SICKAG/sick_scan #1: ERROR with MRS1XXXX

iljastas, 2018-01-04:
Hi guys,

I have this error if I start the launch file:

[ INFO] [1515076997.947821166]: sick_generic_caller V. 001.000.000
[ INFO] [1515076997.947862275]: Program arguments: /home/ilja/017_sick_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1515076997.947875709]: Program arguments: __name:=sick_mrs_1xxx
[ INFO] [1515076997.947885039]: Program arguments: __log:=/home/ilja/.ros/log/a046d38e-f159-11e7-a1a3-00055d09d397/sick_mrs_1xxx-1.log
[ INFO] [1515076997.971388806]: Waiting 5 seconds for device to connect.
[ INFO] [1515076997.971730968]: Succesfully connected to 192.168.1.55:2111
[ INFO] [1515076997.972407794]: Parameter setting for <active_echo: 0>
[ INFO] [1515076997.972534838]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1515076997.975378463]: Receiving: <STX>sAN SetAccessMode 1<ETX>
[ INFO] [1515076997.975438278]: Sending  : <STX>sMN LMCstopmeas<ETX>
[ INFO] [1515076997.999387585]: Receiving: <STX>sAN LMCstopmeas 0<ETX>
[ INFO] [1515076997.999464507]: Sending  : <STX>sWN SetActiveApplications 1 FEVL 0<ETX>
[ INFO] [1515076998.000134267]: Receiving: <STX>sWA SetActiveApplications<ETX>
[ INFO] [1515076998.000197227]: Sending  : <STX>sWN SetActiveApplications 1 RANG 1<ETX>
[ INFO] [1515076998.001052662]: Receiving: <STX>sWA SetActiveApplications<ETX>
[ INFO] [1515076998.001113753]: Sending  : <STX>sRI 0<ETX>
[ INFO] [1515076998.002140249]: Receiving: <STX>sRA DeviceIdent 8 MRS1xxxx 8 1.3.0.0R<ETX>
[ INFO] [1515076998.002200203]: Sending  : <STX>sRN SerialNumber<ETX>
[ INFO] [1515076998.003320055]: Receiving: <STX>sRA SerialNumber 8 17470011<ETX>
[ INFO] [1515076998.003374157]: Deviceinfo sRA DeviceIdent 8 MRS1xxxx 8 1.3.0.0R found and supported by this driver.
[ INFO] [1515076998.003416951]: Sending  : <STX>sRN FirmwareVersion<ETX>
[ INFO] [1515076998.004436720]: Receiving: <STX>sRA FirmwareVersion 9 V 1.3.0.0<ETX>
[ INFO] [1515076998.004501858]: Sending  : <STX>sRN SCdevicestate<ETX>
[ INFO] [1515076998.005580743]: Receiving: <STX>sRA SCdevicestate 2<ETX>
[ERROR] [1515076998.005639405]: Laser reports error state : sRA SCdevicestate 2
[ERROR] [1515076999.006802088]: sendSOPASCommand: failed attempt to read from socket: 125: Operation canceled
[ERROR] [1515076999.006855579]: sendSOPASCommand: no full reply available for read after 1s
[ERROR] [1515076999.006890173]: SOPAS - Error rebooting scanner
[ INFO] [1515076999.006966029]: Sending  : <STX>sRN ODoprh<ETX>
[ERROR] [1515076999.006984847]: sendSOPASCommand: socket not open
[ INFO] [1515076999.007028953]: Receiving: <STX><ETX>
[ERROR] [1515076999.007052371]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sRN ODoprh<ETX>

[ERROR] [1515076999.007076984]: Error reading operation hours
[ INFO] [1515076999.007131312]: Sending  : <STX>sRN ODpwrc<ETX>
[ERROR] [1515076999.007152344]: sendSOPASCommand: socket not open
[ INFO] [1515076999.007183658]: Receiving: <STX><ETX>
[ERROR] [1515076999.007204944]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sRN ODpwrc<ETX>

[ERROR] [1515076999.007234094]: Error reading operation power on counter
[ INFO] [1515076999.007287398]: Sending  : <STX>sRN LocationName<ETX>
[ERROR] [1515076999.007307739]: sendSOPASCommand: socket not open
[ INFO] [1515076999.007339295]: Receiving: <STX><ETX>
[ERROR] [1515076999.007359976]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sRN LocationName<ETX>

[ERROR] [1515076999.007387786]: Error reading Locationname
[ WARN] [1515076999.007421179]: Location command not supported.

[ INFO] [1515076999.008342175]: Sending  : <STX>sRN LMPoutputRange<ETX>
[ERROR] [1515076999.008366846]: sendSOPASCommand: socket not open
[ INFO] [1515076999.008402680]: Receiving: <STX><ETX>
[ERROR] [1515076999.008425305]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sRN LMPoutputRange<ETX>

[ INFO] [1515076999.008464262]: MIN_ANG:   -2.400 [rad] -137.500 [deg]
[ INFO] [1515076999.008488861]: MAX_ANG:    2.400 [rad]  137.500 [deg]
[ INFO] [1515076999.008542727]: Sending  : <STX>sWN LMPoutputRange 1 9C4 FFF8C088 22B6B8<ETX>
[ERROR] [1515076999.008563299]: sendSOPASCommand: socket not open
[ INFO] [1515076999.008602658]: Receiving: <STX><ETX>
[ERROR] [1515076999.008624005]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sWN LMPoutputRange 1 9C4 FFF8C088 22B6B8<ETX>

[ INFO] [1515076999.008677604]: Sending  : <STX>sRN LMPoutputRange<ETX>
[ERROR] [1515076999.008698474]: sendSOPASCommand: socket not open
[ INFO] [1515076999.008736688]: Receiving: <STX><ETX>
[ERROR] [1515076999.008757165]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sRN LMPoutputRange<ETX>

[ INFO] [1515076999.008814783]: Sending  : <STX>sWN LMDscandatacfg 07 00 1 0 0 00 00 0 0 0 0 1<ETX>
[ERROR] [1515076999.008835590]: sendSOPASCommand: socket not open
[ INFO] [1515076999.008868760]: Receiving: <STX><ETX>
[ERROR] [1515076999.008889595]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sWN LMDscandatacfg 07 00 1 0 0 00 00 0 0 0 0 1<ETX>

[ INFO] [1515076999.009476326]: Sending  : <STX>sWN FREchoFilter 1<ETX>
[ERROR] [1515076999.009500138]: sendSOPASCommand: socket not open
[ INFO] [1515076999.009534967]: Receiving: <STX><ETX>
[ERROR] [1515076999.009557541]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sWN FREchoFilter 1<ETX>

[ INFO] [1515076999.009620557]: Sending  : <STX>sMN LMCstartmeas<ETX>
[ERROR] [1515076999.009641957]: sendSOPASCommand: socket not open
[ INFO] [1515076999.009676373]: Receiving: <STX><ETX>
[ERROR] [1515076999.009696971]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN LMCstartmeas<ETX>

[ INFO] [1515076999.009748052]: Sending  : <STX>sMN Run<ETX>
[ERROR] [1515076999.009768864]: sendSOPASCommand: socket not open
[ INFO] [1515076999.009800831]: Receiving: <STX><ETX>
[ERROR] [1515076999.009818613]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN Run<ETX>

[ INFO] [1515076999.009868467]: Sending  : <STX>sRN SCdevicestate<ETX>
[ERROR] [1515076999.009884193]: sendSOPASCommand: socket not open
[ INFO] [1515076999.009903973]: Receiving: <STX><ETX>
[ERROR] [1515076999.009916288]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sRN SCdevicestate<ETX>

[ INFO] [1515076999.009949832]: STATE: 

[ INFO] [1515076999.009967064]: Waiting for scanner ready state since 0 secs
[ INFO] [1515077000.010123397]: Sending  : <STX>sRN SCdevicestate<ETX>
[ERROR] [1515077000.010148778]: sendSOPASCommand: socket not open
[ INFO] [1515077000.010189941]: Receiving: <STX><ETX>
[ERROR] [1515077000.010200525]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sRN SCdevicestate<ETX>

Does anybody have an idea?

Best regards,
Ilja

iljastas, 2018-01-05:
Hi guys,

I fixed it. The SOPAS-Tool showed an errror (also the Scanner was blinking red instead of lighting green).
I switched als I/O-Pins to Input and the error was gone and the ros package was working fine.

Best regards,
Ilja

closed 2018-01-17 by michael1309

omerdurmus61, 2023-03-02:
> Hi guys,
> 
> I fixed it. The SOPAS-Tool showed an errror (also the Scanner was blinking red instead of lighting green). I switched als I/O-Pins to Input and the error was gone and the ros package was working fine.
> 
> Best regards, Ilja

Hi, can you give more information? Where did you get this error on SOPAS and how did you switched pins.