SICKAG/sick_scan #3: Failed to connect to 192.168.0.4:2112

Yvon-Shong, 2018-01-31:
Hello, I am using RMS1000 SICK Laser on ROS on Ubuntu.
I set the ip address as `192.168.0.4` and netmask as 255.255.255.0 in System setting>>Network.
Use Ethernet to connect with LiDAR.
But it didn't work.
```
SOPAS - Error stopping streaming scan data!
sick_scan driver exiting.
[ INFO] [1517387043.694532699]: Waiting 5000 seconds for device to connect.
[ERROR] [1517387043.694580180]: Failed to connect to 192.168.0.4:2112
[FATAL] [1517387043.694595424]: Could not connect to host 192.168.0.4:2112
[FATAL] [1517387043.694615298]: Failed to init device: 1
[ERROR] [1517387043.694646521]: write error for command: sEN LMDscandata 0
```
What's wrong? Could you please help me?

Yvon-Shong, 2018-01-31:
And I could ping 192.168.0.1

Yvon-Shong, 2018-01-31:
When I change the ip address in launch file as 192.168.0.1:2112
```
[ INFO] [1517387700.193179277]: STATE: 

[ INFO] [1517387700.193233552]: Waiting for scanner ready state since 4 secs
[ INFO] [1517387701.193631073]: Sending  : <STX>sRN SCdevicestate<ETX>
[ERROR] [1517387701.193716266]: sendSOPASCommand: socket not open
[ INFO] [1517387701.193855573]: Receiving: <STX><ETX>
[ERROR] [1517387701.193927224]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sRN SCdevicestate<ETX>
```

michael1309, 2018-01-31:
Yvon, can you reach the scanner via Sopas ET?
The scanner must be set to ascii-communication.
Do you use a MRS1104? 

Best regards, michael
-- 
Ing.-Büro Dr. Michael Lehning 
Schulstraße. 21B
D-31137 Hildesheim 
Germany

E: Michael.lehning@lehning.de
T: +49 5121 645-70
M: +49 173 37 69 242

michael1309, 2018-01-31:
Did you check your firewall?
-- 
Ing.-Büro Dr. Michael Lehning 
Schulstraße. 21B
D-31137 Hildesheim 
Germany

E: Michael.lehning@lehning.de
T: +49 5121 645-70
M: +49 173 37 69 242

Yvon-Shong, 2018-02-01:
@michael1309 Thanks for your reply.
Yeah I am using **MRS1104C-111011** LiDAR. This driver didn't support it?
I am using a clean Ubuntu without firewall.
And I have tested the SOPAS software in Windows, and it works, I could see the information of point cloud.

Yeah now I want to make sure **the way to set the ip address.**
In network of system setting, I set the IPV4 settings

```
Address: 192.168.0.4
Netmask: 255.255.255.0
Gateway: none
```

and in the `sick_mrs_1xxx.launch` file of `sick_scan-release-release-kinetic-sick_scan`

```xml
<param name="hostname" type="string" value="192.168.0.4"/>
<param name="port" type="string" value="2112" />
```

Are they right?
I am not clear about how the device work by the ip of Ethernet interface.

michael1309, 2018-02-01:
Use telnet for further debugging:
PROMPT$ telnet 192.168.0.4 2112
Output in case of connection error (e.g. cable problem)
**Trying 192.168.0.4...
telnet: Unable to connect to remote host: Connection refused**

Output if scanner is reachable:
PROMPT$ telnet 192.168.0.4 2112
**Trying 192.168.0.4...
Connected to 192.168.0.4.
Escape character is '^]'.**

michael1309, 2018-02-01:
If there are still errors, try to redirect stderr and stdout to a small log file and post it here:
PROMPT$  **roslaunch sick_scan sick_mrs_1xxx.launch > tinylog.txt 2>&1**
I did it for a couple of possible problems. Please compare your output against the log files attached:

- Correct start: [launch_started_correctly.txt](https://github.com/SICKAG/sick_scan/files/1684984/launch_started_correctly.txt)
- Binary protocol is activated by SOPAS ET: 
[launch_binary_protocol_activated.txt](https://github.com/SICKAG/sick_scan/files/1684978/launch_binary_protocol_activated.txt)
Fixing: Switch to ASCII-Mode using SOPAS ET and save parameters permanently to the scanner
- [launch_log_not_reachable.txt](https://github.com/SICKAG/sick_scan/files/1684981/launch_log_not_reachable.txt)
Fixing: Check cable connection, repower scanner, check firewall connection, try to connect via telnet (e.g. **telnet 192.168.0.4 2112**)
- [launch_log_notsourced.txt](https://github.com/SICKAG/sick_scan/files/168498/launch_log_notsourced.txt)
Fixing: source bash for ROS: **source ./devel/setup.bash**  (started from catkin-workspace-directory)

michael1309, 2018-02-01:
You are using a **MRS1104C-111011**. I tested the firmware against a **MRS1104C-011010**.
I guess that is not the reason for you communication problems.
Good luck!

Yvon-Shong, 2018-02-03:
I try another driver for it (sick_tim) and it work.
still thank you.

closed 2018-02-03 by Yvon-Shong

michael1309, 2018-02-03:
Great to hear. If you have any further information about the previous connection problem using sick_scan please feel free to support.