SICKAG/sick_scan #4: Mismatch Answer for request

Jtlie, 2018-03-01:
Hi, I'm using the Sick Tim561 but I'm getting some errors about mismatched receiving data. 

`[ INFO] [1519902189.025968221]: sick_generic_caller V. 001.000.000
[ INFO] [1519902189.026012121]: Program arguments: /home/jason/catkin_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1519902189.026022885]: Program arguments: __name:=sick_tim_5xx
[ INFO] [1519902189.026029678]: Program arguments: __log:=/home/jason/.ros/log/4f775858-1ba1-11e8-ab10-40e230b7438b/sick_tim_5xx-1.log
[ INFO] [1519902189.073421116]: Waiting 5 seconds for device to connect.
[ INFO] [1519902189.073971742]: Succesfully connected to 169.254.147.31:2112
[ INFO] [1519902189.075414631]: Parameter setting for <active_echo: 0>
[ INFO] [1519902189.075610429]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1519902189.077854856]: Receiving: <STX>sAN SetAccessMode 0<ETX>
[ INFO] [1519902189.077886860]: Receiving: <STX>sAN SetAccessMode 0<ETX>
[ INFO] [1519902189.077950981]: Sending  : <STX>sRI 0<ETX>
[ INFO] [1519902189.078980755]: Receiving: <STX>sRA 0 6 TiM561 E V2.51-21.08.15<ETX>
[ INFO] [1519902189.079001884]: Receiving: <STX>sRA 0 6 TiM561 E V2.51-21.08.15<ETX>
[ INFO] [1519902189.079049914]: Sending  : <STX>sRN SerialNumber<ETX>
[ INFO] [1519902189.080129703]: Receiving: <STX>sRA SerialNumber 8 16160456<ETX>
[ INFO] [1519902189.080149822]: Receiving: <STX>sRA SerialNumber 8 16160456<ETX>
[ INFO] [1519902189.080215428]: Device sRA 0 6 TiM561 E V2.51-21.08.15 V2.51 found and supported by this driver.
[ INFO] [1519902189.080252351]: Sending  : <STX>sRN FirmwareVersion<ETX>
[ INFO] [1519902189.081532916]: Receiving: <STX>sRA FirmwareVersion 5 V2.51<ETX>
[ INFO] [1519902189.081552010]: Receiving: <STX>sRA FirmwareVersion 5 V2.51<ETX>
[ INFO] [1519902189.081607405]: Sending  : <STX>sRN SCdevicestate<ETX>
[ INFO] [1519902189.082697896]: Receiving: <STX>sRA SCdevicestate 1<ETX>
[ INFO] [1519902189.082721769]: Receiving: <STX>sRA SCdevicestate 1<ETX>
[ INFO] [1519902189.082772175]: Sending  : <STX>sRN ODoprh<ETX>
[ INFO] [1519902189.083793114]: Receiving: <STX>sRA ODoprh 924<ETX>
[ INFO] [1519902189.083820096]: Receiving: <STX>sRA ODoprh 924<ETX>
[ INFO] [1519902189.084962675]: Sending  : <STX>sRN ODpwrc<ETX>
[ INFO] [1519902189.085990256]: Receiving: <STX>sRA ODpwrc AC<ETX>
[ INFO] [1519902189.086023284]: Receiving: <STX>sRA ODpwrc AC<ETX>
[ INFO] [1519902189.087135245]: Sending  : <STX>sRN LocationName<ETX>
[ INFO] [1519902189.088178447]: Receiving: <STX>sRA LocationName B not defined<ETX>
[ INFO] [1519902189.088207784]: Receiving: <STX>sRA LocationName B not defined<ETX>
[ INFO] [1519902189.089231776]: Sending  : <STX>sRN LMPoutputRange<ETX>
[ INFO] [1519902189.090314741]: Receiving: <STX>sRA LMPoutputRange 1 D05 FFF92230 225510<ETX>
[ INFO] [1519902189.090358735]: Receiving: <STX>sRA LMPoutputRange 1 D05 FFF92230 225510<ETX>
[ INFO] [1519902189.090434324]: Angle resolution of scanner is 0.33330 [deg]  (in 1/10000th deg: 0xD05)
[ INFO] [1519902189.090458610]: MIN_ANG:   -2.356 [rad] -135.000 [deg]
[ INFO] [1519902189.090479761]: MAX_ANG:    2.356 [rad]  135.000 [deg]
[ INFO] [1519902189.090527082]: Sending  : <STX>sWN LMPoutputRange 1 D05 FFF92230 225510<ETX>
[ INFO] [1519902189.091556183]: Receiving: <STX>sFA A<ETX>
[ INFO] [1519902189.091586593]: Receiving: <STX>sFA A<ETX>
[ERROR] [1519902189.091643015]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX>sWN LMPoutputRange 1 D05 FFF92230 225510<ETX>Answer= >>>sFA A<<<

[ INFO] [1519902189.091723287]: Sending  : <STX>sRN LMPoutputRange<ETX>
[ INFO] [1519902189.092751966]: Receiving: <STX>sRA LMPoutputRange 1 D05 FFF92230 225510<ETX>
[ INFO] [1519902189.092783042]: Receiving: <STX>sRA LMPoutputRange 1 D05 FFF92230 225510<ETX>
[ INFO] [1519902189.095330154]: MIN_ANG (after command verification):   -2.356 [rad] -135.000 [deg]
[ INFO] [1519902189.095375359]: MAX_ANG (after command verification):    2.356 [rad]  135.000 [deg]
[ INFO] [1519902189.095469453]: Sending  : <STX>sWN LMDscandatacfg 01 00 0 0 0 00 00 0 0 0 0 1<ETX>
[ INFO] [1519902189.096527098]: Receiving: <STX>sFA A<ETX>
[ INFO] [1519902189.096567822]: Receiving: <STX>sFA A<ETX>
[ERROR] [1519902189.096641636]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX>sWN LMDscandatacfg 01 00 0 0 0 00 00 0 0 0 0 1<ETX>Answer= >>>sFA A<<<

[ INFO] [1519902189.096741712]: Sending  : <STX>sMN LMCstartmeas<ETX>
[ INFO] [1519902189.097746837]: Receiving: <STX>sFA 1<ETX>
[ INFO] [1519902189.097780605]: Receiving: <STX>sFA 1<ETX>
[ERROR] [1519902189.097845188]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX>sMN LMCstartmeas<ETX>Answer= >>>sFA 1<<<

[ INFO] [1519902189.097920893]: Sending  : <STX>sMN Run<ETX>
[ INFO] [1519902189.099076002]: Receiving: <STX>sAN Run 1<ETX>
[ INFO] [1519902189.099110219]: Receiving: <STX>sAN Run 1<ETX>
[ INFO] [1519902189.099193208]: Sending  : <STX>sEN LMDscandata 1<ETX>
[ INFO] [1519902189.100258410]: Receiving: <STX>sEA LMDscandata 1<ETX>
[ INFO] [1519902189.100299531]: Receiving: <STX>sEA LMDscandata 1<ETX>
`
Any idea what is causing this? I did change Intensity to FALSE in the parameters because I don have RSSI enabled.

michael1309, 2018-03-01:
> [ INFO] [1519902189.075610429]: Sending : sMN SetAccessMode 3 F4724744
> [ INFO] [1519902189.077854856]: Receiving: sAN SetAccessMode 0
> [ INFO] [1519902189.077886860]: Receiving: sAN SetAccessMode 0
> [ INFO] [1519902189.077950981]: Sending : sRI 0
> [ INFO] [1519902189.078980755]: Receiving: sRA 0 6 TiM561 E V2.51-21.08.15
> [ INFO] [1519902189.079001884]: Receiving: sRA 0 6 TiM561 E V2.51-21.08.15
> [ INFO] [1519902189.079049914]: Sending : sRN SerialNumber
> [ INFO] [1519902189.080129703]: Receiving: sRA SerialNumber 8 16160456
> [ INFO] [1519902189.080149822]: Receiving: sRA SerialNumber 8 16160456
> [ INFO] [1519902189.080252351]: Sending : sRN FirmwareVersion
> [ INFO] [1519902189.081532916]: Receiving: sRA FirmwareVersion 5 V2.51
> [ INFO] [1519902189.081552010]: Receiving: sRA FirmwareVersion 5 V2.51
> [ INFO] [1519902189.081607405]: Sending : sRN SCdevicestate
> [ INFO] [1519902189.082697896]: Receiving: sRA SCdevicestate 1
> [ INFO] [1519902189.082721769]: Receiving: sRA SCdevicestate 1
> [ INFO] [1519902189.082772175]: Sending : sRN ODoprh
> [ INFO] [1519902189.083793114]: Receiving: sRA ODoprh 924
> [ INFO] [1519902189.083820096]: Receiving: sRA ODoprh 924
> [ INFO] [1519902189.084962675]: Sending : sRN ODpwrc
> [ INFO] [1519902189.085990256]: Receiving: sRA ODpwrc AC
> [ INFO] [1519902189.086023284]: Receiving: sRA ODpwrc AC
> [ INFO] [1519902189.087135245]: Sending : sRN LocationName
> [ INFO] [1519902189.088178447]: Receiving: sRA LocationName B not defined
> [ INFO] [1519902189.088207784]: Receiving: sRA LocationName B not defined
> [ INFO] [1519902189.089231776]: Sending : sRN LMPoutputRange
> [... 20 lines cut ...]
> [ INFO] [1519902189.099076002]: Receiving: sAN Run 1
> [ INFO] [1519902189.099110219]: Receiving: sAN Run 1
> [ INFO] [1519902189.099193208]: Sending : sEN LMDscandata 1
> [ INFO] [1519902189.100258410]: Receiving: sEA LMDscandata 1
> [ INFO] [1519902189.100299531]: Receiving: sEA LMDscandata 1

Dear Jason,

please wait about 60 secs. for reconnection to your scanner after
terminating a connection or repower the scanner before reconnection.
Sometimes the firmware does not answer correctly if you reconnect too fast.
Also you can try the devel branch of the driver. If you further receive
failure messages please give me as short note. On Mon, 12th of March, I
have access to the TiM561. Than I will investigate the phenoma if this
problem
is not fixed by these hints.

Best regards,
Michael

Jtlie, 2018-03-01:
Waiting after repowering the scanner didn't work. Also, the output when using the devel branch is different, but didn´t solve the problem. 

```
process[sick_tim_5xx-1]: started with pid [29471]
[ INFO] [1519910779.817064301]: sick_generic_caller V. 001.001.000
[ INFO] [1519910779.817109336]: Program arguments: /home/jason/catkin_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1519910779.817124809]: Program arguments: __name:=sick_tim_5xx
[ INFO] [1519910779.817135474]: Program arguments: __log:=/home/jason/.ros/log/4f775858-1ba1-11e8-ab10-40e230b7438b/sick_tim_5xx-1.log
[ INFO] [1519910779.860509036]: Parameter setting for <active_echo: 0>
[ INFO] [1519910780.061174579]: Sending  : sMN SetAccessMode 3 F4724744
[ INFO] [1519910780.071528873]: Receiving: <STX>sAN SetAccessMode 0<ETX>
[ INFO] [1519910780.272029881]: Sending  : sWN EIHstCola 0
[ INFO] [1519910780.282452666]: Receiving: <STX>sFA A<ETX>
[ERROR] [1519910780.282617714]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX>sWN EIHstCola 0<ETX>Answer= >>>sFA A<<<

[ERROR] [1519910780.282725426]: Command or Error message not defined
[ INFO] [1519910780.483162603]: Sending  : sRN FirmwareVersion
[ INFO] [1519910780.493542685]: Receiving: <STX>sRA FirmwareVersion 5 V2.51<ETX>
[ INFO] [1519910780.694193493]: Sending  : sRN SCdevicestate
[ INFO] [1519910780.704624368]: Receiving: <STX>sRA SCdevicestate 1<ETX>
[ INFO] [1519910780.905007449]: Sending  : sRN ODoprh
[ INFO] [1519910780.915411480]: Receiving: <STX>sRA ODoprh 93B<ETX>
[ INFO] [1519910781.119059326]: Sending  : sRN ODpwrc
[ INFO] [1519910781.129356579]: Receiving: <STX>sRA ODpwrc AD<ETX>
[ INFO] [1519910781.330832339]: Sending  : sRN LocationName
[ INFO] [1519910781.341228564]: Receiving: <STX>sRA LocationName B not defined<ETX>
[ INFO] [1519910781.343884873]: Sending  : sRN LMPoutputRange
[ INFO] [1519910781.354219595]: Receiving: <STX>sRA LMPoutputRange 1 D05 FFF92230 225510<ETX>
[ INFO] [1519910781.354480550]: Angle resolution of scanner is 0.33330 [deg]  (in 1/10000th deg: 0xD05)
[ INFO] [1519910781.354539767]: MIN_ANG:   -2.356 [rad] -135.000 [deg]
[ INFO] [1519910781.354579507]: MAX_ANG:    2.356 [rad]  135.000 [deg]
[ INFO] [1519910781.354749847]: Sending  : sWN LMPoutputRange 1 D05 FFF92230 225510
[ INFO] [1519910781.365168638]: Receiving: <STX>sFA A<ETX>
[ERROR] [1519910781.365358291]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX>sWN LMPoutputRange 1 D05 FFF92230 225510<ETX>Answer= >>>sFA A<<<

[ INFO] [1519910781.365506066]: Sending  : sRN LMPoutputRange
[ INFO] [1519910781.375732332]: Receiving: <STX>sRA LMPoutputRange 1 D05 FFF92230 225510<ETX>
[ INFO] [1519910781.375869776]: Angle resolution of scanner is 0.33330 [deg]  (in 1/10000th deg: 0xD05)
[ INFO] [1519910781.378633903]: MIN_ANG (after command verification):   -2.356 [rad] -135.000 [deg]
[ INFO] [1519910781.378675212]: MAX_ANG (after command verification):    2.356 [rad]  135.000 [deg]
[ INFO] [1519910781.378768146]: Sending  : sWN LMDscandatacfg 01 00 0 0 0 00 00 0 0 0 0 1
[ INFO] [1519910781.388978240]: Receiving: <STX>sFA A<ETX>
[ERROR] [1519910781.389120565]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX>sWN LMDscandatacfg 01 00 0 0 0 00 00 0 0 0 0 1<ETX>Answer= >>>sFA A<<<

[ INFO] [1519910781.389245196]: Sending  : sRN LMDscandatacfg
[ INFO] [1519910781.399625072]: Receiving: <STX>sRA LMDscandatacfg 1 0 0 0 0 0 0 0 1 0 0 1<ETX>
[ INFO] [1519910781.399966601]: Sending  : sMN LMCstartmeas
[ INFO] [1519910781.410266222]: Receiving: <STX>sFA 1<ETX>
[ERROR] [1519910781.410440148]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX>sMN LMCstartmeas<ETX>Answer= >>>sFA 1<<<

[ERROR] [1519910781.410532886]: Command or Error message not defined
[ INFO] [1519910781.410615640]: Sending  : sMN Run
[ INFO] [1519910781.420871459]: Receiving: <STX>sAN Run 1<ETX>
[ INFO] [1519910781.421096550]: Sending  : sEN LMDscandata 1
[ INFO] [1519910781.431369353]: Receiving: <STX>sEA LMDscandata 1<ETX>
```

kaidoho, 2018-03-01:
Hi Jason, please follow Michael's advice and switch to the devel branch. This works, at least with my TiM561, quite well. If that doesn't work out - post your launch file.

kaidoho, 2018-03-01:
Hi Jason,

could you post your launch file? Things work out fine with my TIM561 over here.

Jtlie, 2018-03-01:
```
<?xml version="1.0"?>
<!-- start and stop angle is given in [rad] -->
<!-- 
default min_angle is  -135 degree. 
default max_angle is +135 degree.

Check IP-address, if you scanner is not found after roslaunch.
-->
<launch>
<!-- launch file for the tim_5xx series -->
  <param name="robot_description" command="$(find xacro)/xacro.py '$(find sick_scan)/urdf/example.urdf.xacro'" />
  <node name="sick_tim_5xx" pkg="sick_scan" type="sick_generic_caller"  respawn="false" output="screen">

    <!-- Default: min_angle = -135 degree, max_angle = 135 -->
    <!-- default values: -->
    <!--
      <param name="min_ang" type="double" value="-2.35619449" />
      <param name="max_ang" type="double" value="2.35619449" />
      <param name="intensity" type="bool" value="True" />
      <param name="skip" type="int" value="0" />
      <param name="frame_id" type="str" value="laser" />
      <param name="time_offset" type="double" value="-0.001" />
      <param name="publish_datagram" type="bool" value="False" />
      <param name="subscribe_datagram" type="bool" value="false" />
      <param name="device_number" type="int" value="0" />
      <param name="range_min" type="double" value="0.05" />
    -->
    <param name="min_ang" type="double" value="-2.35619449" /><!-- -135° -->
    <param name="max_ang" type="double" value="2.35619449" /><!-- 135° -->
    <param name="range_max" type="double" value="100.0" />
    <param name="intensity" type="bool" value="False" />
    <param name="hostname" type="string" value="169.254.147.31" />
    <param name="port" type="string" value="2112" />
    <param name="timelimit" type="int" value="5" />
    
  </node>
</launch>
<!--
Conversion between degree and rad	
	
DEG	RAD
-180	-3.141592654
-175	-3.054326191
-170	-2.967059728
-165	-2.879793266
-160	-2.792526803
-155	-2.705260341
-150	-2.617993878
-145	-2.530727415
-140	-2.443460953
-137.5	-2,3998277
-135	-2.35619449
-130	-2.268928028
-125	-2.181661565
-120	-2.094395102
-115	-2.00712864
-110	-1.919862177
-105	-1.832595715
-100	-1.745329252
-95	-1.658062789
-90	-1.570796327
-85	-1.483529864
-80	-1.396263402
-75	-1.308996939
-70	-1.221730476
-65	-1.134464014
-60	-1.047197551
-55	-0.959931089
-50	-0.872664626
-45	-0.785398163
-40	-0.698131701
-35	-0.610865238
-30	-0.523598776
-25	-0.436332313
-20	-0.34906585
-15	-0.261799388
-10	-0.174532925
-5	-0.087266463
0	0
5	0.087266463
10	0.174532925
15	0.261799388
20	0.34906585
25	0.436332313
30	0.523598776
35	0.610865238
40	0.698131701
45	0.785398163
50	0.872664626
55	0.959931089
60	1.047197551
65	1.134464014
70	1.221730476
75	1.308996939
80	1.396263402
85	1.483529864
90	1.570796327
95	1.658062789
100	1.745329252
105	1.832595715
110	1.919862177
115	2.00712864
120	2.094395102
125	2.181661565
130	2.268928028
135	2.35619449
137.5	2,3998277
140	2.443460953
145	2.530727415
150	2.617993878
155	2.705260341
160	2.792526803
165	2.879793266
170	2.967059728
175	3.054326191
180	3.141592654
185	3.228859116
190	3.316125579
195	3.403392041
200	3.490658504
205	3.577924967
210	3.665191429
215	3.752457892
220	3.839724354
225	3.926990817
230	4.01425728
235	4.101523742
240	4.188790205
245	4.276056667
250	4.36332313
255	4.450589593
260	4.537856055
265	4.625122518
270	4.71238898
275	4.799655443
280	4.886921906
285	4.974188368
290	5.061454831
295	5.148721293
300	5.235987756
305	5.323254219
310	5.410520681
315	5.497787144
320	5.585053606
325	5.672320069
330	5.759586532
335	5.846852994
340	5.934119457
345	6.021385919
350	6.108652382
355	6.195918845
360	6.283185307
-->
```

This is the one I got when I pulled to devel branch. Only thing I changed again are the hostname and intensity

kaidoho, 2018-03-01:
Hi, 
a) the log shows that your sensor is up and running
b) your launch file works fine with my TiM
However, what we can see from the sensor reply

sAN SetAccessMode 0

that you are unable to log on as user: "service" with password "servicelevel". Mhh what could be a reason for that? On the one hand, the number of connections is limited and you've opened too many. On the other hand, what I've just found out about the TiMs is,  you can change the password. Can you try to log on using SOPAS ET with the above level / password?

kaidoho, 2018-03-01:
Hi,
a) the log shows that your sensor is up and running
b) your launch file works fine with my TiM
However, what we can see from the sensor reply

sAN SetAccessMode 0

that you are unable to log on as user: "Authorized Client" with password "client". Mhh what could be a reason for that? On the one hand, the number of connections is limited and you've opened too many. On the other hand, what I've just found out about the TiMs is, you can change the password. Can you try to log on using SOPAS ET with the above level / password?

Jtlie, 2018-03-05:
Where do u see it using "client" as password? The only thing I see is "Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>".

michael1309, 2018-03-05:
The code is a so called hash code corresponding to the password "client".
You can check this by using sopas et and try to connect as authorized
client. Otherwise you can reset your scanner to factory default. After
resetting and setting the correct ip address the hash code should work.

Best regards, michael

Jtlie, 2018-03-07:
It seems like the password has indeed changed since I can't login with Sopas ET. I've tried to reset it to factory default using the command in the Sopas ET terminal but It also give me a Access denied error. Is there another way to do this?

michael1309, 2018-03-07:
You can try to login with the service level. Be aware of any changes you do after login in this level.

kaidoho, 2018-03-13:
Dear Jason,

I am in a hurry so here's a short answer. Basically you've got two Options.

a)  Return the sensor to SICK AG and let the password be reset. According to the README of this repository you should do this via https://supportportal.sick.com.

b) Modify your local copy of this repository. See document:

https://sick-virginia.data.continum.net/media/docs/7/27/927/Technical_information_Telegram_Listing_Ranging_sensors_LMS1xx_LMS5xx_TiM5xx_MRS1000_MRS6000_NAV310_LD_OEM15xx_LD_LRS36xx_en_IM0045927.PDF

On page 12, several login levels and their passwords are published. Go through the code and whereever you find a login with user level 3 (Authorized Client)

<STX>sMN SetAccessMode 3 F4724744<ETX>

change it to

<STX>sMN SetAccessMode 4 81BE23AA<ETX>

If someone has also changed the service password -> goto a).

Regards

michael1309, 2018-03-14:
Dear Jason, 

After login to service level you can also try to set the password for the client level regarding table 67 in the mentioned document.

The command <STX>sMN{SPC}SetPassword{SPC}03{SPC}F4 72 47 44<ETX> should reset the password for the authorized client mode to default.

michael1309, 2018-04-11:
Since there was no further feedback after the last hints from our side, we would like to close the ticket now. From your point of view, is there anything wrong with it? If not, we'd be closing it by April 15th.

closed 2018-04-18 by michael1309