SICKAG/sick_scan #5: Using ROS node name to define lidar type

baronep, 2018-03-14:
The issue in a nutshell is that the ROS node name is being used to specify the type of laser that sick_generic_caller is connecting to.

Given the provided launch file, everything works as expected with 1 lidar ...
```
<launch>
<node name="sick_tim_5xx" pkg="sick_scan" type="sick_generic_caller">
#Parameters
</node>
</launch>
```

The issue arises when we want to run multiple of the same type. The standard way to configure multiple ros nodes (and avoid using the discouraged anonymous parameter), is by changing this node name such that each node name is unique.

```
<launch>
<node name="sick_tim_5xx_1" pkg="sick_scan" type="sick_generic_caller">
#Parameters
</node>
<node name="sick_tim_5xx_2" pkg="sick_scan" type="sick_generic_caller">
#Parameters
</node>
</launch>
```

The above approach however will fail with the error `Scanner not supported`. Instead, the type of laser that that the caller is attempting to connect to, should be specified in a ros parameter and not using the node name. Alternatively, there could be multiple nodes, 1 for each type of laser.

michael1309, 2018-03-14:
That is a good point. I will discuss it with SICK. If they agree, we will introduce a new parameter field that specifies the scanner type. I will come back to this issue next week.

michael1309, 2018-03-15:
We pushed a new version of the software. The software reads a parameter "scanner_type" from the launch file. The scanner type must correspond to the specific laser scanner type. In your case "sick_tim_5xx". 
To ensure that the combination of two scanners of the same type runs I combined two scanners and prepared a launch file as an example, how you can combine these scanners.
The corresponding launch file is called **sick_tim_5xx_twin.launch** and already uploaded.

Enjoy it!

michael1309, 2018-04-11:
Since we believe that this requirement has been resolved with the new software versions, we would like to close this ticket. From your point of view, is there anything wrong with it? If not, we'd be closing it by April 15th.

closed 2018-04-18 by michael1309