SICKAG/sick_scan #6: no answer received after 2000 ms. Maybe sopas mode is wrong.

yaowenx, 2018-03-26:
Hi, I'm trying to use the driver for mrs6124, but roslaunch gives the error as below:
```
[ INFO] [1522075082.550742490]: Start initialising scanner ...
[ INFO] [1522075082.584104032]: Parameter setting for <active_echo: 0>
[ INFO] [1522075082.784523439]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
[ERROR] [1522075084.806660905]: no answer received after 2000 ms. Maybe sopas mode is wrong.

[ERROR] [1522075084.806721003]: sendSOPASCommand: no full reply available for read after 1s
[ INFO] [1522075084.806823203]: Receiving: <STX><ETX>
[ERROR] [1522075084.806858226]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX>sMN SetAccessMode <ETX>rGD

[ INFO] [1522075084.806980677]: Sending  : sMN SetAccessMode 3 F4724744
[ INFO] [1522075084.817277696]: Receiving: <STX>sAN SetAccessMode 1<ETX>
[ INFO] [1522075085.017781561]: Sending  : sWN EIHstCola 1
[ INFO] [1522075085.028183130]: Receiving: <STX>sWA EIHstCola<ETX>
[FATAL] [1522075085.028697912]: Failed to init scanner: 1
[ INFO] [1522075085.028967663]: Start initialising scanner ...
```

I'm quite new with sick sensors. Is there a option where I can change sopas mode as the error message suggests?

Thanks

michael1309, 2018-03-26:
I recommend to reprogram the scanner to COLA-Binary-protocol permanently. You can do this via using SOPAS ET after login to "authorized client" level. After switching you should save the parameter to the internal memory of the scanner (EEPROM).

yaowenx, 2018-03-27:
Thank you for the help. I've changed the protocol using SOPAS ET. But a new error occured:
```
[ INFO] [1522139498.893607582]: Start initialising scanner ...
[ INFO] [1522139498.937214433]: Parameter setting for <active_echo: 0>
[ INFO] [1522139499.138351702]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
[ INFO] [1522139499.148951484]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1522139499.349845796]: Sending  : sWN EIHstCola \x01
[ INFO] [1522139499.360423893]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1522139499.561644080]: Sending  : sMN LMCstopmeas
[ INFO] [1522139500.585088572]: Receiving: <STX>sAN LMCstopmeas \x00<ETX>
[ INFO] [1522139500.785889030]: Sending  : sRN DeviceIdent
[ INFO] [1522139500.796518770]: Receiving: <STX>sRA DeviceIdent \x00\x08\x4d\x52\x53\x36\x31\x32\x34\x52\x00\x0a\x31\x2e\x30\x2e\x31\x2e\x33\x33\x36\x52<ETX>
IdentLen: 8
[ INFO] [1522139500.797202126]: Deviceinfo MRS6124R V1.0.1.336R found and supported by this driver.
[ INFO] [1522139500.998284324]: Sending  : sRN FirmwareVersion
[ INFO] [1522139501.008877317]: Receiving: <STX>sRA FirmwareVersion \x00\x0a\x31\x2e\x30\x2e\x31\x2e\x33\x33\x36\x52<ETX>
[ INFO] [1522139501.209769751]: Sending  : sRN SCdevicestate
[ INFO] [1522139501.220375401]: Receiving: <STX>sRA SCdevicestate \x01<ETX>
[ INFO] [1522139501.421244748]: Sending  : sRN ODoprh
[ INFO] [1522139501.432008044]: Receiving: <STX>sRA ODoprh \x00\x00\x08\x7b<ETX>
[ INFO] [1522139501.636605832]: Sending  : sRN ODpwrc
[ INFO] [1522139501.647164790]: Receiving: <STX>sRA ODpwrc \x00\x00\x00\x4d<ETX>
[ INFO] [1522139501.853567867]: Sending  : sRN LocationName
[ INFO] [1522139501.864100890]: Receiving: <STX>sRA LocationName \x00\x0b\x6e\x6f\x74\x20\x64\x65\x66\x69\x6e\x65\x64<ETX>
[ INFO] [1522139501.868801843]: Sending  : sRN LMPoutputRange
[ INFO] [1522139501.879325019]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x05\x14\x00\x04\x93\xe0\x00\x16\xe3\x60<ETX>
[ INFO] [1522139501.879681206]: MIN_ANG:   -1.047 [rad]  -60.000 [deg]
[ INFO] [1522139501.879836914]: MAX_ANG:    1.047 [rad]   60.000 [deg]
[ INFO] [1522139501.880163266]: Sending  : sWN LMPoutputRange \x00\x01\x00\x00\x05\x14\x00\x04\x93\xe0\x00\x16\xe3\x60
[ INFO] [1522139501.890516571]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1522139501.890699259]: Sending  : sRN LMPoutputRange
[ INFO] [1522139501.900931632]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x05\x14\x00\x04\x93\xe0\x00\x16\xe3\x60<ETX>
[ INFO] [1522139501.903980528]: MIN_ANG (after command verification):   -1.047 [rad]  -60.000 [deg]
[ INFO] [1522139501.904015460]: MAX_ANG (after command verification):    1.047 [rad]   60.000 [deg]
[ INFO] [1522139501.904139271]: Sending  : sWN LMDscandatacfg \x1f\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x01
[ INFO] [1522139501.914323634]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1522139501.914440747]: Sending  : sRN LMDscandatacfg
[ INFO] [1522139501.924636436]: Receiving: <STX>sRA LMDscandatacfg \x1f\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x01<ETX>
[ INFO] [1522139501.925559782]: Sending  : sWN FREchoFilter \x02
[ INFO] [1522139501.935715882]: Receiving: <STX>sWA FREchoFilter <ETX>
[ INFO] [1522139501.935850195]: Sending  : sMN LMCstartmeas
[ INFO] [1522139502.468069147]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1522139502.468548681]: Sending  : sMN Run
[ INFO] [1522139502.479135372]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1522139502.479625218]: Sending  : sEN LMDscandata \x01
[ INFO] [1522139502.490005015]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ WARN] [1522139504.491332880]: Timeout during waiting of new datagram
[ERROR] [1522139504.491516085]: Read Error when getting datagram: 1.
[ INFO] [1522139504.491904087]: Start initialising scanner ...
```
I can hear the sound the sensor makes. I assume that now the initialization is working but maybe the datagram setting might need change as well?

michael1309, 2018-03-27:
Hello, yaowenx,

We can reproduce that problem. Depending on the scanner status and the scanner firmware, it may take longer than 2 seconds for the first scan data to arrive. 
Therefore, the value for the timeout "maxWaitInMs" should be changed to 30000 ms.  
You can find the timeout in the file driver/src/sick_scan_common_tcp.cpp
This file should then be changed as follows:
```
711	int SickScanCommonTcp::get_datagram(unsigned char* receiveBuffer, int bufferSize, int* actual_length, bool isBinaryProtocol)
	{
712		this->setReplyMode(1);
713		const int maxWaitInMs = 30000; 
```
We will fix this problem in the next release thanks for the hint.

yaowenx, 2018-03-27:
Hi Michael,

Thank you again for the help. Now it is running without error. But the cloud looks a little bit weird
[image]
The part in the black box is a flat wall surface and the view is topdown. But it looks like that there's multiple layers of the wall and each layer consists of small arcs.
I first think it is the multi-echo feature but I tried first echo, all echo and last echo settings, they all looks the same. It is actually the same as what I saw in SOPASair. So it is probably not a bug in your code. But do you have any idea why the cloud looks like this? Maybe something wrong in the firmware? My sensor is using 1.0.1.336R firmware.
Thank you

michael1309, 2018-03-27:
Hello yaowenx,

I think that what you see is not a bug in the scanner or the code,but for objects that are close to the scanner, the distance quantisation of 12.5 cm can be seen in Rviz.
If you don't use a top-down orthographic camera in rviz, there will be additional distortions caused by the projection.   
The MRS6124 delivers impressive recordings in the far field. 
If you have the opportunity, take a test outdoors. 

Best regards Michael

4eyedwit, 2018-04-09:
Hello,
I am also trying to launch the driver for MRS6124R. But even after changing the protocol to Binary, I get the below error. 

```
[ INFO] [1523273349.547360455]: Start initialising scanner ...
1523273349.5475 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273369.927893985]: no answer received after 20000 ms. Maybe sopas mode is wrong.

[ERROR] [1523273369.928035805]: sendSOPASCommand: no full reply available for read after 20000 ms

SOPAS - Error stopping streaming scan data!
[ WARN] [1523273369.928374863]: Disconnecting TCP-Connection.
sick_scan driver exiting.
1523273499.7376 ERROR: Tcp::open: Failed to open TCP connection to 192.168.0.24, aborting.
[ INFO] [1523273499.743618277]: Parameter setting for <active_echo: 0>
[ INFO] [1523273499.944271010]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
1523273499.9444 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273505.039736946]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1523273505.039904852]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1523273505.040206259]: Receiving: <STX><ETX>
[ERROR] [1523273505.040513862]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX>sMN SetAccessMode <ETX>rGD

[ INFO] [1523273505.041279125]: Sending  : sMN SetAccessMode 3 F4724744
1523273505.0414 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273510.135328661]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1523273510.135490299]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1523273510.135741709]: Receiving: <STX><ETX>
[ERROR] [1523273510.135886521]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN SetAccessMode 3 F4724744<ETX>

[ERROR] [1523273510.136251626]: Error Access Mode
[ INFO] [1523273510.336915862]: Sending  : sWN EIHstCola 1
1523273510.3372 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273515.431311557]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1523273515.431508571]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1523273515.431857316]: Receiving: <STX><ETX>
[ERROR] [1523273515.432196197]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sWN EIHstCola 1<ETX>

[ERROR] [1523273515.432697692]: Command or Error message not defined
[ERROR] [1523273515.432890800]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "sopas_protocol_type" to overwrite the default settings of the driver.
[ INFO] [1523273515.433070341]: Start initialising scanner ...
1523273515.4332 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273535.812481859]: no answer received after 20000 ms. Maybe sopas mode is wrong.

[ERROR] [1523273535.812639219]: sendSOPASCommand: no full reply available for read after 20000 ms

SOPAS - Error stopping streaming scan data!
[ WARN] [1523273535.812993309]: Disconnecting TCP-Connection.
sick_scan driver exiting.
1523273665.6256 ERROR: Tcp::open: Failed to open TCP connection to 192.168.0.24, aborting.
```
Is there anything else that I might be missing?

Thanks.

michael1309, 2018-04-09:
Hi,

 

could you ping the ip-address 192.168.0.24?

Did you use the most recent version in the devel branch?

 

Best regards,

Michael

 

Hello,
I am also trying to launch the driver for MRS6124R. But even after changing the protocol to Binary, I get the below error.

[ INFO] [1523273349.547360455]: Start initialising scanner ...
1523273349.5475 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273369.927893985]: no answer received after 20000 ms. Maybe sopas mode is wrong.
 
[ERROR] [1523273369.928035805]: sendSOPASCommand: no full reply available for read after 20000 ms
 
SOPAS - Error stopping streaming scan data!
[ WARN] [1523273369.928374863]: Disconnecting TCP-Connection.
sick_scan driver exiting.
1523273499.7376 ERROR: Tcp::open: Failed to open TCP connection to 192.168.0.24, aborting.
[ INFO] [1523273499.743618277]: Parameter setting for <active_echo: 0>
[ INFO] [1523273499.944271010]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
1523273499.9444 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273505.039736946]: no answer received after 5000 ms. Maybe sopas mode is wrong.
 
[ERROR] [1523273505.039904852]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1523273505.040206259]: Receiving: <STX><ETX>
[ERROR] [1523273505.040513862]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX>sMN SetAccessMode <ETX>rGD
 
[ INFO] [1523273505.041279125]: Sending  : sMN SetAccessMode 3 F4724744
1523273505.0414 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273510.135328661]: no answer received after 5000 ms. Maybe sopas mode is wrong.
 
[ERROR] [1523273510.135490299]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1523273510.135741709]: Receiving: <STX><ETX>
[ERROR] [1523273510.135886521]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN SetAccessMode 3 F4724744<ETX>
 
[ERROR] [1523273510.136251626]: Error Access Mode
[ INFO] [1523273510.336915862]: Sending  : sWN EIHstCola 1
1523273510.3372 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273515.431311557]: no answer received after 5000 ms. Maybe sopas mode is wrong.
 
[ERROR] [1523273515.431508571]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1523273515.431857316]: Receiving: <STX><ETX>
[ERROR] [1523273515.432196197]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sWN EIHstCola 1<ETX>
 
[ERROR] [1523273515.432697692]: Command or Error message not defined
[ERROR] [1523273515.432890800]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "sopas_protocol_type" to overwrite the default settings of the driver.
[ INFO] [1523273515.433070341]: Start initialising scanner ...
1523273515.4332 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1523273535.812481859]: no answer received after 20000 ms. Maybe sopas mode is wrong.
 
[ERROR] [1523273535.812639219]: sendSOPASCommand: no full reply available for read after 20000 ms
 
SOPAS - Error stopping streaming scan data!
[ WARN] [1523273535.812993309]: Disconnecting TCP-Connection.
sick_scan driver exiting.
1523273665.6256 ERROR: Tcp::open: Failed to open TCP connection to 192.168.0.24, aborting.

Is there anything else that I might be missing?

Thanks.

4eyedwit, 2018-04-10:
Hi Michael,

Thanks for your reply. Yes I used the most recent version of the devel branch. However I am unable to ping the ip address. I could do so with Windows OS but not with Ubuntu.

michael1309, 2018-04-10:
Hello,

if the scanner is not reachable via Ping, I recommend you to check the network settings. Under Ubuntu you can check the IP settings with the command ifconfig. The local IP address must be in the same namespace. You are using the IP address 192.168.0.24 for the scanner. Therefore your computer should have an address according to the pattern 192.168.0.xxx (e.g. 192.168.0.100). The subnet mask should be 255.255.255.0.  

You can set the IP address of the scanner in the corresponding launch file. The IP address set in the launch file and the IP address assigned via SOPAS-ET must match. The IP address set via SOPAS-ET must be stored in the scanner's eeprom. Therefore, the scanner must also be accessible from the Windows command line after closing SOPAS ET and repowering the scanner via ping. You must also disable DHCP for the used ethernet interface and switch to static ip addresses. As long as the scanner cannot be reached via ping, the ROS driver cannot run. I hope these hints will help you.

 

Best regards

Michael

closed 2018-04-18 by michael1309

dpelaez10, 2020-04-12:
> 
> 
> I recommend to reprogram the scanner to COLA-Binary-protocol permanently. You can do this via using SOPAS ET after login to "authorized client" level. After switching you should save the parameter to the internal memory of the scanner (EEPROM).

Hi Michael,

I have started using a TIM 781 for a project. I too have had similar issues and saw in the launch file that I should change to CoLa B protocol using SOPAS, however I am not sure how to do this within the program. The only area I am able to find such a selection is within the console when issuing the sensor commands. Any info on how to achieve this would be immensely appreciated.

Thank you for your work,
Daniel

michael1309, 2020-04-16:
@dpelaez10 
You can try it out relatively easily by switching the lidar off and on again (repowering). The lidar then starts with the protocol stored in the EEPROM. Usually this is SOPAS-A (ASCII protocol). After switching to the binary mode, the network connection has to be disconnected and re-established. The lidar then uses the binary mode. 

You can follow this very well in the console output.

naba4u, 2021-06-22:
Hi michael, i am new to sick mrs 1xxxx lidar, i was trying to connect mrs to ROS but not able to do. Even though i searched and tried since 1 month. could you please help me.  i changed ip, ip connections are successful every time, sopas mode changed from ASCII to binary. still facing issue.

ARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /sick_new_ip/hostname: 192.168.0.1
 * /sick_new_ip/new_IP_address: 192.168.0.69
 * /sick_new_ip/port: 2112
 * /sick_new_ip/scanner_type: sick_tim_5xx
 * /sick_new_ip/timelimit: 5
 * /sick_new_ip/use_binary_protocol: True

NODES
  /
    sick_new_ip (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://localhost:11311

process[sick_new_ip-1]: started with pid [9209]
[ INFO] [1624347649.074164889]: sick_generic_caller V. 1.10.1
[ INFO] [1624347649.074276037]: Program arguments: /home/nabin/ros_catkin_ws/install/lib/sick_scan/sick_generic_caller
[ INFO] [1624347649.074315570]: Program arguments: __name:=sick_new_ip
[ INFO] [1624347649.074351663]: Program arguments: __log:=/home/nabin/.ros/log/131f33f2-d328-11eb-bf27-c94be21df935/sick_new_ip-1.log
[ INFO] [1624347649.091443062]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1624347649.092064701]: Binary protocol activated
[ INFO] [1624347649.092171078]: Start initialising scanner [Ip: 192.168.0.1] [Port: 2112]
[ INFO] [1624347649.134657657]: Publishing laserscan-pointcloud2 to cloud
1624347779.6300 ERROR: Tcp::open: Failed to open TCP connection to 192.168.0.1, aborting.
[ INFO] [1624347779.631451449]: Parameter setting for <active_echo: 0>
[ INFO] [1624347779.832234863]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
1624347779.8323 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1624347787.262386274]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1624347787.262658235]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1624347787.262804049]: Receiving: <STX><ETX>
[ INFO] [1624347787.262857539]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>

[ INFO] [1624347787.262966948]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
1624347787.2630 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1624347794.772732995]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1624347794.772921603]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1624347794.773051517]: Receiving: <STX><ETX>
[ INFO] [1624347794.773097199]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN SetAccessMode 3 F4724744<ETX>

[ERROR] [1624347794.773154464]: Error Access Mode
[ INFO] [1624347794.773268209]: Sending  : <STX>sWN EIIpAddr C0 A8 00 45<ETX>
1624347794.7733 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1624347802.221615216]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1624347802.221930567]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1624347802.222095763]: Receiving: <STX><ETX>
[ INFO] [1624347802.222220652]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sWN EIIpAddr C0 A8 00 45<ETX>

[ INFO] [1624347802.222458271]: Sending  : <STX>sMN mEEwriteall<ETX>
1624347802.2226 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1624347809.682087047]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1624347809.682221876]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1624347809.682314457]: Receiving: <STX><ETX>
[ INFO] [1624347809.682363761]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN mEEwriteall<ETX>

[ INFO] [1624347809.682443111]: Sending  : <STX>sMN Run<ETX>
1624347809.6827 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1624347817.089753827]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1624347817.090862743]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1624347817.091576060]: Receiving: <STX><ETX>
[ INFO] [1624347817.092319557]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN Run<ETX>

[ INFO] [1624347817.093186843]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
1624347817.0939 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1624347824.697167783]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1624347824.697497015]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1624347824.697712163]: Receiving: <STX><ETX>
[ INFO] [1624347824.698001907]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN SetAccessMode 3 F4724744<ETX>

[ INFO] [1624347824.698230153]: Sending  : <STX>sMN mSCreboot<ETX>
1624347824.6983 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1624347832.202590684]: no answer received after 5000 ms. Maybe sopas mode is wrong.

please help me asap.

naba4u, 2021-06-22:
if i will run on this bellow command with a specific ip like

roslaunch sick_scan sick_mrs_1xxx.launch hostname:=192.168.0.117

then i am getting this.
ARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /sick_mrs_1xxx/cloud_topic: cloud
 * /sick_mrs_1xxx/frame_id: cloud
 * /sick_mrs_1xxx/hostname: 192.168.0.117
 * /sick_mrs_1xxx/intensity_resolution_16bit: False
 * /sick_mrs_1xxx/max_ang: 2.3998277
 * /sick_mrs_1xxx/min_ang: -2.3998277
 * /sick_mrs_1xxx/port: 2112
 * /sick_mrs_1xxx/scanner_type: sick_mrs_1xxx
 * /sick_mrs_1xxx/sw_pll_only_publish: True
 * /sick_mrs_1xxx/timelimit: 5

NODES
  /
    sick_mrs_1xxx (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://localhost:11311

process[sick_mrs_1xxx-1]: started with pid [10049]
[ INFO] [1624350855.206909783]: sick_generic_caller V. 1.10.1
[ INFO] [1624350855.207076128]: Program arguments: /home/nabin/ros_catkin_ws/install/lib/sick_scan/sick_generic_caller
[ INFO] [1624350855.207145973]: Program arguments: __name:=sick_mrs_1xxx
[ INFO] [1624350855.207210425]: Program arguments: __log:=/home/nabin/.ros/log/8a2093fe-d334-11eb-bf27-c94be21df935/sick_mrs_1xxx-1.log
[ INFO] [1624350855.232881510]: Start initialising scanner [Ip: 192.168.0.117] [Port: 2112]
[ INFO] [1624350855.288203805]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1624350855.297626394]: Parameter setting for <active_echo: 0>
[ INFO] [1624350855.499235463]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1624350855.505474696]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1624350855.707877874]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1624350855.715127686]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1624350855.917108070]: Sending  : <STX><STX><STX><STX><Len=0015>sMN LMCstopmeas CRC:<0x10>
[ INFO] [1624350855.925128582]: Receiving: <STX>sAN LMCstopmeas \x00<ETX>
[ INFO] [1624350856.129623078]: Sending  : <STX><STX><STX><STX><Len=0033>sWN SetActiveApplications 0x00 0x01 0x46 0x45 0x56 0x4c 0x00 CRC:<0x35>
[ INFO] [1624350856.131651735]: Receiving: <STX>sWA SetActiveApplications <ETX>
[ INFO] [1624350856.333114150]: Sending  : <STX><STX><STX><STX><Len=0033>sWN SetActiveApplications 0x00 0x01 0x52 0x41 0x4e 0x47 0x01 CRC:<0x37>
[ INFO] [1624350856.336987774]: Receiving: <STX>sWA SetActiveApplications <ETX>
[ INFO] [1624350856.538235455]: Sending  : <STX><STX><STX><STX><Len=0015>sRN DeviceIdent CRC:<0x25>
[ INFO] [1624350856.547123402]: Receiving: <STX>sRA DeviceIdent \x00\x08\x4d\x52\x53\x31\x78\x78\x78\x43\x00\x08\x32\x2e\x31\x2e\x30\x2e\x30\x52<ETX>
[ INFO] [1624350856.548345836]: Deviceinfo MRS1xxxC V2.1.0.0R found and supported by this driver.
[ INFO] [1624350856.749635393]: Sending  : <STX><STX><STX><STX><Len=0016>sRN SerialNumber CRC:<0x4c>
[ INFO] [1624350856.756726254]: Receiving: <STX>sRA SerialNumber \x00\x08\x32\x30\x31\x31\x30\x30\x30\x35<ETX>
[ INFO] [1624350856.959051058]: Sending  : <STX><STX><STX><STX><Len=0019>sRN FirmwareVersion CRC:<0x24>
[ INFO] [1624350856.961077423]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x20\x32\x2e\x31\x2e\x30\x2e\x30<ETX>
[ INFO] [1624350857.161797363]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1624350857.163072624]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1624350857.363592796]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODoprh CRC:<0x41>
[ INFO] [1624350857.365801280]: Receiving: <STX>sRA ODoprh \x00\x00\x01\x57<ETX>
[ INFO] [1624350857.569423431]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODpwrc CRC:<0x52>
[ INFO] [1624350857.578081587]: Receiving: <STX>sRA ODpwrc \x00\x00\x00\x2b<ETX>
[ INFO] [1624350857.781226174]: Sending  : <STX><STX><STX><STX><Len=0016>sRN LocationName CRC:<0x55>
[ INFO] [1624350857.783732270]: Receiving: <STX>sRA LocationName \x00\x0b\x6e\x6f\x74\x20\x64\x65\x66\x69\x6e\x65\x64<ETX>
[ INFO] [1624350857.786700928]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1624350857.789005493]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xf8\xc0\x88\x00\x22\xb6\xb8<ETX>
[ INFO] [1624350857.790010769]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1624350857.790770888]: [From:To] -47.50000 [deg] to 227.50000 [deg] (in 1/10000th deg: from 0xFFF8C088 to 0x22B6B8)
[ INFO] [1624350857.791416893]: MIN_ANG:   -2.400 [rad] -137.500 [deg]
[ INFO] [1624350857.795777867]: MAX_ANG:    2.400 [rad]  137.500 [deg]
[ INFO] [1624350857.796641431]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x09 0xc4 0xff 0xf8 0xc0 0x88 0x00 0x22 0xb6 0xb8 CRC:<0xd4>
[ INFO] [1624350857.798778554]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1624350857.799672667]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1624350857.806698250]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xf8\xc0\x88\x00\x22\xb6\xb8<ETX>
[ INFO] [1624350857.814777924]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1624350857.818869353]: MIN_ANG (after command verification):   -2.400 [rad] -137.500 [deg]
[... 57 lines cut ...]
[ INFO] [1624350870.097132173]: 5 / 6 Packet dropped Software PLL not yet locked.
[ INFO] [1624350870.115359406]: 6 / 6 Packet dropped Software PLL not yet locked.
[ INFO] [1624350870.116083913]: Software PLL is expected to be ready now!
[... cut ...]
[ INFO] [1624350964.694441283]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1624350964.694647417]: Receiving: <STX><ETX>
[ INFO] [1624350964.695398924]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN SetAccessMode 3 F4724744<ETX>

[ERROR] [1624350964.696201240]: Error Access Mode
[ INFO] [1624350964.897321434]: Sending  : <STX>sWN EIHstCola 1<ETX>
1624350964.8976 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1624350972.189371019]: no answer received after 5000 ms. Maybe sopas mode is wrong.

michael1309, 2021-06-24:
Hello,

basically everything looks very good. After the startup sequence runs through, Pointcloud2 messages are sent and the log output is muted. We recommend you to read the introductory guide. 

For startup with rviz etc. you can refer to e.g. the ticket https://github.com/SICKAG/sick_scan/issues/122 for help. 

Good luck!

naba4u, 2021-06-24:
Thank you michael i solved... thanks for your help. i Really love it.