SICKAG/sick_scan #7: PointCloud2 data has to much information

SponTanious, 2018-04-11:
Firstly I want  to say thank you for your work. I have been using your repository with the MRS6000 SICK LiDAR with no issues for the past few weeks. 

My issue arose when I wanted to transform the LiDAR data from my laser frame to my map frame it was taking approximately 0.4 seconds on my system. I noticed that when I used ROS's readPoints function located at */ros/lunar/lib/python2.7/dist-packages/sensor_msgs* that it returned 88704 points instead of 22176 points. After further investigation I found that the height listed on the PointCloud2 message is 96 not 24, 4 times larger which results in 4 times as many points as expected.

Is the PointCloud2 message returning the same set of points 4 times? Or does each set of points represent something different, e.g. echos?

I created a work around where I created my own readPoints function where i changed
```python
for v in range(height):
```
to
```python
for v in range(0,height,4):
```

Will this effect the quality of the data I am receiving?

michael1309, 2018-04-11:
Hi Daniel,

Great to hear that you were able to successfully use the driver. Your observation is correct. The PointCloud2 message follows this logic:

1. 4 destinations are sent per space direction in the default setting. The data is nested as follows:

- Top level: Index of the respective target[0...3]
- Middle level: Line of the respective scan s[0...23]
- Lowest level: Column of the respective scan[0...923]

If you now want to read the nth target at position x,y, the address is as follows:
Address in PointCloud2 field: n * 24 * 924 + y * 924 + x

2. Using the entry filter_echos in the parameter file you can switch from 4 targets per direction to 1 target per direction according to the following logic:

- 0:  First echo
- 1 : All echoes (4 shots per direction)
- 2: Last echo

Please change the value to 0 or 2 and follow the output of the FREchoFilter command when starting.
Many greetings
Michael

SponTanious, 2018-04-11:
Thanks for your help Michael that definitely clears things up.

Daniel

closed 2018-04-11 by SponTanious