SICKAG/sick_scan #10: Tim561-20590101 wtih "use_binary_protocol" : topic "/scan" has a empty frame_id, but also both of angle_min and angle_max with a wrong value

idealkr, 2018-06-20:
Equipment: Tim561-20590101 
Protocol: "use_binary_protocol"
Launch file: "sick_tim_5xx.launch"

michael1309, 2018-06-22:
Dear „idealkr“,

 

Thanks for you hints. We will check your remarks and will try to fix these minor issues in the next couple of days.

 

Best regards,

Michael

michael1309, 2018-06-26:
Dear Tony,

we fixed theses issues now. Here is a dump from "rostopic echo /scan".

header: 
  seq: 1076
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: laser
angle_min: -0.994837701321
angle_max: 0.994837582111
angle_increment: 0.0174532923847
time_increment: 0.000185185184819
scan_time: 0.0666666701436
range_min: 11.1000003815
range_max: 101.099998474
ranges: [1.3819999694824219, ...

As you can see all the necessary parameters are now included.
Also we tested the following scanner types: tim551, tim561, tim571.

All changes are pushed to the "devel"-branch.

Best regards,
Michael

closed 2018-06-26 by michael1309