SICKAG/sick_scan #9: Tim561-20500101: "The time_increment, scan_time and angle_increment values reported by the scanner are inconsistent! "

idealkr, 2018-06-20:
[ WARN] [1529475586.849910849]: The time_increment, scan_time and angle_increment values reported by the scanner are inconsistent! Expected time_increment: 0.000185185, reported time_increment: 0.000061728. Perhaps you should set the parameter time_increment to the expected value. This message will print every 60 seconds.

Tim561-20500101 running with  "sick_tim_5xx.launch", it's turn out to be a 1° laser_sacn msg returned, wasn't the expected 0.33° laser_sacn msg.

this [WARN]  didn't dispear， until you added "<param name="time_increment" type="double" value="0.000185185" />" to launch file.

michael1309, 2018-06-25:
Dear Idealkr,
we checked your issues. You are right. The code contains inconsistencies in the details concerned. We are in the correction process and will also take over the checks into the test bed in this context. As mentioned, we expect the bugs to be fixed by the end of the week.
Thank you for your comments!

closed 2018-06-26 by michael1309

QGBSA, 2021-10-21:
I experience the same problem, have you rectified the bug?