SICKAG/sick_scan #11: Error unexpected Sopas answer for request sWN SetActiveApplications

chrisfernam2, 2018-06-29:
Hi everyone i just received the mrs1000 LiDAR and i'm having some problems. I'm running the launch file by entering: roslaunch sick_scan sick_mrs_1xxx.launch but i'm facing quite a bit of errors. Could you please help me?

Here is the behavior within my terminal:  

ros-vm@ROS-vm:~/catkin_ws/src/sick_scan/launch$ roslaunch sick_scan sick_mrs_1xxx.launch 
... logging to /home/ros-vm/.ros/log/c477f7dc-7bbb-11e8-beb2-0800274a3ab6/roslaunch-ROS-vm-2325.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro.py is deprecated; please use xacro instead
started roslaunch server http://localhost:39475/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.2
 * /sick_mrs_1xxx/hostname: 192.168.0.1
 * /sick_mrs_1xxx/intensity_resolution_16bit: False
 * /sick_mrs_1xxx/max_ang: 2.3998277
 * /sick_mrs_1xxx/min_ang: -2.3998277
 * /sick_mrs_1xxx/port: 2112
 * /sick_mrs_1xxx/scanner_type: sick_mrs_1xxx
 * /sick_mrs_1xxx/timelimit: 5

NODES
  /
    sick_mrs_1xxx (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://localhost:11311

process[sick_mrs_1xxx-1]: started with pid [2343]
[ INFO] [1530290727.003895157]: sick_generic_caller V. 001.002.001
[ INFO] [1530290727.003985764]: Program arguments: /home/ros-vm/catkin_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1530290727.004012680]: Program arguments: __name:=sick_mrs_1xxx
[ INFO] [1530290727.004029852]: Program arguments: __log:=/home/ros-vm/.ros/log/c477f7dc-7bbb-11e8-beb2-0800274a3ab6/sick_mrs_1xxx-1.log
[ INFO] [1530290727.016150117]: Start initialising scanner ...
[ INFO] [1530290727.043051528]: Parameter setting for <active_echo: 0>
[ INFO] [1530290727.243853840]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
[ INFO] [1530290727.255578714]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1530290727.458298011]: Sending  : sWN EIHstCola \x01
[ INFO] [1530290727.469194857]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1530290727.670414146]: Sending  : sMN LMCstopmeas
[ INFO] [1530290727.703497948]: Receiving: <STX>sAN LMCstopmeas \x00<ETX>
[ INFO] [1530290727.904011287]: Sending  : sWN SetActiveApplications \x01\x20\x46\x45\x56\x4c\x20\x00
[ INFO] [1530290727.916623526]: Receiving: <STX>sFA\x00\x45<ETX>
[ERROR] [1530290727.916850355]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX>"sWN SetActiveApplications  FEVL 5Answer= >>>sFA\x00\x45<<<

[ERROR] [1530290727.917062333]: Command or Error message not defined
[ INFO] [1530290728.118791900]: Sending  : sWN SetActiveApplications \x01\x20\x52\x41\x4e\x47\x20\x01
[ INFO] [1530290728.131115200]: Receiving: <STX>sFA\x00\x45<ETX>
[ERROR] [1530290728.131560221]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX>"sWN SetActiveApplications  RANG 7Answer= >>>sFA\x00\x45<<<

[ERROR] [1530290728.131687105]: Command or Error message not defined
[ INFO] [1530290728.332318527]: Sending  : sRI 0
[ INFO] [1530290728.343563829]: Receiving: <STX>sFA\x00\x03<ETX>
[ERROR] [1530290728.343697775]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX>sRI 0xAnswer= >>>sFA\x00\x03<<<

[ERROR] [1530290728.343803174]: Error reading device ident
[ INFO] [1530290728.544384533]: Sending  : sRN SerialNumber
[ INFO] [1530290728.554818372]: Receiving: <STX>sRA SerialNumber \x00\x08\x31\x37\x34\x35\x30\x30\x32\x33<ETX>
[ WARN] [1530290728.554976773]: Device ??? V-1.-1 found and maybe unsupported by this driver.
[ WARN] [1530290728.555022895]: Full SOPAS answer: Command or Error message not defined
[ INFO] [1530290728.756571314]: Sending  : sRN FirmwareVersion
[ INFO] [1530290728.767168275]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x20\x31\x2e\x33\x2e\x31\x2e\x30<ETX>
[ INFO] [1530290728.967667563]: Sending  : sRN SCdevicestate
[ INFO] [1530290728.978611015]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1530290729.179266167]: Sending  : sRN ODoprh
[ INFO] [1530290729.189693093]: Receiving: <STX>sRA ODoprh \x00\x00\x00\xf8<ETX>
[ INFO] [1530290729.391517477]: Sending  : sRN ODpwrc
[ INFO] [1530290729.401865706]: Receiving: <STX>sRA ODpwrc \x00\x00\x00\x02<ETX>
[ INFO] [1530290729.604090387]: Sending  : sRN LocationName
[ INFO] [1530290729.615077248]: Receiving: <STX>sRA LocationName \x00\x0b\x53\x4e\x20\x31\x37\x34\x35\x30\x30\x32\x33<ETX>
[sick_mrs_1xxx-1] process has died [pid 2343, exit code -11, cmd /home/ros-vm/catkin_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_mrs_1xxx __log:=/home/ros-vm/.ros/log/c477f7dc-7bbb-11e8-beb2-0800274a3ab6/sick_mrs_1xxx-1.log].
log file: /home/ros-vm/.ros/log/c477f7dc-7bbb-11e8-beb2-0800274a3ab6/sick_mrs_1xxx-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
ros-vm@ROS-vm:~/catkin_ws/src/sick_scan/launch$

michael1309, 2018-06-30:
Hi, we will check this issue next week and give you a detailed feedback. Best regards, Michael

[ INFO] [1530290727.255578714]: Receiving: sAN SetAccessMode \x01
[ INFO] [1530290727.469194857]: Receiving: sWA EIHstCola 
[ INFO] [1530290727.703497948]: Receiving: sAN LMCstopmeas \x00
[ INFO] [1530290727.916623526]: Receiving: sFA\x00\x45
[ERROR] [1530290727.916850355]: Error Sopas answer mismatch Error unexpected Sopas Answer for request "sWN SetActiveApplications FEVL 5Answer= >>>sFA\x00\x45<<<

[ INFO] [1530290728.131115200]: Receiving: sFA\x00\x45
[ERROR] [1530290728.131560221]: Error Sopas answer mismatch Error unexpected Sopas Answer for request "sWN SetActiveApplications RANG 7Answer= >>>sFA\x00\x45<<<

[ INFO] [1530290728.343563829]: Receiving: sFA\x00\x03
[ERROR] [1530290728.343697775]: Error Sopas answer mismatch Error unexpected Sopas Answer for request sRI 0xAnswer= >>>sFA\x00\x03<<<

[ INFO] [1530290728.554818372]: Receiving: sRA SerialNumber \x00\x08\x31\x37\x34\x35\x30\x30\x32\x33
[ INFO] [1530290728.767168275]: Receiving: sRA FirmwareVersion \x00\x09\x56\x20\x31\x2e\x33\x2e\x31\x2e\x30
[ INFO] [1530290728.978611015]: Receiving: sRA SCdevicestate \x00
[ INFO] [1530290729.189693093]: Receiving: sRA ODoprh \x00\x00\x00\xf8
[ INFO] [1530290729.401865706]: Receiving: sRA ODpwrc \x00\x00\x00\x02
[ INFO] [1530290729.615077248]: Receiving: sRA LocationName \x00\x0b\x53\x4e\x20\x31\x37\x34\x35\x30\x30\x32\x33

michael1309, 2018-07-01:
Hi Chris,

in the first step you can ignore the warnings. Some scanner types does not support the full range of commands. Can you receive data from the MRS1xxx?

Best regards,
Michael

michael1309, 2018-07-01:
The MRS1xxx does not support some legacy commands. So I changed the command processing similar to the MRS6124.  The output starting with V.001.002.002 is like this:

[ INFO] [1530430494.072798301]: sick_generic_caller V. 001.002.002
[ INFO] [1530430494.072857178]: Program arguments: /home/michael/catkin_sickag_devel_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1530430494.072875282]: Program arguments: __name:=sick_mrs_1xxx
[ INFO] [1530430494.118973641]: Start initialising scanner ...
[ INFO] [1530430494.521775846]: Parameter setting for <active_echo: 0>
[ INFO] [1530430494.722729142]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
[ INFO] [1530430494.746603450]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1530430494.948155898]: Sending  : sWN EIHstCola \x01
[ INFO] [1530430494.959109516]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1530430495.160597830]: Sending  : sMN LMCstopmeas
[ INFO] [1530430495.192428888]: Receiving: <STX>sAN LMCstopmeas \x00<ETX>
[ INFO] [1530430495.393304351]: Sending  : sRN DeviceIdent
[ INFO] [1530430495.403826492]: Receiving: <STX>sRA DeviceIdent \x00\x08\x4d\x52\x53\x31\x78\x78\x78\x78\x00\x08\x31\x2e\x33\x2e\x30\x2e\x30\x52<ETX>
[ INFO] [1530430495.404019258]: Deviceinfo MRS1xxxx V1.3.0.0R found and supported by this driver.
[ INFO] [1530430495.604321816]: Sending  : sRN FirmwareVersion
[ INFO] [1530430495.615344630]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x20\x31\x2e\x33\x2e\x30\x2e\x30<ETX>
[ INFO] [1530430495.815703185]: Sending  : sRN SCdevicestate
[ INFO] [1530430495.826633990]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1530430496.027230813]: Sending  : sRN ODoprh
[ INFO] [1530430496.038068903]: Receiving: <STX>sRA ODoprh \x00\x00\x02\x5c<ETX>
[ INFO] [1530430496.240775305]: Sending  : sRN ODpwrc
[ INFO] [1530430496.252793971]: Receiving: <STX>sRA ODpwrc \x00\x00\x00\x0f<ETX>
[ INFO] [1530430496.457036391]: Sending  : sRN LocationName
[ INFO] [1530430496.468266243]: Receiving: <STX>sRA LocationName \x00\x0b\x53\x4e\x20\x31\x37\x34\x35\x30\x30\x31\x35<ETX>
[ INFO] [1530430496.470928561]: Sending  : sRN LMPoutputRange
[ INFO] [1530430496.481915251]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xf8\xc0\x88\x00\x22\xb6\xb8<ETX>
[ INFO] [1530430496.482063160]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1530430496.482101973]: MIN_ANG:   -2.400 [rad] -137.500 [deg]
[ INFO] [1530430496.482122579]: MAX_ANG:    2.400 [rad]  137.500 [deg]
[ INFO] [1530430496.482188355]: Sending  : sWN LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xf8\xc0\x88\x00\x22\xb6\xb8
[ INFO] [1530430496.492883848]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1530430496.493029638]: Sending  : sRN LMPoutputRange
[ INFO] [1530430496.503827584]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xf8\xc0\x88\x00\x22\xb6\xb8<ETX>
[ INFO] [1530430496.503972204]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1530430496.506923288]: MIN_ANG (after command verification):   -2.400 [rad] -137.500 [deg]
[ INFO] [1530430496.507007984]: MAX_ANG (after command verification):    2.400 [rad]  137.500 [deg]
[ INFO] [1530430496.507086222]: MRS 1xxx detected overwriting resolution flag (only 8 bit supported)
[ INFO] [1530430496.507176276]: Sending  : sWN LMDscandatacfg \x07\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01
[ INFO] [1530430496.517809197]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1530430496.517957464]: Sending  : sRN LMDscandatacfg
[ INFO] [1530430496.528736597]: Receiving: <STX>sRA LMDscandatacfg \x07\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01<ETX>
[ INFO] [1530430496.530129982]: Sending  : sWN FREchoFilter \x01
[ INFO] [1530430496.540727421]: Receiving: <STX>sWA FREchoFilter<ETX>
[ INFO] [1530430496.540897186]: Sending  : sMN LMCstartmeas
[ INFO] [1530430496.562667284]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1530430496.562823184]: Sending  : sMN Run
[ INFO] [1530430496.573634045]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1530430496.573802048]: Sending  : sRN SCdevicestate
[ INFO] [1530430496.584610981]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1530430496.584763341]: Waiting for scanner ready state since 0 secs
[ INFO] [1530430497.585955317]: Sending  : sRN SCdevicestate
[ INFO] [1530430497.596889617]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1530430497.597039747]: Waiting for scanner ready state since 1 secs
.... <continued> ...

michael1309, 2018-07-01:
I will push this version at latest tomorrow morning (2nd of July) after testing.

michael1309, 2018-07-01:
Hi all,

I just pushed a new version in the devel branch, which should solve the mentioned issue.

Best regards,
Michael

closed 2018-07-01 by michael1309

chrisfernam2, 2018-07-02:
This is my first product with SICKAG and i'm not fully experienced using
this. I downloaded the sick_scan directory and right now i'm facing
different issues:

1st)  Its failing to open TCP connection.
2nd) I cant even connect to ET tool even though i was able to connect 3
days ago, i custom search for LiDAR by giving ip 192.168.0.1 but its unable
to establish connection.

3rd) If the sick_generic_caller is supposed to be V.001.002.002, why i'm i
seeing 001.001.001?

michael1309, 2018-07-02:
Did you choose the devel branch? Did you build your app. via catkin_make? Can you ping your device?

chrisfernam2, 2018-07-02:
I did build the app via catkin_make, and yes i am able to ping the device
successfully.

However, how can i choose the devel branch? I have been learning ROS for
the past month so some things are new to me.

Thanks for the help by the way!

michael1309, 2018-07-02:
You can choose the devel branch in the drop down menu on the github page of
the project.

michael1309, 2018-07-02:
Here you can see the branches.

Michael Lehning <michael.lehning@googlemail.com> schrieb am Mo., 2. Juli
2018, 21:33:

michael1309, 2018-07-02:
You can you also the command line:

git checkout devel

Best regards, michael

chrisfernam2, 2018-07-02:
Ok i just downloaded the devel version and remade my file. Although it now  shows sick_generic_caller V.001.002.002, and most errors are gone, i still getting the same final error causing a disconnection on TCP-connection, here is the full log :

michael1309, 2018-07-02:
Command handling is ok.
But it seems that the scanner does not send the datagram. Can you receive data via sopas et? Could you send me the firmware version? I will discuss this issue with SICK.
-- 
Ing.-Büro Dr. Michael Lehning 
Schulstraße. 21B
D-31137 Hildesheim 
Germany

E: Michael.lehning@lehning.de
T: +49 5121 645-70
M: +49 173 37 69 242

chrisfernam2, 2018-07-02:
Actually i think it's working now. The final terminal log info printed was:

INFO Receiving:<STX>sEA LMDscandata \x01<ETX>

It hasn't given me the error anymore. To verify, entered rostopic list and
i get the following topics:

/cloud
/diagnostics
/radar
/rosout
/rosout_agg
/scan
/sick_mrs_1xxx/parameter_descriptions
/sick_mrs_1xxx/parameter_updates

I "rostopic echo /cloud" and "rostopic echo /scan" in the terminal and i
can confirm i'm receiving values of some sort.

When i "roslaunch sick_scan sick_mrs_1xxx.launch" , isn't this supposed to
launch rviz for me automatically?
If not how may i see the scanned point clouds in rviz; that is, what
command?

roslaunch sick_scan sick_mrs_1xxx.launch
rosrun rviz rviz
publish to point cloud         (how can i do this?) and see point cloud
data on rviz?

I appreciate your patience and support.

michael1309, 2018-07-02:
Sounds great. Just start rviz and follow the hints I gave in the ticket for
the mrs6124.

michael1309, 2018-07-02:
Rviz: change the fixed frame under global options to "laser" instead of "map".
-- 
Ing.-Büro Dr. Michael Lehning 
Schulstraße. 21B
D-31137 Hildesheim 
Germany

E: Michael.lehning@lehning.de
T: +49 5121 645-70
M: +49 173 37 69 242

chrisfernam2, 2018-07-02:
I opened rviz and chose to add a display (by topic), i added /cloud   PointCloud2.  I specified the topic to be /cloud, but almost instantaneously i get:

 Status: error Transform[sender=unknown_publisher]         For frame[laser]:Fixed Frame[map] does not exist.

It should be pretty much plug & play from this point onward, what does this error mean?

chrisfernam2, 2018-07-03:
Thank you the hint is very helpful for me for differentiating between each layer!

chrisfernam2, 2018-07-05:
I'm still having trouble with adding topic LaserScan to rviz, i get error Transform[sender=unkown_publisher]
Why does this happen?

chrisfernam2, 2018-07-05:
I'm still receiving the same "Transform[sender=unknown_publisher]" error ; that is, 
For frame[laser_+250_DIST3]:Fixed Frame[laser_+250_DIST1] does not exist.

How can i fix this?

michael1309, 2018-07-05:
You can use the command tf static transform as a node in the launch file.

michael1309, 2018-07-05:
One remark: We recommend to use pointcloud2 instead of laserscan.

Michael Lehning <michael.lehning@googlemail.com> schrieb am Do., 5. Juli
2018, 23:37:

michael1309, 2018-07-06:
Hi,

for a better understanding of the ideas and concepts I created today a test launch file and opened a ticket about this issue (Ticket #13).
It explains in detail and hands on how you can use pointcloud2 and laserscan (including rviz config file and screenshot)

Best regards,
Michael

anupritdangi, 2018-07-24:
Getting the same error @chrisfernam2 . Nothing is published on scan topic. How you removed this error. can you please help?

LOG->
-------------------------
    <param name="scanner_type" type="string" value="sick_mrs_1xxx"/>
    <param name="min_ang" type="double" value="-2.3998277" />
    <param name="max_ang" type="double" value="+2.3998277" />
    <param name="intensity_resolution_16bit" type="bool" value="False" />
    <param name="hostname" type="string" value="169.254.100.120" />
    <param name="port" type="string" value="2112" />
    <param name="timelimit" type="int" value="5" />
    <param name="subscribe_datagram" type="bool" value="false" />
---------------------------

ROS_MASTER_URI=http://localhost:11311

process[sick_mrs_1xxx-1]: started with pid [18897]
[ INFO] [1532422730.910066728]: sick_generic_caller V. 001.002.004
[ INFO] [1532422730.910277132]: Program arguments: /home/nvidia/catkin_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1532422730.910344429]: Program arguments: __name:=sick_mrs_1xxx
[ INFO] [1532422730.910428462]: Program arguments: __log:=/home/nvidia/.ros/log/e12895f0-8f1e-11e8-9e14-00044ba7df0f/sick_mrs_1xxx-1.log
[ INFO] [1532422730.958080648]: Start initialising scanner ...
[ INFO] [1532422731.257183790]: Parameter setting for <active_echo: 0>
[ INFO] [1532422731.461172324]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
[ INFO] [1532422731.473282717]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1532422731.675159250]: Sending  : sWN EIHstCola \x01
[ INFO] [1532422731.686425759]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1532422731.888174418]: Sending  : sMN LMCstopmeas
[ INFO] [1532422731.920882663]: Receiving: <STX>sAN LMCstopmeas \x00<ETX>
[ INFO] [1532422732.123843116]: Sending  : sRN DeviceIdent
[ INFO] [1532422732.136092712]: Receiving: <STX>sRA DeviceIdent \x00\x08\x4d\x52\x53\x31\x78\x78\x78\x78\x00\x08\x31\x2e\x33\x2e\x30\x2e\x30\x52<ETX>
[ INFO] [1532422732.138111015]: Deviceinfo MRS1xxxx V1.3.0.0R found and supported by this driver.
[ INFO] [1532422732.340192639]: Sending  : sRN FirmwareVersion
[ INFO] [1532422732.352384154]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x20\x31\x2e\x33\x2e\x30\x2e\x30<ETX>
[ INFO] [1532422732.554089804]: Sending  : sRN SCdevicestate
[ INFO] [1532422732.565260695]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1532422732.768058778]: Sending  : sRN ODoprh
[ INFO] [1532422732.780229205]: Receiving: <STX>sRA ODoprh \x00\x00\x00\xea<ETX>
[ INFO] [1532422732.994985391]: Sending  : sRN ODpwrc
[ INFO] [1532422733.007086569]: Receiving: <STX>sRA ODpwrc \x00\x00\x00\x27<ETX>
[ INFO] [1532422733.222065350]: Sending  : sRN LocationName
[ INFO] [1532422733.234544613]: Receiving: <STX>sRA LocationName \x00\x0b\x53\x4e\x20\x31\x37\x34\x37\x30\x30\x32\x33<ETX>
[ INFO] [1532422733.244302907]: Sending  : sRN LMPoutputRange
[ INFO] [1532422733.255426277]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xf8\xc0\x88\x00\x22\xb6\xb8<ETX>
[ INFO] [1532422733.256589623]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1532422733.257103135]: MIN_ANG:   -2.400 [rad] -137.500 [deg]
[ INFO] [1532422733.257324802]: MAX_ANG:    2.400 [rad]  137.500 [deg]
[ INFO] [1532422733.258225360]: Sending  : sWN LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xf8\xc0\x88\x00\x22\xb6\xb8
[ INFO] [1532422733.269740481]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1532422733.270823505]: Sending  : sRN LMPoutputRange
[ INFO] [1532422733.281993468]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xf8\xc0\x88\x00\x22\xb6\xb8<ETX>
[ INFO] [1532422733.282864362]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1532422733.314694065]: MIN_ANG (after command verification):   -2.400 [rad] -137.500 [deg]
[ INFO] [1532422733.315348443]: MAX_ANG (after command verification):    2.400 [rad]  137.500 [deg]
[ INFO] [1532422733.315781058]: MRS 1xxx detected overwriting resolution flag (only 8 bit supported)
[ INFO] [1532422733.317125399]: Sending  : sWN LMDscandatacfg \x07\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01
[ INFO] [1532422733.328296098]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1532422733.329294929]: Sending  : sRN LMDscandatacfg
[ INFO] [1532422733.340812898]: Receiving: <STX>sRA LMDscandatacfg \x07\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01<ETX>
[ INFO] [1532422733.355271359]: Sending  : sWN FREchoFilter \x01
[ INFO] [1532422733.358047370]: Receiving: <STX>sWA FREchoFilter<ETX>
[ INFO] [1532422733.360535440]: Sending  : sMN LMCstartmeas
[ INFO] [1532422733.383033769]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1532422733.384965446]: Sending  : sMN Run
[ INFO] [1532422733.397572391]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1532422733.399289154]: Sending  : sRN SCdevicestate
[ INFO] [1532422733.411037462]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1532422733.415876384]: Waiting for scanner ready state since 0 secs
[ INFO] [1532422734.417671061]: Sending  : sRN SCdevicestate
[ INFO] [1532422734.428398425]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[... 8 lines cut ...]
[ INFO] [1532422737.473227254]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1532422737.474270022]: Waiting for scanner ready state since 4 secs
[ INFO] [1532422738.475584373]: Sending  : sRN SCdevicestate
[ INFO] [1532422738.486512892]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[... cut ...]
[ INFO] [1532422743.553504937]: Sending  : sRN SCdevicestate
[ INFO] [1532422743.565851494]: Receiving: <STX>sRA SCdevicestate \x01<ETX>
[ INFO] [1532422743.566976919]: Scanner ready for measurement after 10 [sec]
[ INFO] [1532422743.567882693]: Sending  : sEN LMDscandata \x01
[ INFO] [1532422743.580065344]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ WARN] [1532422763.584522966]: Timeout during waiting for new datagram
[ERROR] [1532422763.585447012]: Read Error when getting datagram: 1.
[ INFO] [1532422763.586557845]: Start initialising scanner ...

michael1309, 2018-07-24:
@anupritdangi 
Which MRS1xxx scanner do you use exactly? For multilayer scanners the driver does not publish scan messages but only PointCloud2 messages in topic /cloud. The individual layer of the scans are not in the same coordinate frame, so the PointCloud2 representation is much more practical. 

For the visualization in rviz you should select "laser" as fixed frame or publish a tf transformation "laser"-->"map".

Best regards, 
Michael

chrisfernam2, 2018-07-24:
Are you sure it does not publish LaserScan messages? Mine is. Mine is setup
with a tirtlebot and is currently working fine.

michael1309, 2018-07-25:
Hello,

this was a little mix-up in our internal company coordination. You are correct. Two types of messages are published for the MRS1xxx.

/cloud as Pointcloud2 message.
/scan as laser scanner messages. Each level and echotype has its own name.

For more than 4 layers (i.e. MRS6xxx), only the Pointcloud2 message is published.

michael1309, 2018-07-25:
One additional hint: By comparing the timestamps you can see the "old" values for the timeout in the timestamp difference It is round about 20 sec.
 INFO] [1532422743.580065344]: Receiving: sEA LMDscandata \x01
[ WARN] [1532422763.584522966]: Timeout during waiting for new datagram

anupritdangi, 2018-07-25:
Everything is working fine now.I changed nothing just restarted the scanner 2-3 times.
Thanks

michael1309, 2018-07-26:
@anupritdangi 

Great to hear. 

So you have left the timeout time at the set value of 20 seconds?

If you can derive any rules when the scanner does not send datagrams, although the initialization sequence was successfully run before, we would be pleased if you could report this briefly.