SICKAG/sick_scan #14: Unable to connect to remote host: Connection refused

THAC00, 2018-07-18:
Hello,

I'm having trouble getting the driver started
I'm using the MRS6214 and when I try the startup sequence:

source ~/ros_catkin_ws/install/setup.bash
roslaunch sick_scan sick_mrs_6xxx.launch

I get a bunch of errors related to not being able to connect to the scanner.
The scanner works in windows/sopas and is set to binary.
The static IP Address is 169.254.10.247 and is pingable and set in the launchfile in the install folder.

When I use telnet with the port 2112 ( or any port ) I get this error message:
Unable to connect to remote host: Connection refused

I Would really appreciate some help and thank you for the hard work.
Kind regards.

Here is the Launch Error Log:

_________________________________________________________________________________

... logging to /home/mert/.ros/log/32faa446-8a8c-11e8-b505-a402b94d4cfc/roslaunch-mert-hp-4008.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mert-hp:43115/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.2
 * /sick_mrs_6xxx/filter_echos: 0
 * /sick_mrs_6xxx/hostname: 169.254.10.247
 * /sick_mrs_6xxx/max_ang: 1.047197333
 * /sick_mrs_6xxx/min_ang: -1.047197333
 * /sick_mrs_6xxx/port: 2112
 * /sick_mrs_6xxx/range_max: 250.0
 * /sick_mrs_6xxx/range_min: 0.1
 * /sick_mrs_6xxx/scanner_type: sick_mrs_6xxx
 * /sick_mrs_6xxx/timelimit: 5

NODES
  /
    sick_mrs_6xxx (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://localhost:11311

process[sick_mrs_6xxx-1]: started with pid [4023]
[ INFO] [1531919555.254069396]: sick_generic_caller V. 001.002.004
[ INFO] [1531919555.254113876]: Program arguments: /home/mert/ros_catkin_ws/install/lib/sick_scan/sick_generic_caller
[ INFO] [1531919555.254120863]: Program arguments: __name:=sick_mrs_6xxx
[ INFO] [1531919555.254126256]: Program arguments: __log:=/home/mert/.ros/log/32faa446-8a8c-11e8-b505-a402b94d4cfc/sick_mrs_6xxx-1.log
[ INFO] [1531919555.264481030]: Start initialising scanner ...
1531919555.2852 ERROR: Tcp::open: Failed to open TCP connection to 169.254.10.247, aborting.
[ INFO] [1531919555.286518386]: Parameter setting for <active_echo: 0>
[ INFO] [1531919555.487762772]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
1531919555.4881 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1531919560.588075899]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1531919560.588309246]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1531919560.588619553]: Receiving: <STX><ETX>
[ERROR] [1531919560.588750416]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX>sMN SetAccessMode <ETX>rGD

[ INFO] [1531919560.589169193]: Sending  : sMN SetAccessMode 3 F4724744
1531919560.5894 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1531919565.674493837]: no answer received after 5000 ms. Maybe sopas mode is wrong.
[... 103 lines cut ...]
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ INFO] [1531919643.249096371]: Start initialising scanner ...
1531919643.2494 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1531919663.538088686]: no answer received after 20000 ms. Maybe sopas mode is wrong.

[ERROR] [1531919663.538336646]: sendSOPASCommand: no full reply available for read after 20000 ms

SOPAS - Error stopping streaming scan data!
[ WARN] [1531919663.538698792]: Disconnecting TCP-Connection.
sick_scan driver exiting.
1531919663.6688 ERROR: Tcp::open: Failed to open TCP connection to 169.254.10.247, aborting.
[ INFO] [1531919663.671406449]: Parameter setting for <active_echo: 0>
[ INFO] [1531919663.871700499]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
1531919663.8718 Warning: Tcp::write: Failed to send data to socket.
^C[sick_mrs_6xxx-1] killing on exit
1531919668.9829 Warning: Tcp::write: Failed to send data to socket.
1531919674.2948 Warning: Tcp::write: Failed to send data to socket.
1531919679.3635 Warning: Tcp::write: Failed to send data to socket.
[sick_mrs_6xxx-1] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done

michael1309, 2018-07-18:
Hi, please verify that you can reach the scanner via ping. Please follow the trouble shooting instruction and use the "devel" branch.

michael1309, 2018-07-18:
Please check you firewall to ensure that the port is open.

THAC00, 2018-07-18:
Danke für die Rückmeldung . Ich antworte am besten einfach auf deutsch.

Pingen kann ich die ip. Nur scheint die TCP Verbindung mit dem Port nicht funktionieren zu wollen.

Ein paar weitere Infos :
Die ip ist statisch selbst Gesetzt (ipv4) unter den Linux Einstellung für den Ethernet Stecker.

Die ip steht so auch bei sopas ( manuell ) und ist mit Ping zu erreichen.

Die ip ist so im Launchfile für mrs6xxx eingetragen und gespeichert.

Vielen Dank  noch einmal für die Mühe !

THAC00, 2018-07-18:
Und auch die Firewall ist abgeschaltet , ich hatte sie testweise auch eingeschaltet und die Ports freigegebenen , aber danach wieder abgeschaltet .

Wenn sie noch weitere Ideen haben würde ich  gerne alles versuchen , ansonsten bastle ich weiter dran Rum bis es läuft

Beste Grüsse

From: Michael Lehning
Sent: Wednesday, July 18, 18:42
Subject: Re: [SICKAG/sick_scan] Unable to connect to remote host: Connection refused (#14)
To: SICKAG/sick_scan
Cc: Orman, Mertcan Nurhak, Author

Please check you firewall to ensure that the port is open.

THAC00, 2018-07-19:
Thank you for the support! Both via Github and via Phone. I really appreciate it! Glad I've got it to work with your help. Would have never thought of the solution myself.

Best wishes and kind regards,
Mert

closed 2018-07-19 by THAC00

michael1309, 2018-07-19:
You are welcome.
The setup steps for setting a fix scanner ip address please follow the instructions under https://github.com/SICKAG/sick_scan/blob/master/doc/ipconfig/ipconfig.md