SICKAG/sick_scan #16: initialization deadlocks, node relies on SIGTERM when closing roslaunch

kikass13, 2018-08-15:
Hello, 

your driver node will not close/exit properly because of your blocking init() function. You rely on another process just SIGTERM'ing your node instead of returning from your main thread the proper way.

... also having those 2 cascaded while loops might not be a good design.

```
while (ros::ok())
{
    s = new sick_scan::SickScanCommonTcp(hostname, port, timelimit, parser, colaDialectId);
    result = s->init();
    while (ros::ok() && (result == sick_scan::ExitSuccess))
    {
      ros::spinOnce();
      result = s->loopOnce();
    }
    if (result == sick_scan::ExitFatal)
      return result;
  }
```
a better way would be something like this (just a mockup):

```
s = new sick_scan::SickScanCommonTcp(hostname, port, timelimit, parser, colaDialectId);
while (ros::ok())
{
   result = s->initNonBlocking();
   if(result == sick_scan::ExitSuccess)
        break;
}

while (ros::ok() && result == sick_scan::ExitSuccess)
{
    // is ros spin really necessary? you dont even have incoming messages?! ... but its fine i guess ...
    ros::spinOnce();
    // this is fine
    result = s->loopOnce();
}

return result;
```

michael1309, 2018-08-19:
Hello,

Thanks for your useful hint. We will integrate your idea in our next release.

Best regards,
Michael

michael1309, 2018-08-21:
Hello,

We restructured the startup process. It is now a small state machine that works similarly to the proposed methodology. Thanks for the advice.

closed 2018-08-21 by michael1309

kikass13, 2018-08-22:
hello, thanks for these changes. 
it looks better now, but the issue of the blocking initialization function was not approached.

when i start one of the included launchfiles the following happens:

[image]

Is it possible to thread the initialization process, so that we can exit from that node the proper/intended way?

kikass13, 2018-08-23:
after i had a little dive into your code, i guess overwriting the signal handler of the ros handle should work fine ... this of course depends on whether or not your `close_device()` function and the following `m_nw.disconnect();` works as intended. 

i will try commit b0bf6976621bee5e96d55f38946936751787d33a  tomorrow and give you the feedback.

Side-Topic:

after looking through your init_device function i am not really sure why you deadlock in the first place (did i miss something special?), because all i can see is the fact that you do `m_nw.connect();` after registering some internal callback functions.

kikass13, 2018-08-23:
appendix to https://github.com/SICKAG/sick_scan/issues/16#issuecomment-415573626

ok ... after hunting for 5 incredibly cascaded `open()` calls i have found it 🍡 

https://github.com/SICKAG/sick_scan/blob/b335a04d985c1f4bbc01f91e7e044dfee1cacae0/driver/src/tcp/tcp.cpp#L172
https://github.com/SICKAG/sick_scan/blob/b335a04d985c1f4bbc01f91e7e044dfee1cacae0/driver/src/tcp/tcp.cpp#L150

would a be a good approach/is it possible to create a nonblocking socket with (http://man7.org/linux/man-pages/man2/socket.2.html)
> SOCK_NONBLOCK

? I dont understand your internal structure well enough to answer that question but im fine with the fix you have given earlier anyways , thank you for the trouble :)

kikass13, 2018-08-24:
after testing your signal handler it seems that has not been an improvement with the results.

[image]

as long as the main thread is idling inside your socket.connect (as mentioned in https://github.com/SICKAG/sick_scan/issues/16#issuecomment-415576976) we will not be able to close the node properly

michael1309, 2018-08-27:
Hello Nick,
thank you for your advice. Since we are planning further driver modifications in the next weeks, we will consider your hints in further development in order to further improve the exiting of the driver.
Many greetings,
Michael

kikass13, 2018-09-26:
is something happening ? :)

michael1309, 2018-09-26:
In the last weeks we have built a first version of the driver for ROS2. We plan to merge the ROS1 and ROS2 drivers this year. In the course of this we will also check the ticket again. Until then we would like to ask you for patience, because the abort with Ctrl-C now works on many systems. We'll come back to you then.