SICKAG/sick_scan #17: /cloud topic missing multiple echo data with SICK MRS1000

ssheikki, 2018-09-17:
When launching SICK MRS1000 scanner with
roslaunch sick_scan test_003_mrs1xxx_tf_show.launch
I am able to see data from all the echoes (parsed from /scan topic) but the data from later (2nd and 3rd) echoes is not present in the /cloud topic. I guess the later echoes should be also in /cloud topic because the topic data has place for the later echoes (size of /cloud/data is three times bigger with "filter_echoes=1" than if getting just first echo with "filter_echoes=0"). The problem seems to be that the x,y,z values are for some reason always zero.

michael1309, 2018-09-18:
Hi, we will check this issue during this week and will give you a feedback until Friday.

ssheikki, 2018-09-24:
Removing [this](https://github.com/SICKAG/sick_scan/blob/955c7f7a0edc913395b0f1a1dc21dfab30fb5d3b/driver/src/sick_scan_common.cpp#L2719) if clause seems to fix the problem. Now the cosAlphaTablePtr and sinAlphaTablePtr tables are calculated only for the first echo and are always zero for all the other echos. I think this affects all the scanners with multiple echoes.

michael1309, 2018-09-25:
Thank you for pointing that out. The lookup table for the elevation angle should now work for more than one echo. For debugging I used a transparent plastic cover (plastic: PP) and held the cover about 30 cm in front of the scanner and removed it again. One can clearly see how the scanner sees more than one echo in this direction.
[image] (e.g. you can order this from amazon: https://www.amazon.de/Hammarplast-3er-Set-Clipbox-Smart-Classic/dp/B00NKD7Y1W/ref=sr_1_5?ie=UTF8&qid=1537870427&sr=8-5&keywords=kunststoffbox+15).

Please try commit e789b2c68eafa8236806a1d233fbf510b622d72f or later from the devel branch.

michael1309, 2018-09-25:
Here are the intensity and distance image with and without cover in front of the scanner. These images are generated by using the command 

> roslaunch sick_scan pcl_converter.launch

closed 2018-09-25 by michael1309