SICKAG/sick_scan #19: build in release (gcc)

kikass13, 2018-09-26:
Hello,

your driver node will segfault if build in release mode.

Any options? :)

kikass13, 2018-09-28:
Cmakefile is below, i relayed this information from one of our developers ... 
we are setting the `CMAKE_BUILD_TYPE` variable to Release but after inspecting the CMakefile, there is no proper usage of this variable 

in other modules, we use

```
### default is Debug, we write this variable while using catkin from ros.
### catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release
if(NOT CMAKE_BUILD_TYPE)
  set(CMAKE_BUILD_TYPE Debug) #None, Debug, Release
endif()
```

Here is the Cmakefile from our repository:

```
####################################################
# CMakeLists.txt for project sick_scan
####################################################
cmake_minimum_required(VERSION 2.8.3)
project(sick_scan)

# By uncommenting the following lines you can list all variable names ...
# --- CUT ---
# get_cmake_property(_variableNames VARIABLES)
# list (SORT _variableNames)
# foreach (_variableName ${_variableNames})
#	message(STATUS "${_variableName}=${${_variableName}}")
# endforeach()
# --- CUT ---

find_package(catkin REQUIRED
    COMPONENTS
      roscpp
      sensor_msgs
      diagnostic_updater
      dynamic_reconfigure
      geometry_msgs
      std_msgs
      sensor_msgs
	  message_generation
		)

find_package(PkgConfig REQUIRED)

find_package(Boost REQUIRED COMPONENTS system)

generate_dynamic_reconfigure_options(cfg/SickScan.cfg)

add_message_files(
		FILES
		RadarObject.msg
		RadarPreHeaderDeviceBlock.msg
		RadarPreHeaderMeasurementParam1Block.msg
		RadarPreHeaderStatusBlock.msg
        RadarPreHeader.msg
		RadarScan.msg
)

generate_messages(
        DEPENDENCIES
        std_msgs
        geometry_msgs
        sensor_msgs
)

catkin_package(
        DEPENDS Boost
        CATKIN_DEPENDS roscpp sensor_msgs diagnostic_updater dynamic_reconfigure message_runtime
        LIBRARIES sick_scan_lib
        INCLUDE_DIRS include )

include_directories(include include/tcp include/tinyxml ${catkin_INCLUDE_DIRS} )

add_library(sick_scan_lib
    driver/src/sick_scan_common.cpp
    driver/src/abstract_parser.cpp
	driver/src/tcp/tcp.cpp
	driver/src/tcp/Mutex.cpp
	driver/src/tcp/SickThread.cpp
    driver/src/tcp/errorhandler.cpp
    driver/src/tcp/toolbox.cpp
    driver/src/tcp/Time.cpp
    driver/src/tcp/colaa.cpp
    driver/src/tcp/colab.cpp
	driver/src/binScanf.cpp
	driver/src/sick_scan_common_tcp.cpp
	driver/src/sick_generic_radar.cpp
	driver/src/sick_generic_parser.cpp
	driver/src/sick_scan_common_nw.cpp)

add_dependencies(sick_scan_lib ${PROJECT_NAME}_gencfg ${catkin_EXPORTED_TARGETS})
target_link_libraries(sick_scan_lib
    ${catkin_LIBRARIES})

add_executable(sick_generic_caller
    driver/src/sick_generic_laser.cpp
    driver/src/sick_generic_caller.cpp
	)

add_executable(sick_scan_radar_object_marker
        tools/src/sick_scan_radar_object_marker.cpp
        )

target_link_libraries(sick_generic_caller sick_scan_lib)
target_link_libraries(sick_scan_radar_object_marker sick_scan_lib)

add_executable(sick_scan_test
	test/src/sick_scan_test.cpp
	driver/src/tinyxml/tinystr.cpp
	driver/src/tinyxml/tinyxml.cpp
	driver/src/tinyxml/tinyxmlerror.cpp
	driver/src/tinyxml/tinyxmlparser.cpp
	)
target_link_libraries(sick_scan_test sick_scan_lib)

# Destinations:
# LIB for library
# BIN for Binaries
# INCLUDE for Includes
# SHARE for config files, launch files etc.

install(TARGETS sick_scan_lib
    DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION})

install(
    TARGETS
        sick_generic_caller
    RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION})

install(
		TARGETS
		sick_scan_test
		RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION})

install(FILES include/${PROJECT_NAME}/abstract_parser.h
              include/${PROJECT_NAME}/sick_scan_common.h
    DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION})

install(DIRECTORY test/
        DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/test 
	FILES_MATCHING PATTERN "*.xml"
	PATTERN "test/src" EXCLUDE)

install(DIRECTORY launch/
    DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/launch)

install(DIRECTORY meshes/
    DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/meshes)

install(DIRECTORY urdf/
    DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/urdf)

```

michael1309, 2018-09-28:
Thanks. We will check this issue.

michael1309, 2018-10-06:
Hi Nick,

we used your CMakeLists.txt to build the release version. The release ran smoothly in our test environment.  

Our environment was:
- Ubuntu 18.04.
- ROS1 Melodic
- x86-64 hardware
- Testing against MRS1104.
- Test against devel-branch

Could you tell us this information (Unix version, ROS version, platform, scanner type, branch) so that we can reproduce the crash in the release version? We would like to help you here. But we need the background information. It would also be also helpful to get a screenshot of the crash or coredump information.

Many greetings,
Michael

michael1309, 2018-11-29:
Hello, Nick,

I hope you were able to build the driver and the problems are solved.
So I will close this issue on 6.12.

Best regards,
Michael

closed 2019-01-07 by michael1309