SICKAG/sick_scan #20: Read sick_scan fingerprint

Ly0n, 2018-10-10:
Dear SICK Team,

thanks for creating this driver. I want to read a finder print of the sick configuration and SICK Firmware that is running on the scanner. Do you see any way how I can get this information out of your system from the ROS Site or any other communication. I want to use this information to create a system fingerprint with all the configurations that are running at the moment.

Kind regards
Tobias

michael1309, 2018-10-10:
Hi Tobias,

you could start the node and pipe the output to a file.
Grabbing the lines  like
[ INFO] [1539168876.250428942]: Receiving: <STX>sRA DeviceIdent 8 MRS1xxxx 8 1.4.1.0B<ETX>
[ INFO] [1539168876.250605851]: Deviceinfo MRS1xxxx V1.4.1.0B found and supported by this driver.
[ INFO] [1539168876.451751635]: Sending  : sRN SerialNumber
[ INFO] [1539168876.462541794]: Receiving: <STX>sRA SerialNumber 8 18340008<ETX>
[ INFO] [1539168876.663329519]: Sending  : sRN FirmwareVersion
[ INFO] [1539168876.675649844]: Receiving: <STX>sRA FirmwareVersion 9 T 1.4.1.0<ETX>
could be used to create a fingerprint.
Best regards,
Michael

closed 2018-11-30 by Ly0n

Ly0n, 2018-11-30:
Hi Michael,
thanks for your support. I integrated the fingerprint like you said.
For the future it would also be nice to read the used configuration of the Lidar, but this will work for now on.
Best regards
Tobias