SICKAG/sick_scan #21: Nodelet

RhysMcK, 2018-10-31:
I was wondering if there has been any work completed to implement this package in nodelet form? As it stands it isn't super practical to use the current implementation. 

Cheers!

michael1309, 2018-11-05:
After discussion with my colleagues, this is not at the top of our agenda. The current solution is designed to connect multiple scanners across computer boundaries. Can you explain why you currently prefer the solution as a Nodelet?

RhysMcK, 2018-11-05:
Hey @michael1309,

Having it implemented as a Nodelet will allow for this sensor to be used in more "realtime" robotic applications. I believe by not doing so really limits the use cases for this sensor in practical applications.

closed 2019-01-17 by michael1309