SICKAG/sick_scan #25: Melodic Release

JWhitleyAStuff, 2018-12-21:
We are moving our development over to Melodic and we would like to be able to use a binary version of this driver. Any ETA on a release for Melodic?

michael1309, 2018-12-22:
Hello,

building and creating a binary version under Melodic is no problem. You can build it and then use the binary.

JWhitleyAStuff, 2018-12-22:
@michael1309 I was asking for SICK to create a release for Melodic using the procedure here: http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease

michael1309, 2018-12-22:
I know what you mean. We also put this issue on the task list for Jan. 2019.

JWhitleyAStuff, 2018-12-22:
Wonderful. Thank you!

closed 2019-01-17 by michael1309

vincentrou, 2019-01-25:
this issue was closed but it is not released for melodic ?

JWhitleyAStuff, 2019-01-25:
@vincentrou - You're correct, I just checked and it hasn't gone into shadow-fixed yet. @michael1309 - Is this also planned for before the end of the month?

michael1309, 2019-01-28:
We will reopen this issue and plan to start the Melodic release until end of this week.