SICKAG/sick_scan #26: Linux Release of SOPAS ET

JWhitleyAStuff, 2018-12-21:
I know this isn't directly related to this driver, but it was the easiest way to contact you regarding this issue. Any chance there will be a Linux release of the SOPAS Engineering Tool?

michael1309, 2018-12-22:
SOPAS ET is a Windows tool and is therefore not available for Linux. But you can try to start Sopas ET under "Wine" within a Linux environment.

JWhitleyAStuff, 2018-12-22:
@michael1309 I attempted this but encountered a segfault during the installation procedure. Is there any way to change the sensor IP address without this tool?

michael1309, 2018-12-22:
Yes, we started the integration of such a tool into this software bundle Currently we deloped such a solution in a separate development branch.  It will be integrated into the main devel in Jan. 2019. Is this ok for you?

JWhitleyAStuff, 2018-12-22:
That's perfect. Thank you!

michael1309, 2019-01-24:
Hello,

we have now developed a tool to recognize the connected scanners with their IP addresses under Linux. Details can be found under the following link. 

https://github.com/SICKAG/sick_scan/blob/master/tools/sick_generic_device_finder/sick_generic_device_finder.py

We have also started with the integration of the IP address assignment. Details can be found under the link:
https://github.com/SICKAG/sick_scan/blob/devel/doc/ipconfig/ipconfig.md
We plan, as promised, to finish this work (allocation of IP addresses) by the end of next week.

michael1309, 2019-01-31:
Today we checked and updated the address assignment for several scanner types and standardized the process for the IP address assignment. We have also implemented the process for the binary Sopas.
Good luck!