SICKAG/sick_scan #30: Field breach feature not working in MRS1104c

anupritdangi, 2019-01-04:
I made an application in SOPAS tool for generating an output on i/o pins of laser whenever a field breach happens in a pre-defined field. It works fine when i start the laser. But whenever if i launch the ros package of the scanner, the package works but the breach feature is not working.

michael1309, 2019-01-04:
Hi,

the ROS node is set up so that it works in the so-called "Ranging" mode. For this purpose the Range application is activated internally and thus the Field application is deactivated. In the sense of ROS, one would then have to write an application that subscribes to the scan data and then performs the field evaluation itself.

What kind of ROS messages would you expect if a field was breached? 

Regards
Michael

michael1309, 2019-01-04:
Hi,

here are the lines of code that disable the field applications and enable the Ranging application. You may be able to branch the application if you only want to start the scanner under Linux and don't need any other data. In this case you can comment out the last ones and modify the rest of the code as you need it.

sopasCmdChain.push_back(CMD_APPLICATION_MODE_FIELD_OFF);
sopasCmdChain.push_back(CMD_APPLICATION_MODE_RANGING_ON);

Regards
Michael