SICKAG/sick_scan #31: Devel and master branch do not work with MRS1104C

BaltashovIlia, 2019-01-04:
I have MRS1104C-111011 with software version – V1.4.0.0R. It works with SOPAS correctly. 

If I use this driver of release version 0.0.13, it starts with warnings and errors but works correctly.  I actually read in previous issues that it is because the MRS1xxx does not support some legacy commands.

If I use the master branch, I have an error every time “Timeout during waiting for new datagram” and don’t get data on topics. I tried to change settings of this driver, reboot the lidar and increase timeout but every time have the same error. Maybe I have to change some other settings or change settings on MRS1104C. 

If I use the devel branch, it sometimes starts correctly, but sometimes crashes while starting. 

Here is the output of driver for each version of the driver. 
v 0.0.13: [v0013.txt](https://github.com/SICKAG/sick_scan/files/2728563/v0013.txt)
master: [branch-master.txt](https://github.com/SICKAG/sick_scan/files/2728567/branch-master.txt)
devel: [branch-devel.txt](https://github.com/SICKAG/sick_scan/files/2728568/branch-devel.txt)

michael1309, 2019-01-07:
Hello BaltashovIlia,

The problem with your scanner seems to be the experimental imu support in the devel branch.

You should disable this in the launchfile by changing this line.
`<param name="imu_enable" type="bool" value="False" />`
Then the driver should work.

On your advice I also updated this in the repo.

BaltashovIlia, 2019-01-08:
Hello michael1309, 

Now devel branch works for me, thank you. 

And it is needed to add C++11 flag to CMakeLists.txt

michael1309, 2019-01-10:
Hello BaltashovIlia,

As you suggested I added the c++11 flag.

best regards Michael

closed 2019-01-10 by michael1309