SICKAG/sick_scan #32: SICK MRS6124 doesn't filter points <min_range.

GrantDare, 2019-01-06:
Hi, 

I am using the [SICK MRS6124](https://www.sick.com/de/en/detection-and-ranging-solutions/3d-lidar-sensors/mrs6000/mrs6124r-131001/p/p533545?ff_data=JmZmX2lkPXA1MzM1NDUmZmZfbWFzdGVySWQ9cDUzMzU0NSZmZl90aXRsZT1NUlM2MTI0Ui0xMzEwMDEmZmZfcXVlcnk9JmZmX3Bvcz0xJmZmX29yaWdQb3M9MSZmZl9wYWdlPTEmZmZfcGFnZVNpemU9OCZmZl9vcmlnUGFnZVNpemU9OCZmZl9zaW1pPTkzLjA=) with this package and have experienced a problem in that the LIDAR occasionally provides a PointCloud2 data point at (0,0) relative to the origin frame. In the user manual for this LIDAR it states it has a [working range](https://cdn.sick.com/media/docs/0/40/540/Operating_instructions_MRS6000_en_IM0076540.PDF) of 0.5m - 200m. On the SOPAS software they do filter points under 0.5m however this package does not filter the points for <0.5m. This is causing problems for me integrating the LIDAR with [Costmap_2D](http://wiki.ros.org/costmap_2d) as the false data points at <0.5m are being registered as obstacles.

Are you able to integrate a depth filter for this LIDAR that filters points to the specifications of the user manual? In the meantime are you able to recommend a depth filter package?

michael1309, 2019-01-10:
Hello, Grant,

The measured values output by the scanner are arranged in the point cloud as a matrix according to their position in space. For this reason, no measured values are filtered out.
Does the status LED of the scanner turn orange when the interference occurs?
An Orange Status LED indicates occlusions of the field of view.

Best regards Michael

GrantDare, 2019-01-10:
Hi Michael, thank you for your suggestion. 

> Does the status LED of the scanner turn orange when the interference occurs?

I have tested this now and can confirm that the status LED remains a solid green over the duration when this phantom data point is present. 

> An Orange Status LED indicates occlusions of the field of view.

I assume that this would be the same as the _service needed_ indicator on the SOPAS Engineering Tool software. I also connected the sensor to this software before creating this issue to test if it was my unit that required the service however, on the software, I never received a service needed indicator.

Wissal1992, 2019-01-11:
> Hi,
>
> I am using the [SICK [...]

Hello, 
I'm also using a Lidar MRS6124 for my thesis and I'm a beginner. I would like to know how did you save your Output from the Sensor? I tried to import the measurements using the Data Recoder in Sopas in a .csv file. But when I open it it just contains only the measurements of the last Layer (ED) from every scan. Do you know how to import all of the 24 Layers into the .csv file? That would be of a great help. 

Thank you
Best regards

GrantDare, 2019-01-13:
@Wissal1992 This is not the place to ask that. This is the Git repo for the sick_scan ROS package not the SOPAS software. The issue I have raised is in relation to the functionality and conformance of this package with the specifications of the MRS6124 I would suggest using the official [SICK support portal](https://supportportal.sick.com).

GrantDare, 2019-01-14:
Hi @michael1309, still having issues with this. I have been [recommended](https://answers.ros.org/question/312336/how-to-filter-pointcloud2-data-based-on-distances/?answer=312364#post-id-312364) to use the squared distances of points to filter based on distance, however this is using [another package ](http://wiki.ros.org/pcl_ros) which I am experiencing problems with.

> The measured values output by the scanner are arranged in the point cloud as a matrix according to their position in space. For this reason, no measured values are filtered out.

Could you implement a version of this perhaps using [Euclidean and Euclidean Squared distances](http://www.improvedoutcomes.com/docs/WebSiteDocs/Clustering/Clustering_Parameters/Euclidean_and_Euclidean_Squared_Distance_Metrics.htm) to filter these points so that it conforms to the technical specifications given by SICK?

michael1309, 2019-01-17:
Hello Grant,

We have also observed that in the first column of the MRS6xxx scan, the values may fall below the minimum distance in some cases. We have reported this issue to SICK and are in contact with the relevant departments. Following the feedback from SICK, we recommend to ignore the first scan point of each layer from the scan area. In the next firmware release of MRS6xxx (Q1 of 2019 by plan), the number of scan points will be 923 per layer. Then, there is no need to care about this issue. An example for the adapted configuration can be found in the launch file sick_mrs_6xxx.launch.

Apart from this, there are three other reasons why measured values at the 0.0 point can occur.
1. The length is greater than the measuring range of the scanner, the ToF measurement results in a timeout and no valid measured value can be generated.
2. The reflectivity of the surface is too low, no laser light is reflected back and no valid value can be generated.
3. the distance to the object is smaller than 50 cm, the laser pulse returns during the dead time of the measuring system and can therefore not be detected.
These cases cannot be distinguished in the laser scanner and must therefore be treated differently depending on the application. This is why we will not filter these points.

The [solution suggested to](https://answers.ros.org/question/312336/how-to-filter-pointcloud2-data-based-on-distances/?answer=312364#post-id-312364) you we consider suitable to solve your problem because it is based on the Euclidean distance to the scanner.

closed 2019-02-07 by michael1309

GrantDare, 2019-08-29:
Hi @michael1309 can you confirm if this has been solved by now?

heizmreSICKAG, 2019-08-30:
Hi @GrantDare ,
From software version 1.1.0.634R (April 2019) this issue is solved. 
MRS6124 sends out 923 data points per row.

GrantDare, 2019-09-16:
> Hi @GrantDare ,
> From software version 1.1.0.634R (April 2019) this issue is solved.
> MRS6124 sends out 923 data points per row.

Hi, Could you please direct me to how to update the firmware :)

heizmreSICKAG, 2019-09-17:
Hi @GrantDare , please register to our SICK support portal https://supportportal.sick.com and open a ticket for MRS6000. Then one of our application engineers will directly contact you.