SICKAG/sick_scan #33: Error after closing the node

muttistefano, 2019-01-09:
Hi, 
i am experiencing a random error sometimes when i close the node.
i'm using 2 nodes for 2 lms151 lasers on a mobile platform, sometimes after i close the nodes i cannot open them anymore and my computer cannot communicate anymore with both the lasers.
If i check ifconfig i see that the communication is still active and that the packages are being dropped as errors.
After the error occurs i cannot ping the lasers and dmesg shows me :"igb 0000:01:00.1 eth2: malformed Tx packet detected and dropped, LVMMC:0xffffffff""
After this i reboot the computer and everything goes back to normal.
I run the nodes as superuser cause they are part of a biggen launchfile in which i need to be superuser.
Thanks for the help.

michael1309, 2019-01-10:
Hello muttistefano,

I can only partially reproduce your problem on my development system.
This is probably because I did not install/start ros as a super user.
There is no need to start the sick_scan node as root, so you shouldn't do this either.
Here you will find instructions on how to start only certain parts of a launch file as root:
https://answers.ros.org/question/165246/launch-node-with-root-permissions/

How do you end the nodes?
If you terminate the launch script with kill -9, the node will not close properly and the tcp connection will remain active and therefore cannot be reestablished.
Therefore you should always use CRT+C or Kill -2 (SIGINT) to terminate the script.

I hope this information will help you solve the problem.

Best regards Michael

P.S. On my system the scanner always remains pingable even if the node was not properly terminated. Possibly another program interferes with the termination of your ip stack and causes the packages to drop.

Translated with www.DeepL.com/Translator

muttistefano, 2019-01-12:
Thanks for the answer.
I will investigate to see in another programs disrupts the nodes.

closed 2019-01-17 by michael1309