SICKAG/sick_scan #34: ros topic "/scan" has no frame_id

jxhstrive, 2019-01-13:
I have set parameters in the launch file, for example, angle_max、angle_max etc.But, there are angle_min and frame_id have no the value which has been set. When I echo the topic "/scan", the frame id display empty.　So, I want to know how to set my parameters and get parameter from topic. Thank you very match!

michael1309, 2019-01-13:
Could you post the launch file and some background information about the used scanner type?

jxhstrive, 2019-01-14:
Hi michael,
Thank you! I install sick driver by the command sudo apt get install ros-kinetic-sick-scan. I get the header information: 
header: 
  seq: 360
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
angle_min: -5.0
angle_max: -1.68387436867
angle_increment: 0.00872664619237
time_increment: 0.0
scan_time: 0.0
range_min: 0.0
range_max: 0.0
These parameters are not consistent of the launch file.
Best regards!

michael1309, 2019-01-15:
Can you repeat this step by cloning the most recent firmware from the devel branch? 
In this branch are a couple of optimizations integrated.

Ttamb0, 2019-02-04:
Have these devel changes been incorporated into the master branch yet? I am having the same issue with both my lms1xx and tim_5xx lidars.

Thanks,
Thomas

michael1309, 2019-02-04:
Hi Thomas,

we will check this issue again during this week.

Best regards,
Michael

Ttamb0, 2019-02-04:
Thanks!

In case it helps, here's some more info on what I've found, based on the source code: I think the problem is occurring between lines 2965-2989 of driver/src/sick_scan_common.cpp . I believe the scanners I'm using (lms1xx and tim_5xx) are not being marked as single-layer for some reason, so the code is entering the wrong part of the if-statement and is never copying the frame id from the parameters to the temporary variable `szTmp`. This empty temporary variable is then being copied into outgoing message. I am not sure if the best solution involves marking these scanners as single-layer, or if the multilayer scanner part of the if-statement should also be copying the frame id (at which point you could just pull that out of the if-statement entirely and always copy it), or if there's a 3rd option that works better with the way you have written your code. Let me know if I can pass along any more info or logs to help you guys out!

Thomas

Ttamb0, 2019-02-12:
Hey Michael, any chance you have had a chance to take a look at this?

Thanks,
Thomas

michael1309, 2019-02-12:
Thanks for asking. The last days were a bit busy, so I'm not finished with all topics concerning the LMS5xx yet. But I have scheduled it for Thursday. In this context I am also revising the timestamp. To increase the transparency I open this ticket again.

michael1309, 2019-02-14:
Hello, Thomas,

in version V001.003.002 ( 8800570c1a5696d2e9a00b746c058577679b14d4 ) the error you described does not occur anymore. 
I tested this with an LMS111 and the following launchfile:
```
<?xml version="1.0"?>
<!-- FOR FUTURE USE. NOT SUPPORTED NOW. DO NOT USE IT. -->
<launch>
  <arg name="hostname" default="192.168.0.1" />
    <!-- robot_description and robot_state_publisher hier evtl. einbauen  -->
  <node name="sick_lms_1xx" pkg="sick_scan" type="sick_generic_caller" respawn="false" output="screen">
    <!-- default values: -->
    <!--
      <param name="intensity" type="bool" value="True" />
      <param name="skip" type="int" value="0" />
      <param name="frame_id" type="str" value="laser" />
      <param name="time_offset" type="double" value="-0.001" />
      <param name="publish_datagram" type="bool" value="False" />
      <param name="subscribe_datagram" type="bool" value="false" />
      <param name="device_number" type="int" value="0" />
      <param name="range_min" type="double" value="0.05" />
	<param name="min_ang" type="double" value="-1.658" />
	<param name="max_ang" type="double" value="1.658" />
    -->
      <param name="frame_id" type="str" value="laser_test" />
    	<param name="use_binary_protocol" type="bool" value="False" />
	<param name="scanner_type" type="string" value="sick_lms_1xx"/>
	<param name="range_max" type="double" value="25.0" />
        <param name="hostname" type="string" value="$(arg hostname)" />
	<param name="port" type="string" value="2112" />
	<param name="timelimit" type="int" value="5" />

    
  </node>
</launch>
```
The `rostopic echo /scan` returns the following: 

---
header: 
  seq: 374
  stamp: 
    secs: 1550140591
    nsecs: 462321807
  frame_id: "laser_test"
angle_min: -2.35619449615
angle_max: 2.35619449615
angle_increment: 0.00872664619237
time_increment: 2.77777780866e-05
scan_time: 0.019999999553
range_min: 0.0500000007451
range_max: 25.0
ranges: [0.07000000029802322, 0.07000000029802322, 0.05299999937415123, 0.06400000303983688,

I hope that solves your problems.

best regards Michael

closed 2019-02-28 by michael1309