SICKAG/sick_scan #35: Error during launching process with driver LMS5XX

saikishor, 2019-01-23:
I have been trying to get the scan data from the sensor LMS511-10100 but whenever I launch I am getting the following error:
```

Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://egypt:41511/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /sick_lms_5xx/hostname: 192.168.0.1
 * /sick_lms_5xx/intensity: True
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 1.658062789
 * /sick_lms_5xx/min_ang: -1.658062789
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_max: 100.0
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: True

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [10204]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3cd37a84-1efb-11e9-a66f-b86b23e49ec3
process[rosout-1]: started with pid [10217]
started core service [/rosout]
process[sick_lms_5xx-2]: started with pid [10227]
[ INFO] [1548240002.320133878]: sick_generic_caller V. 001.001.001
[ INFO] [1548240002.320182711]: Program arguments: /opt/ros/kinetic/lib/sick_scan/sick_generic_caller
[ INFO] [1548240002.320198790]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1548240002.320208939]: Program arguments: __log:=/home/pal/.ros/log/3cd37a84-1efb-11e9-a66f-b86b23e49ec3/sick_lms_5xx-2.log
[ INFO] [1548240002.335761219]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1548240002.335812793]: Binary protocol activated
[ INFO] [1548240002.335839401]: Start initialising scanner ...
[sick_lms_5xx-2] process has died [pid 10227, exit code -11, cmd /opt/ros/kinetic/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/pal/.ros/log/3cd37a84-1efb-11e9-a66f-b86b23e49ec3/sick_lms_5xx-2.log].
log file: /home/pal/.ros/log/3cd37a84-1efb-11e9-a66f-b86b23e49ec3/sick_lms_5xx-2*.log
```
TCP connection result
```
nc -z -v -w5 192.168.0.1 2112
Connection to 192.168.0.1 2112 port [tcp/*] succeeded!
```

I found my sensor IP to be `192.168.0.1` and I used the same in my hostname address, but I am not successful to get the data in ros topic.

Any idea what could be wrong?

michael1309, 2019-01-23:
Hi,

you can try the following actions:
a) Trying the most recent version of the "devel"-branch.
b) Switching the sensor to ASCII-Mode by setting use_binary_protocol to False.
c) You can try also the brand new tool https://github.com/SICKAG/sick_scan/blob/devel/tools/sick_generic_device_finder/sick_generic_device_finder.py
which searches for connected scanners and generate for each found scanner an xml file.
d) paranoid-check: Please disconnect your scanner and retry nc -z -v -w5 192.168.0.1 2112.
   It should give you - of course - an error in this case.

saikishor, 2019-01-23:
@michael1309 isn't the `devel` branch unstable?

saikishor, 2019-01-23:
@michael1309 This is the output I got first time I ran the launch file

```
roslaunch sick_scan sick_lms_5xx.launch 
... logging to /home/pal/.ros/log/6125b6bc-1f2e-11e9-8025-b86b23e49ec3/roslaunch-egypt-20927.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://egypt:37974/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /sick_lms_5xx/hostname: 192.168.0.1
 * /sick_lms_5xx/intensity: True
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 1.658062789
 * /sick_lms_5xx/min_ang: -1.658062789
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_max: 100.0
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: False

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [20938]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6125b6bc-1f2e-11e9-8025-b86b23e49ec3
process[rosout-1]: started with pid [20951]
started core service [/rosout]
process[sick_lms_5xx-2]: started with pid [20964]
[ INFO] [1548261967.591785701]: sick_generic_caller V. 001.002.010
[ INFO] [1548261967.591834035]: Program arguments: /home/pal/sick_80m_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1548261967.591853415]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1548261967.591864353]: Program arguments: __log:=/home/pal/.ros/log/6125b6bc-1f2e-11e9-8025-b86b23e49ec3/sick_lms_5xx-2.log
[ INFO] [1548261967.607659100]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1548261967.607726280]: ASCII protocol activated
[ INFO] [1548261967.607762787]: Start initialising scanner [Ip: 192.168.0.1] [Port: 2112]
[ INFO] [1548261967.649678772]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1548261967.659516730]: Parameter setting for <active_echo: 0>
[ INFO] [1548261967.859929362]: Sending  : sMN SetAccessMode 3 F4724744
[ERROR] [1548261972.937359476]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1548261972.937424875]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1548261972.937501142]: Receiving: <STX><ETX>
[ERROR] [1548261972.937577496]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN SetAccessMode 3 F4724744<ETX>

[ INFO] [1548261972.937768792]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
[ INFO] [1548261972.948048734]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1548261973.148444080]: Sending  : sWN EIHstCola \x00
[ INFO] [1548261973.158851969]: Receiving: <STX>sWA EIHstCola <ETX>
[ERROR] [1548261973.159047023]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ INFO] [1548261973.159183121]: Start initialising scanner [Ip: 192.168.0.1] [Port: 2112]
[ WARN] [1548261973.159220273]: Disconnecting TCP-Connection.
1548261974.1508 Info: Tcp::readInputData: Read 0 bytes - connection is lost!
sick_scan driver exiting.
[... 553 lines cut ...]
done)
```

c) This is the output of your generic device finder python script
```
rosrun sick_scan sick_generic_device_finder.py 
If you have more than one network interface it may happen that no scanner is found because the broadcast ip address does not match and the discovery packets are sent to the wrong interface.To fix this problem change the parameter <UDP_IP = "192.168.0.255"> to the broadcast address that ifconfig returns for your network interface e.g. "192.168.178.255".
UDP target IP: 192.168.0.255
UDP target port: 30718
The scan result is available in 10 seconds.
Unhandled exception in thread started by 
```

d) I checked the same as of your instructions
When the device is disconnected to the system
```
nc -z -v -w5 192.168.0.1 2112
nc: connect to 192.168.0.1 port 2112 (tcp) failed: Connection refused
```

When connected:
```
nc -z -v -w5 192.168.0.1 2112
Connection to 192.168.0.1 2112 port [tcp/*] succeeded!
```

saikishor, 2019-01-24:
@michael1309 I have also added two PR's for both master and devel addressing the compilation issues that persist right at the moment. Please kindly consider them and help me with the driver issue

michael1309, 2019-01-24:
Thanks for your PR. We will integrate this into the driver and really appreciate your input.
Regarding the crash: I see that you are now able to start the communication. Can you try to switch back to "binary"-mode and retry your test? 
Could you attach current launch file to the ticket?
Best regards,
Michael

saikishor, 2019-01-24:
@michael1309 Sure! I will do that and update you.

saikishor, 2019-01-24:
@michael1309  please keep me posted on this issue.
Thanks in advance.

michael1309, 2019-01-24:
We use your launch file in our test. We could not see any problems using your launch file with most recent version from today. Please try the devel branch again. If you have further problems, please try to compile the node as debug version and start it with gdb. This would be very helpful.
Does the node crash directly after sending the last command?

saikishor, 2019-01-24:
@michael1309 Sure! Yes it do crashes immediately after the last command.

saikishor, 2019-02-07:
@michael1309 Sure, I will give it a try tomorrow and get back to you. Thanks for updating me on this.

saikishor, 2019-02-07:
@michael1309 btw, where can I check the Version: 001.003.001 that you were mentioning. I can only see the tag ´0.0.15´. Let me know, if you are referring to a different one.

michael1309, 2019-02-07:
Good point. I tagged the most recent version with "V001.003.002".

You can find the version information in the file containing the main-routine.
It is combination of the following defines:
#define SICK_GENERIC_MAJOR_VER "001"
#define SICK_GENERIC_MINOR_VER "003"
#define SICK_GENERIC_PATCH_LEVEL "002"

You can find this combination also in the console output after launch the node in the first
INFO-Line. 
Example
```
ROS_MASTER_URI=http://localhost:11311

process[sick_lms_5xx-1]: started with pid [28045]
[ INFO] [1549568743.199436231]: sick_generic_caller V. 001.003.002
```

saikishor, 2019-02-08:
@michael1309  I have no luck trying this version too. Below is the output I get with the new version. 

```
~/sick_80m_ws/src$ nc -z -v -w5 192.168.0.1 2112
Connection to 192.168.0.1 2112 port [tcp/*] succeeded!
```
```
~/sick_80m_ws/src$ roslaunch sick_scan sick_lms_5xx.launch 
... logging to /home/pal/.ros/log/96a1c17c-2bb4-11e9-9139-b86b23e49ec3/roslaunch-egypt-20802.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://egypt:38943/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /sick_lms_5xx/hostname: 192.168.0.1
 * /sick_lms_5xx/intensity: True
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 1.658062789
 * /sick_lms_5xx/min_ang: -1.658062789
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_max: 100.0
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: True

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [20812]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 96a1c17c-2bb4-11e9-9139-b86b23e49ec3
process[rosout-1]: started with pid [20825]
started core service [/rosout]
process[sick_lms_5xx-2]: started with pid [20838]
[ INFO] [1549639023.835237497]: sick_generic_caller V. 001.003.002
[ INFO] [1549639023.835325988]: Program arguments: /home/pal/sick_80m_ws/devel_release/lib/sick_scan/sick_generic_caller
[ INFO] [1549639023.835349144]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1549639023.835361212]: Program arguments: __log:=/home/pal/.ros/log/96a1c17c-2bb4-11e9-9139-b86b23e49ec3/sick_lms_5xx-2.log
[ INFO] [1549639023.851236638]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1549639023.851308240]: Binary protocol activated
[ INFO] [1549639023.851342000]: Start initialising scanner [Ip: 192.168.0.1] [Port: 2112]
[ INFO] [1549639023.891209582]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1549639023.899674472]: Parameter setting for <active_echo: 0>
[ INFO] [1549639024.099955195]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ERROR] [1549639029.666758376]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1549639029.666817350]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1549639029.666865024]: Receiving: <STX><ETX>
[ERROR] [1549639029.666898149]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>

[ INFO] [1549639029.667006640]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1549639029.670248756]: Receiving: <STX>sAN SetAccessMode 1<ETX>
[ INFO] [1549639029.870479674]: Sending  : <STX>sWN EIHstCola 1<ETX>
[ INFO] [1549639029.871638918]: Receiving: <STX>sWA EIHstCola<ETX>
[ERROR] [1549639029.871714821]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ INFO] [1549639029.871784213]: Start initialising scanner [Ip: 192.168.0.1] [Port: 2112]
[ WARN] [1549639029.871822651]: Disconnecting TCP-Connection.
1549639030.8727 Info: Tcp::readInputData: Read 0 bytes - connection is lost!
sick_scan driver exiting.
[ INFO] [1549639030.928003183]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1549639030.938648836]: Parameter setting for <active_echo: 0>
[ INFO] [1549639031.138894788]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
1549639031.1819 ERROR: Tcp::readInputData: Failed to read data from socket, aborting!
[ERROR] [1549639036.710606904]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1549639036.710659931]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1549639036.710703638]: Receiving: <STX><ETX>
[ERROR] [1549639036.710741066]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>

[ INFO] [1549639036.710803614]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
1549639036.7108 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1549639042.280135011]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1549639042.280184842]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1549639042.280229715]: Receiving: <STX><ETX>
[ERROR] [1549639042.280260313]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN SetAccessMode 3 F4724744<ETX>

[ERROR] [1549639042.280313524]: Error Access Mode
[ INFO] [1549639042.480616002]: Sending  : <STX>sWN EIHstCola 1<ETX>
1549639042.4808 Warning: Tcp::write: Failed to send data to socket.
^C[sick_lms_5xx-2] killing on exit
[rosout-1] killing on exit
^C[master] killing on exit
^Cshutting down processing monitor...
... shutting down processing monitor complete
done
```

michael1309, 2019-02-08:
Please set sick_lms_5xx/use_binary_protocol to False in the launch file and try again after reboot.

michael1309, 2019-02-09:
I checked your output. It looks like the destructor of a vector crashes. 
Could you try if this crash also happens in ASCII mode? Would you be able to start the application in an IDE with a debugger? 
In any case the communication with the scanner starts correctly.

michael1309, 2019-02-28:
No more feedback. So we guess the problem was solved.

closed 2019-02-28 by michael1309