SICKAG/sick_scan #39: No data of topic /scan sick_lms_5xx

jxhstrive, 2019-02-21:
Hi,
I have set IP address with LMS5XX by GUI software. And, I could see laser scan data by the monitor. But, when I start the launch file and echo the topic "/scan", I could not get laser scan data. I got the following logs,
  PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /sick_lms_5xx/hostname: 10.158.51.32
 * /sick_lms_5xx/intensity: True
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 1.658062789
 * /sick_lms_5xx/min_ang: -1.658062789
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_max: 100.0
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: True

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://10.158.51.29:11311

process[sick_lms_5xx-1]: started with pid [24395]
[ INFO] [1550722702.086664959]: sick_generic_caller V. 001.003.000
[ INFO] [1550722702.086692882]: Program arguments: /home/qomolo/Documents/rosbag_dir/xuhui/sick_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1550722702.086716148]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1550722702.086722743]: Program arguments: __log:=/home/qomolo/.ros/log/eeb27d7c-357b-11e9-bf32-78d00424d2f4/sick_lms_5xx-1.log
[ INFO] [1550722702.097072508]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1550722702.097100779]: Binary protocol activated
[ INFO] [1550722702.097112719]: Start initialising scanner [Ip: 10.158.51.32] [Port: 2112]
[ INFO] [1550722702.115367444]: Publishing laserscan-pointcloud2 to cloud
1550722831.5599 ERROR: Tcp::open: Failed to open TCP connection to 10.158.51.32, aborting.
[ INFO] [1550722831.562960441, 8561.610000000]: Parameter setting for <active_echo: 0>
[ INFO] [1550722831.761597736, 8561.810000000]: Sending  : <STX><STX><STX><STX><Len=0x385875968>sMN SetAccessMode 0x030xf40x720x470x440xb3
1550722831.7617 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1550722837.077645091, 8567.080000000]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1550722837.077713629, 8567.080000000]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1550722837.077920499, 8567.080000000]: Receiving: <STX><ETX>
[ERROR] [1550722837.077960882, 8567.080000000]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0x385875968>sMN SetAccessMode 0x030xf40x720x470x440xb3

[ INFO] [1550722837.078136507, 8567.080000000]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
1550722837.0782 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1550722842.395589192, 8572.360000000]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1550722842.395642207, 8572.360000000]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1550722842.395740799, 8572.360000000]: Receiving: <STX><ETX>
[ERROR] [1550722842.395776243, 8572.360000000]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN SetAccessMode 3 F4724744<ETX>

[ERROR] [1550722842.395837867, 8572.360000000]: Error Access Mode
[ INFO] [1550722842.593395393, 8572.560000000]: Sending  : <STX>sWN EIHstCola 1<ETX>
1550722842.5934 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1550722847.910055973, 8577.830000000]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1550722847.910115626, 8577.830000000]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1550722847.910209830, 8577.830000000]: Receiving: <STX><ETX>
[ERROR] [1550722847.910255128, 8577.830000000]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sWN EIHstCola 1<ETX>

[ERROR] [1550722847.910328478, 8577.830000000]: Command or Error message not defined
[ERROR] [1550722847.910402470, 8577.830000000]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ INFO] [1550722847.910455437, 8577.830000000]: Start initialising scanner [Ip: 10.158.51.32] [Port: 2112]
[ WARN] [1550722847.910498577, 8577.830000000]: Disconnecting TCP-Connection.
sick_scan driver exiting.

jxhstrive, 2019-02-21:
Hi,
I git clone the devel branch code. I have four sick lasers, the others work well. I have not found the problem of this situation. Maybe something wrong with configure.  Thank you for reply!

michael1309, 2019-02-21:
Hi,

In your log you can the following line:
1550722831.5599 ERROR: Tcp::open: Failed to open TCP connection to 10.158.51.32, aborting

This means, that the driver is no able to establish a TCP connection to your scanner.
Could you check you ip connection by the following command?

nc -z -v -w5 10.158.51.32 2112

For further details see "Troubleshooting".

What is about the other scanners? Could you give some feedback about these types and connections?
Is there something different compared to your "problem scanner"?

Could you switch cables, ip addresses, power supply etc. to cross check your installation between the laser scanners?

jxhstrive, 2019-02-21:
Hi michael,
I check ip connection by the command "nc -z -v -w5 10.158.51.32 2112", The connection is succeeded. And I get scan data. It's cool. But I get some errors sometimes. Giving the logs as follows:

[ INFO] [1550742816.188649982]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ERROR] [1550742819.729566995]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ERROR] [1550742819.729613019]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1550742819.729643758]: Receiving: <STX><ETX>
[ERROR] [1550742819.729663065]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX>sMN SetAccessMode <ETX>rGD

[ INFO] [1550742819.729700960]: Sending  : sMN SetAccessMode 3 F4724744
[ INFO] [1550742819.739833947]: Receiving: <STX>sAN SetAccessMode 1<ETX>
[ INFO] [1550742819.940112522]: Sending  : sWN EIHstCola 1
[ INFO] [1550742819.950366172]: Receiving: <STX>sWA EIHstCola<ETX>
[ERROR] [1550742819.950486835]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "sopas_protocol_type" to overwrite the default settings of the driver.
[ INFO] [1550742819.950650744]: Start initialising scanner ...
[ERROR] [1550742840.227516750]: no answer received after 20000 ms. Maybe sopas mode is wrong.

[ERROR] [1550742840.227624204]: sendSOPASCommand: no full reply available for read after 20000 ms

SOPAS - Error stopping streaming scan data!
[ WARN] [1550742840.227754621]: Disconnecting TCP-Connection.
1550742840.9767 Info: Tcp::readInputData: Read 0 bytes - connection is lost!
sick_scan driver exiting.
[ INFO] [1550742842.012274417]: Parameter setting for <active_echo: 0>
[ INFO] [1550742842.212628143]: Sending  : sMN SetAccessMode \x03\xf4\x72\x47\x44
[ INFO] [1550742842.222946202]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1550742842.423331668]: Sending  : sWN EIHstCola \x01
[ INFO] [1550742842.433645162]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1550742842.634006282]: Sending  : sRN FirmwareVersion
[ INFO] [1550742842.644351813]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x31\x2e\x38\x30\x20\x20\x20\x20<ETX>
[ INFO] [1550742842.844733395]: Sending  : sRN SCdevicestate
[ INFO] [1550742842.865203705]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1550742843.065543228]: Sending  : sRN ODoprh
[ INFO] [1550742843.086194564]: Receiving: <STX>sRA ODoprh \x00\x00\x02\xe7<ETX>
[ INFO] [1550742843.310335495]: Sending  : sRN ODpwrc
[ INFO] [1550742843.320767844]: Receiving: <STX>sRA ODpwrc \x00\x00\x00\x30<ETX>
[ INFO] [1550742843.537928615]: Sending  : sRN LocationName
[ INFO] [1550742843.548276360]: Receiving: <STX>sRA LocationName \x00\x0b\x53\x4e\x20\x31\x38\x31\x34\x30\x31\x31\x30<ETX>
[ INFO] [1550742843.567610485]: Sending  : sRN LMPoutputRange
[ INFO] [1550742843.577990378]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x13\x88\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1550742843.578177138]: Angle resolution of scanner is 0.50000 [deg]  (in 1/10000th deg: 0x1388)
[ INFO] [1550742843.578269212]: MIN_ANG:   -1.658 [rad]  -95.000 [deg]
[ INFO] [1550742843.578349633]: MAX_ANG:    1.658 [rad]   95.000 [deg]
[ INFO] [1550742843.578473241]: Sending  : sWN LMPoutputRange \x00\x01\x00\x00\x13\x88\xff\xff\x3c\xb0\x00\x1c\x3a\x90
[ INFO] [1550742843.588760232]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1550742843.588906162]: Sending  : sRN LMPoutputRange
[ INFO] [1550742843.599234074]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x13\x88\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1550742843.599435251]: Angle resolution of scanner is 0.50000 [deg]  (in 1/10000th deg: 0x1388)
[ INFO] [1550742843.632538336]: MIN_ANG (after command verification):   -1.658 [rad]  -95.000 [deg]
[ INFO] [1550742843.632959068]: MAX_ANG (after command verification):    1.658 [rad]   95.000 [deg]
[ INFO] [1550742843.633039472]: LMS 5xx detected overwriting output channel flag ID
[ INFO] [1550742843.633134449]: LMS 5xx detected overwriting resolution flag (only 8 bit supported)
[ INFO] [1550742843.633310095]: Sending  : sWN LMDscandatacfg \x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01
[ INFO] [1550742843.643581206]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1550742843.643763080]: Sending  : sRN LMDscandatacfg
[ INFO] [1550742843.654066906]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01<ETX>
[ INFO] [1550742843.654288098]: Sending  : sMN LMCstartmeas
[ INFO] [1550742843.664672365]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1550742843.664829357]: Sending  : sMN Run

Best regards,
jxh

michael1309, 2019-02-21:
Hi Jxh,

great to hear. You can ignore the error messages. These are standard messages during handling the bin/ascii mode switching. Maybe we should declare these messages in the near future as infos messages. 

Also you can update to the most recent version. It gives you an optimized feedback for the log messages.

Have fun and best regards,
Michael

closed 2019-02-21 by michael1309