SICKAG/sick_scan #43: Is there a way to know whether the mounted laser is TiM 571 OR TiM 561?

saikishor, 2019-03-29:
Hello!!

I have a question, may be its unrelated. With the TiM 5XX series you have the same product ID and vendor ID. So, it's hard to differentiate the sensor programmatically. Is there a way to know the difference between sensors, using software?

michael1309, 2019-04-01:
Hello,

maybe a small hint: You could try to configure the max. distance to e.g. 15m and then read the set values from the scanner again.  Because the TiM561 only scans up to 10m and the TiM571 up to 25m, the values should differ.

Best regards,
Michael

saikishor, 2019-04-01:
@michael1309 Thanks for your input. Right now, we are following the same method to differentiate between the sensors. It seems that's the only way to differentiate. Do you know if there is a pattern with respect to the serial number assigned to the sensors, that could help?.

michael1309, 2019-04-01:
I will ask the R&D department at SICK for more background information during this week and will give you an update.

saikishor, 2019-04-01:
@michael1309 Thanks a lot for the help. Will be waiting for your reply.

michael1309, 2019-04-03:
You can use the SOPAS-Command "Dltype". Attached you find the command structure. By using this command you can retrieve the information about the device type. 

Thanks to @heizmreSICKAG for his support!
[image]

closed 2019-04-03 by michael1309