SICKAG/sick_scan #45: LMS511-20100 PRO compatibility

themightyoarfish, 2019-06-27:
I am trying to use this driver with the [LMS511-20100 PRO](https://www.sick.com/us/en/detection-and-ranging-solutions/2d-lidar-sensors/lms5xx/lms511-20100-pro/p/p216240) device.

- I have verified that the firmware is recent enough (v1.80.1, which is enough according to the [telegram listing](https://cdn.sick.com/media/docs/7/27/927/Technical_information_Telegram_Listing_Ranging_sensors_LMS1xx_LMS5xx_TiM5xx_TiM7xx_LMS1000_MRS1000_MRS6000_NAV310_LD_OEM15xx_LD_LRS36xx_LMS4000_en_IM0045927.PDF) p. 6)
- The SOPAS ET also shows the scans

However, the ROS node only manages to exchange configuration data with the sensor, the connection times out while waiting for the Scan data.

```
started roslaunch server http://carrera-Inspiron-3470:37347/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /sick_lms_5xx/frame_id: cloud
 * /sick_lms_5xx/hostname: 192.168.178.75
 * /sick_lms_5xx/intensity: True
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 1.658062789
 * /sick_lms_5xx/min_ang: -1.658062789
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_max: 100.0
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: True

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [27064]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 80cd4df8-98bf-11e9-a466-7440bb669aff
process[rosout-1]: started with pid [27077]
started core service [/rosout]
process[sick_lms_5xx-2]: started with pid [27080]
[ INFO] [1561628388.635597677]: sick_generic_caller V. 001.003.018
[ INFO] [1561628388.635702190]: Program arguments: /home/carrera/ros_workspace/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1561628388.635741878]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1561628388.635766187]: Program arguments: __log:=/home/carrera/.ros/log/80cd4df8-98bf-11e9-a466-7440bb669aff/sick_lms_5xx-2.log
[ INFO] [1561628388.687573896]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1561628388.687649963]: Binary protocol activated
[ INFO] [1561628388.687687535]: Start initialising scanner [Ip: 192.168.178.75] [Port: 2112]
[ INFO] [1561628388.754458974]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1561628388.771312146]: Parameter setting for <active_echo: 0>
[ INFO] [1561628388.972265947]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1561628388.977205522]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1561628389.177832408]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1561628389.180351792]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1561628389.380970419]: Sending  : <STX><STX><STX><STX><Len=0019>sRN FirmwareVersion CRC:<0x24>
[ INFO] [1561628389.383774925]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x31\x2e\x38\x30\x2e\x31\x20\x20<ETX>
[ INFO] [1561628389.584454158]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1561628389.587045913]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1561628389.787704937]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODoprh CRC:<0x41>
[ INFO] [1561628389.790249458]: Receiving: <STX>sRA ODoprh \x00\x00\x02\xe4<ETX>
[ INFO] [1561628389.992744883]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODpwrc CRC:<0x52>
[ INFO] [1561628389.995109682]: Receiving: <STX>sRA ODpwrc \x00\x00\x00\x24<ETX>
[ INFO] [1561628390.196151839]: Sending  : <STX><STX><STX><STX><Len=0016>sRN LocationName CRC:<0x55>
[ INFO] [1561628390.198597626]: Receiving: <STX>sRA LocationName \x00\x0b\x53\x4e\x20\x31\x39\x31\x32\x31\x30\x37\x33<ETX>
[ INFO] [1561628390.200412487]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1561628390.202827123]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x06\x83\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1561628390.202948240]: Angle resolution of scanner is 0.16670 [deg]  (in 1/10000th deg: 0x683)
[ INFO] [1561628390.202993736]: MIN_ANG:   -1.658 [rad]  -95.000 [deg]
[ INFO] [1561628390.203019947]: MAX_ANG:    1.658 [rad]   95.000 [deg]
[ INFO] [1561628390.203128499]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x06 0x83 0xff 0xff 0x3c 0xb0 0x00 0x1c 0x3a 0x90 CRC:<0xc5>
[ INFO] [1561628390.205467923]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1561628390.205622923]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1561628390.207967062]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x06\x83\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1561628390.208085994]: Angle resolution of scanner is 0.16670 [deg]  (in 1/10000th deg: 0x683)
[ INFO] [1561628390.210879157]: MIN_ANG (after command verification):   -1.658 [rad]  -95.000 [deg]
[ INFO] [1561628390.210936437]: MAX_ANG (after command verification):    1.658 [rad]   95.000 [deg]
[ INFO] [1561628390.210981492]: LMS 5xx detected overwriting output channel flag ID
[ INFO] [1561628390.211013171]: LMS 5xx detected overwriting resolution flag (only 8 bit supported)
[ INFO] [1561628390.211148542]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x42>
[ INFO] [1561628390.213475927]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1561628390.213626819]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[ INFO] [1561628390.216030863]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01\x00\x01<ETX>
[ INFO] [1561628390.218778720]: Sending  : <STX><STX><STX><STX><Len=0016>sMN LMCstartmeas CRC:<0x68>
[ INFO] [1561628390.221183741]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1561628390.221348258]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1561628390.223717597]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1561628390.223873229]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1561628390.226185308]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ WARN] [1561628410.230353418]: Timeout during waiting for new datagram
[ERROR] [1561628410.230449886]: Read Error when getting datagram: 1.
```

It seems that the sensor acknowledges the request **EA LMDscandata**, but the excpected telegram stream does not come through.

I realise this device is not officially supported, but I expect there are no changes in the protocol between the different lms5xx sensors, right?

michael1309, 2019-06-27:
Hello Rasmus

Even if the LMS 511 is not officially supported, I tested the software with a LMS511-10100 (firmware V1.80). There the communication with the scanner is successfully initialized.

Can you check with wireshark or similar if the scanner sends scan data? The data packets should look like this:
[image]

Best regards Michael

themightyoarfish, 2019-06-27:
Thanks for your reply.

The wireshark capture does not show any packets arriving after the `ROS_ERROR` about the timeout. The few `LMDscandata` packets are much shorter than in your screenshot and seem to concern the scanner's acknowledgement of the request.

[image]

themightyoarfish, 2019-06-27:
I should note that a colleague of mine apparently used the same model successfully with the node, so I'm a bit baffled what the issue can be.
I tried both with the scanner available in the network (static IP, as in the screenshot above), and directly ehternet-connected to the computer running the Node. Same result both times.

I'm grateful for any avenues of debugging.

michael1309, 2019-06-27:
We recommend that you test your colleague's scanner in your setup. If the other scanner works and your scanner doesn't, we recommend that you contact SICK Support directly. We close this ticket because it looks more like a scanner problem than a software problem. 

Have you possibly set and saved a completely different scan rate or similar at an earlier time?
What we recommend:
Reset the scanner with its parameters to the factory default state and then save these parameters permanently in the scanner. You can do all this with SOPAS ET if you are logged in as an "Authorized Client".
 
If you have any new findings, you are welcome to reopen the ticket.

closed 2019-06-27 by michael1309

themightyoarfish, 2019-07-04:
As another debugging vector, I inspected the Wireshark capture when scanning with the SOPAS ET software. The software does not seem to use verbs/commands which are documented in the Telegram listing, is that expected? E.G. there were verbs such as `sRI`, which are not documented.
Therefore, I could not ascertain whether maybe one of the methods of getting scan data has a bug in the firmware version.

michael1309, 2019-07-04:
This is the default behaviour of Sopas Et. Sopas ET uses a short form/special form of command ids. Could you try the scanner  of your colleague?

themightyoarfish, 2019-07-05:
Thanks for this information. I will be in contact with Sick support soon; unfortunately the working scanner was a device borrowed from Sick and we do not have it anymore.

themightyoarfish, 2019-07-08:
Update: We have moved the scanner to a different Network, and it works now. I have no idea what network property was the culprit, but I invite everyone experiencing similar issues to try their device in different networks to make sure it is not related to the environment.

michael1309, 2019-07-09:
Very good to hear. If you still find differences that explain the behavior, I am grateful for every update.
Ideas:

- 100 Mbit vs GigE
- Switches and Routers
- Wiring and cables

Thank you very much for your feedback.