SICKAG/sick_scan #46: ERROR in: sudo apt-get install ros-<rosdistro>-sick_scan

dylanDC-Combi, 2019-07-04:
this command is wrong, when you run 

> apt search ros-<ROS-DISTRO> 

for the distro melodic it returns the package as sick-scan and not sick_scan, as a result this command only works as:

> sudo apt-get install ros-melodic-sick-scan

there are also several errors that come up as a result, I'm assuming, where ROS cannot launch the drivers either as sick_scan or sick-scan while following the quick-start guide 

[image]

michael1309, 2019-07-04:
Good point. We will check this for the next release. Meanwhile we recommend the direct git pull from the master branch.

michael1309, 2019-08-29:
We've checked the package names. It is the rule that the underscore is converted to a minus sign. So it is correct that you use "sudo apt-get install ros-melodic-sick-scan" for the installation.

closed 2019-08-29 by michael1309