SICKAG/sick_scan #47: SICK MRS6124 header stamp is in the future

GrantDare, 2019-07-17:
Hi Developer,

I am using ros Kinetic Kame and have installed the debian package for this using

`sudo apt-get install ros-kinetic-sick-scan`

and am experiencing that the PointCloud2 message that is produced is on average 4300 seconds in the future? The issue is when using a `static_transform_publisher` from the tf library to provide the transform from laser->base_link, i receive this error

```Transform [sender=unknown_publisher]
For frame [laser_link]: No transform to fixed frame [base_link]. TF error: [Lookup would require extrapolation into the future. Requested time 1563407610.392146348 but the latest data is at time 1563404409.628742959, when looking up transform from frame [laser_link] to frame [base_link]]```

I wrote a quick analysis node to test the time differences and there is a huge difference between the PointCloud2 message and ros::Time::now(). When I republish the message with the correct time stamp, the visualization is correct from the base_link frame.

```#include <ros/ros.h>
#include <sensor_msgs/PointCloud2.h>

ros::Publisher cloud_pub;

void publisher_analysis(const sensor_msgs::PointCloud2::ConstPtr& msg){
    ros::Time msg_time = msg->header.stamp;
    ros::Time now = ros::Time::now();
    ros::Duration diff = now - msg_time;
    ROS_INFO("msg: %f\tnow: %f\tDifference %f",msg_time.toSec(), now.toSec(), diff.toSec() );
    sensor_msgs::PointCloud2 new_msg = *msg;
    new_msg.header.stamp = ros::Time::now();
    cloud_pub.publish(new_msg);
}

int main(int argc, char** argv){
    ros::init(argc, argv, "analysis_tool");
	ros::NodeHandle n;

    ros::Subscriber subs = n.subscribe<sensor_msgs::PointCloud2>("/cloud", 10, &publisher_analysis);
    cloud_pub = n.advertise<sensor_msgs::PointCloud2>("/cloud_corrected", 10);
    ros::spin();
}```

Here is a sample terminal message to demonstrate the readout I am experiencing

```[ INFO] [1563407656.560052456]: msg: 1563411944.716735  now: 1563407656.559996  Difference -4288.156739
[ INFO] [1563407656.661517240]: msg: 1563411966.456520  now: 1563407656.661473  Difference -4309.795047
[ INFO] [1563407656.762293701]: msg: 1563411984.364266  now: 1563407656.762240  Difference -4327.602026
[ INFO] [1563407656.858648356]: msg: 1563411919.840781  now: 1563407656.858621  Difference -4262.982160
[ INFO] [1563407656.965505452]: msg: 1563411902.182181  now: 1563407656.965480  Difference -4245.216702
[ INFO] [1563407657.062846706]: msg: 1563411964.189110  now: 1563407657.062805  Difference -4307.126304```

michael1309, 2019-07-18:
Hi,

for the correct calculation of the time stamps, we have developed a software PLL which compares the internal 32-bit counter of the laser scan with the ROS time stamps and generates the time stamp of origin from this. This should also work for the MRS6124. We will consult with SICK on this and, if appropriate, investigate the time stamp generation for the MRS6124 again in the coming weeks. In the meantime, we recommend that you comment out the call of the software PLL. Then you will get the timestamp of the packets directly after receiving the datagrams. Since the receipt of datagrams and the sending of ROS messages run in different processes, we recommend the use of the timestamp retrieved in line 2426 (all line numbers are referenced to https://github.com/SICKAG/sick_scan/blob/master/driver/src/sick_scan_common.cpp ).
Please comment out line 2593.

`bRet = SoftwarePLL::instance().getCorrectedTimeStamp(recvTimeStamp.sec, recvTimeStamp.nsec,SystemCountScan);`
change to:
`// bRet = SoftwarePLL::instance().getCorrectedTimeStamp(recvTimeStamp.sec, recvTimeStamp.nsec,SystemCountScan);`

This should give you a valid timestamp as a workaround.

BaltashovIlia, 2019-08-25:
Hi,
Occasionally I have the same problem with MRS1104C-111011. Time difference is between 2 and 4300 seconds. 
Commenting of line 2593 solved the problem. But I have not tested it enough yet. 
Has this solution any drawbacks?

michael1309, 2019-08-26:
Hi,
that is a valid workaround. Maybe this could yield a small (ignorable) time jitter in your scan data.

michael1309, 2019-08-29:
We investigated the problem and were able to reproduce it. It is an overflow problem that we will correct in September 2019. Until then, however, the workaround is valid.

michael1309, 2019-09-05:
@GrantDare and @BaltashovIlia

We have examined the issue more closely. 

The so-called software PLL is used to precisely determine the generation time of a scan. In order to determine this time stamp, we compare the receiving time stamp of the IP packet in the PC with the time stamp of transmission (leaving the scanner) in the datagram. The datagram contains two time stamps:
a) the generation time stamp (in ticks)
b) the transmission time stamp (in ticks)
Both are indicated in so-called ticks. In order to compensate jitter and possible network latencies, we calculate a moving average value over the receive time stamps to generate a homogeneous time stamp progression. For details, please have a look at the code and the following explanations.

As mentioned the driver has a time correction algorithm called `SoftwarePLL`. This calculates the actual time at which the scan was created. In the datagrams there are two 32 bit time stamps (microseconds since scanner start) containing the time when the scan was started (`SystemCountScan`) and the time when the SOPAS data packet was sent (`SystemCountTransmit`). The driver also generates a time stamp (`recvTimeStamp`) when the network stack receives the packet.
The` SoftwarePLL` is updated with the function `updatePLL(recvTimeStamp.sec, recvTimeStamp.nsec,SystemCountTransmit)` .
In the `SoftwarePLL`  the correlation between `SystemCountTransmit` stamp and packet reception time stamp is calculated.
Then for each packet the time stamp at which the scan was started is calculated from the `SystemCountScan` time stamp.
For the MRS6000 this leads to the following timing results:

| ROS Timestamp | uncorrected time |corrected time|SystemCountTransmit|SystemCountScan|uncorrected time-corrected time|
|---------------|------------------|--------------|-------------------|---------------|-------------------------------|
|[ INFO] [1567685648.217552425]| 1567685648.216935|1567685648.041867|95603878|95428734|0.175068|
|[ INFO] [1567685648.218119646]| 1567685648.217529|1567685648.041867|95604316|95428734|0.175663|
|[ INFO] [1567685648.218637057]| 1567685648.217947|1567685648.041867|95604717|95428734|0.176080|
|[ INFO] [1567685648.219105852]| 1567685648.218404|1567685648.041867|95605189|95428734|0.176537|
|[ INFO] [1567685648.219565695]| 1567685648.218723|1567685648.041867|95605519|95428734|0.176856|
|[ INFO] [1567685648.220026850]| 1567685648.219095|1567685648.041867|95605821|95428734|0.177228|
|[ INFO] [1567685648.220491092]| 1567685648.219456|1567685648.066925|95606138|95453734|0.152531|
|[ INFO] [1567685648.220945576]| 1567685648.219456|1567685648.066925|95606457|95453734|0.152531|
|[ INFO] [1567685648.221448060]| 1567685648.219983|1567685648.066925|95606777|95453734|0.153057|
|[ INFO] [1567685648.221921251]| 1567685648.220339|1567685648.066925|95607084|95453734|0.153414|
|[ INFO] [1567685648.222393874]| 1567685648.220665|1567685648.066925|95607404|95453734|0.153740|
|[ INFO] [1567685648.222853046]| 1567685648.220665|1567685648.066925|95607710|95453734|0.153740|
|[ INFO] [1567685648.223319450]| 

It can be seen that 6 layers always have the same `SystemCountScan` time stamp but arrive at different arrival times in the pc system and were transmitted at different times from the scanner.
The `SoftwarePLL` assigns the same time stamp to each of these scans because they were created at the same time.
The time difference between corrected and uncorrected time stamp corresponds to the duration of the scanner needs to rotate the mirrors and send the datagrams.

We tested the improved algorithm against MRS6124, TiM571, MRS1104 and LMS4111.
The most recent patches are included in the current devel branch.

closed 2019-09-05 by michael1309