SICKAG/sick_scan #49: Is there anyway to check encoder value from ROS LMS1xx & LMS5xx?

adrianwslee, 2019-09-04:
Hello, I'm currently using LMS151 with rotary encoder connected for the project.

Telegram Listing shows I can get the amount of encoder, speed, and position.

The problem is I can't find where to get this data from the code. Can you give me a hint?

Thanks

michael1309, 2019-09-04:
Hi,

please use the devel-branch to resolve this issue. Currently we did not export the encoder settings to a ROS message. 

Currently, we have not exported the encoder settings to an ROS message. A good start for parsing the encoder settings is the line no. 2632 in https://github.com/SICKAG/sick_scan/blob/devel/driver/src/sick_scan_common.cpp

We patched the code here last week to process encoder messages correctly.
For more details, I have attached the most important information on this topic from the telegram list.

[image]

adrianwslee, 2019-09-05:
Thank you for your fast reply!

When I use Windows to set LMS151 to get encoder value using SOPAS and check from the terminal, I see a datagram numOfEncoder part gets changed from 0 to 1.

However, when I move back to Linux and run the launch file, the datagram numofEncoder part sets back to 0.

I assume this has to do with setting up the sensor. Is there a part (LICsrc) where I can set Incremental source to Encoder?

michael1309, 2019-09-05:
I think that after setting the values in SOPAS ET you probably have to write the values back to the flash memory of the device to keep it permanently.
To test this, proceed as follows:
a) Start SOPAS ET, make your settings and save them in the device
b) Close SOPAS ET
c) Restart scanner (power off,  wait 10 sec., power on)
d) Start SOPAS ET
e) Check encoder settings
f) If the settings are then correct, this is proof that the values are stored correctly in the flash memory.
g) If this is successful, try again under Linux.

adrianwslee, 2019-09-06:
I've tried that but I had no luck. So I went back to check the code.

Here is what I did:

*Using devel branch
on line 910

`sopasCmdMaskVec[CMD_SET_PARTIAL_SCANDATA_CFG] = "\x02sWN LMDscandatacfg %02d 00 %d %d 0 00 00 0 0 0 1 1\x03";`

I seemed like configuration command is setting encoder setting as 0 as a default value.

So I switched to 

`\x02sWN LMDscandatacfg %02d 00 %d %d 0 **01** 00 0 0 0 1 1\x03;`

from that, I was able to get the amount of encoder, position, and speed.

Thenceforth, I changed _sick_generic_parser.cpp_ fields(+2 after field[17]) to correspond with changes made in fields number. I was able to get ROS laserscan data too.

I really appreciate your help in resolving my issue.

closed 2019-09-06 by adrianwslee