SICKAG/sick_scan #51: False laser data

fayyazpocker, 2019-09-10:
I am using **SICK LMS101** and **sick_lms_1xx** node to read laser data. However, i am getting false data towards the left side of the SICK sensor as shown in the image. 
[image]

I am using the following configuration:

    <node name="sick_lms_1xx" pkg="sick_scan" type="sick_generic_caller" respawn="false">
        <param name="intensity" type="bool" value="False" />
        <param name="intensity_resolution_16bit" type="bool" value="false" />
        <param name="min_ang" type="double" value="-1.658" />
        <param name="max_ang" type="double" value="1.658" />
        <param name="frame_id" type="str" value="$(arg sick_laser_frame_id)" />
        <param name="use_binary_protocol" type="bool" value="false" />
        <param name="scanner_type" type="string" value="sick_lms_1xx"/>
        <param name="range_max" type="double" value="18.0" />
        <param name="range_min" type="double" value="0.5" />
        <param name="hostname" type="string" value="$(arg ip_sick)" />
        <param name="port" type="string" value="2112" />
        <param name="timelimit" type="int" value="1" />
    </node>

michael1309, 2019-09-10:
Please check the scanner in other installation situations. A good possibility is e.g. scanning against diffuse room walls. I don't think that this is a software issue at the moment. Please contact SICK Support here if necessary.

fayyazpocker, 2019-09-10:
Is it possible that this behavior is due to any configuration set in Sopas ??

michael1309, 2019-09-10:
Possibly. In order to be completely safe, you can reset the scanner to the factory settings using SOPAS ET and save the reset changes in the EEPROM of the scanner.

michael1309, 2019-09-24:
We guess that you were able to solve this issue by the given hints. Otherwise you can reopen this ticket.

closed 2019-09-24 by michael1309