SICKAG/sick_scan #53: No IMU values posted to topic /imu

GusGoliath96, 2019-09-24:
Hello, I am trying to SLAM using the IMU values ​​provided by my SICK MRS1000 sensor, but when I run "rostopic echo /imu", no data is posted in this topic. My SOPAS setting is set to ranging, RSSI parameter is enabled, communication is set to binary.
I added these parameters to my node:
<param name="imu_enable" type="bool" value="True" />
<param name="imu_enable_additional_info" type="bool" value="True" />

I'm trying to use gmapping node but without the IMU values it's not possible.
Thanks for help

michael1309, 2019-09-24:
Hi,

did you enable the imu flag in the launch file?

**<param name="imu_enable" type="bool" value="True" />**

See IMU.md for details.

GusGoliath96, 2019-09-24:
yes, this booth parameters 

<param name="imu_enable" type="bool" value="true" />
<param name="imu_enable_additional_info" type="bool" value="true" />

michael1309, 2019-09-25:
Did you check your MRS1xxx order to see if it contains an IMU? 
Could you upload the startup log and the launch file?

SICK-WestCoast, 2019-09-25:
All MRS1000 from May builds on have IMU! Andrew

michael1309, 2019-09-25:
@GusGoliath96 

Please post your serial number and firmware version. SICK can check whether your MRS1xxx contains an IMU.

GusGoliath96, 2019-09-25:
Yes, i did see the flag in the IMU.md and added to my launch file. 
I suspect it's because i have bought the sensor i'm using before may 2019 and it's firmware version is 1.4.0.0R
Probably I need to upgrade the firmware version to get the IMU support.

michael1309, 2019-10-20:
Clarified with support.

closed 2019-10-20 by michael1309