SICKAG/sick_scan #55: A possible mistake about parse code

JinwenJay, 2019-10-21:
In sick_scan_common.cpp, line 3109. I think it should be 
elevationAngleDegree =  -elevationAngleDegree / 180.0 * M_PI instead of 
elevationAngleDegree =  elevationAngleDegree / 180.0 * M_PI
for the calculation after it.
If not, the present code recognize MRS1104 scanner's 4 lines as -2.5, 0, 2.5 and 5 degree, which is not correspond to its documents(-5, -2.5, 0 and 2.5 degree).

michael1309, 2019-10-21:
Hi  @JinwenJay ,
did you derive this information only from the manual?
During development we checked the angle orientation against a real target. 
You can check it yourself, for example, by moving a triangle with its tip pointing downwards through the layers from the top to bottom of your room. You can check the sequence by looking at the point cloud in rviz. If this test confirms your statement, we will like to take up this topic again.

michael1309, 2019-10-24:
Hi @JinwenJay ,

to be on the safe side, we tested the layer orientation once again. For this purpose we have executed the following task. We placed the MRS1104 on a flat floor and let it look against a wall with a cardboard box on top. The rviz screenshot in the appendix shows how the scan planes cut the cardboard from bottom to top. This corresponds to the expectation.

[image]
[image]
[image]

closed 2019-10-24 by michael1309