SICKAG/sick_scan #58: SICK S300 ROS Driver?

seanyen, 2019-11-08:
I am curious is there any plan to bring up ROS driver for `SICK S300` from `SICKAG`?

I think the problem I am having is that I saw there are some projects working on that, but it is unclear to me which one would be the most current (and stable) to use:
- https://github.com/ipa320/cob_driver
- https://github.com/RobotnikAutomation/s3000_laser
- https://github.com/RobotnikAutomation/sick_s300_laser

michael1309, 2019-11-09:
Currently there are no concrete plans to support this lidar. However, you are welcome to use the code and adapt it for your purposes.

closed 2019-11-09 by michael1309

NicolasLoeffler, 2019-11-11:
@seanyen in addition to @michael1309 post: there will be no official ROS driver for our S300 family. Other customers gave us positive reviews on the first driver you've mentioned. 

We officially support our latest safety laser scanner families: microScan3, outdoorScan3 and nanoScan3. You'll find the ROS driver here: https://github.com/SICKAG/sick_safetyscanners.