SICKAG/sick_scan #59: lidar mrs1000 doesn't publish anything at /cloud

richardz03, 2019-11-15:
Hi Michael,

I ran sick_mrs_1xxx.launch and followed your quickstart, but it doesn't publish anything in /cloud, I verify it with rostopic echo. Is there anything additional that I need to do to get this part working? The connection is fine.

Hopefully I am allowed to ask you a general question here, if not please ignore this part. I ran the lidar with another package sick_tim mrs1000 launch file using LaserScan type msg, and noticed that usually the intensity is 0 at the edges of my hand, and If I move any object relatively fast in front of the lidar,  values with 0 intensity are returned from some of the angles that the object cover.  Is it common for the LIDAR to react like this?

michael1309, 2019-11-15:
Hi Richard,

please send a log of the startup phase for an improved analysis.

Best regards, 
Michael

richardz03, 2019-11-16:
Sorry, I just noticed that a while after succesfully connecting the LIDAR ip, it just fails to open TCP connection to port 2112.  I believe changing to 2111 should fix my problem.

closed 2019-11-16 by richardz03