SICKAG/sick_scan #60: Corrupted /cloud pointcloud when publishing IMU on cloud_output_mode =2

ROS-lover, 2019-11-15:
Hi,

First of all, thanks for the useful development! I am using Sick-MRS1000 and after enabling the following parameters. (using the sick_mrs_1xxx_cartographer.launch)
      <param name="imu_enable" type="bool" value="True" />
      <param name="cloud_output_mode" type="int" value="2" />
      <param name="imu_enable_additional_info" type="bool" value="True" />
Why setting "cloud_output_mode" = 2 ? this was empirical, any other value will result in /imu topic with fixed (0) orientation variables.
ISSUE: when I view the /cloud on rviz, the points seem to be corrupted, as if only parts of the points were being published.
It might be that I am doing something wrong, but I am not sure. 
[image]

Thanks!

michael1309, 2019-11-15:
Please use the most recent version of the driver. We removed the fix values of orientation in this version and fixed a minor bug in the imu time stamping.

For your mentioned setup the Pointcloud2 is chopped into small parts. That is a feature to generate partial point clouds at a high frequency rate output to support slam in a better way.

You could check this in rviz by increasing the decay value in the Pointcloud2 viewer control.

ROS-lover, 2019-11-17:
Thanks @michael1309 . Followed the instruction and everything works just fine.

closed 2019-11-17 by ROS-lover