SICKAG/sick_scan #61: Cannot recieve LMDradardata when running sick_rms_3xx.launch

kwildermuth1, 2019-11-15:
Hello, 

Thank you for building and managing this software. I'm running sick scan with the RMS320 radar unit. When I run sick_rms_3xx.launch, I get a connection to the radar but no target tracking or object messages are being published. I thought this was the expected behavior. It seems like the application is getting hung up when trying to recieve the LMDradardata. Below is an output of the log (sorry that it is an image and not text, this is all that I have at the moment)

[image]

Some other notes:

- We do see targets when we run the visualization in SOPAS
- when I run 'rostopic list' while sick_rms_3xx.launch is running, I see the following topics, but I do not see any messages on the topics:
/cloud
/cloud_radar_rawtarget
/cloud_radar_track
/diagnostics
/imu
/radar
/rosout
/rosout_agg
/scan
/sick_rms_3xx/parameter_descriptions
/sick_rms_3xx/parameter_updates

I know the repo documentation describes RMS320 being under development, so I'm wondering if this is related to that or if it is possibly a user error.

Thanks for your help

michael1309, 2019-11-17:
Hi,

all in all the start sequence looks ok. We will check the radar interface during the next week and will give a feedback. In the meanwhile you can checkout b92f60493ff560cb6db216a92a1c5002ab12e7db commit 2018-07-19.
In this commit, we have made adaptations to the radar. Possibly the source code additions in the lidar domain didn't completely take into account something in the radar domain. Feedback is very welcome.

Regards,

kwildermuth1, 2019-11-18:
@michael1309  Thanks for the quick reply. It will be much appreciated to get feedback on the radar interface from your perspective when you test this week. 

Should there be any additional logging in the terminal when I run the application if it were successfully transmitting object data from the radar?

Also, I do not see any changes about the radar in the commit you reference, it only shows a few comments changed on a ipconfig.md file

kwildermuth1, 2019-11-18:
Also is there any additional configuration I need to do in order to get radar data sending? It just seems peculiar that it is hanging at this last command <STX>sEN LMDradardata 1<ETX>

michael1309, 2019-11-20:
Hi,

please try to use the following command to check the radar messages:
`rostopic echo /radar
`
Here is a typical output:
`
---
header: 
  seq: 3859
  stamp: 
    secs: 1574249935
    nsecs: 591188907
  frame_id: "radar"
radarPreHeader: 
  uiVersionNo: 1
  radarPreHeaderDeviceBlock: 
    uiIdent: 1
    udiSerialNo: 18020073
    bDeviceError: False
    bContaminationWarning: False
    bContaminationError: False
  radarPreHeaderStatusBlock: 
    uiTelegramCount: 36989
    uiCycleCount: 36991
    udiSystemCountScan: 2386444184
    udiSystemCountTransmit: 2386446915
    uiInputs: 0
    uiOutputs: 0
  radarPreHeaderMeasurementParam1Block: 
    uiCycleDuration: 46272
    uiNoiseLevel: 0
  radarPreHeaderArrayEncoderBlock: 
    - 
      udiEncoderPos: 0
      iEncoderSpeed: 0
targets: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  height: 0
  width: 0
  fields: []
  is_bigendian: False
  point_step: 0
  row_step: 0
  data: []
  is_dense: False
objects: 
  - 
    id: 0
    tracking_time: 
      secs: 0
      nsecs:         0
    last_seen: 
      secs: 0
      nsecs:         0
    velocity: 
      twist: 
        linear: 
          x: 0.0
          y: 0.0
          z: 0.0
        angular: 
          x: 0.0
          y: 0.0
          z: 0.0
      covariance: [1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0]
    bounding_box_center: 
      position: 
        x: 2.08000016212
        y: -0.416000008583
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    bounding_box_size: 
      x: 1.0
      y: 1.7
      z: 1.7
    object_box_center: 
      pose: 
        position: 
          x: 2.08000016212
          y: -0.416000008583
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
      covariance: [1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0]
    object_box_size: 
      x: 1.0
      y: 1.7
      z: 1.7
    contour_points: []
  - 
    id: 0
    tracking_time: 
      secs: 0
      nsecs:         0
    last_seen: 
      secs: 0
      nsecs:         0
    velocity: 
      twist: 
        linear: 
          x: 0.0
          y: 0.20000000298
          z: 0.0
        angular: 
          x: 0.0
          y: 0.0
          z: 0.0
      covariance: [1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0]
    bounding_box_center: 
      position: 
        x: 6.80000019073
        y: -1.26400005817
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.707106769085
        w: 0.707106769085
    bounding_box_size: 
      x: 1.0
      y: 1.7
      z: 1.7
    object_box_center: 
      pose: 
        position: 
          x: 6.80000019073
          y: -1.26400005817
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.707106769085
          w: 0.707106769085
      covariance: [1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0]
    object_box_size: 
      x: 1.0
      y: 1.7
      z: 1.7
    contour_points: []
---
`

michael1309, 2019-11-20:
[radarlog.txt](https://github.com/SICKAG/sick_scan/files/3869120/radarlog.txt)
I have now checked the behaviour of the radar again and rebuilt the code so that the radar code is now well encapsulated in a singleton. We also activated the raw targets and tracking objects as Pointcloud2 data for your debugging. In the appendix you can find sample output (search for "command" and "output").
What surprised me about your screenshot was the output "Read Error when getting datagram". In my test setup, targets are always output several times per second, even in a stationary environment. 

In summary please now try the following commands:
`rostopic echo /radar
rostopic echo /cloud_radar_rawtarget
rostopic echo /cloud_radar_track
`

michael1309, 2019-11-20:
The new version is 1.5.0

michael1309, 2019-12-10:
Closed to inactivity due to discussion rules.

closed 2019-12-10 by michael1309

alfred31, 2020-02-28:
Hi, I reopen this question.

First of all, thank you for the amazing job you made.

The thing is that I am also using a sick radar, and I successfully launched the driver, that is already working, with no errors. The problem I am getting is that I don't realy know which topic should give the information of the radar, I mean, and don't see any changes in the topics when I put objects closer or further to the sensor. Besides, I was able to see the pointclouds coming from the topics in rviz, but I just don't get what it works, they appear to be always the same...

Could you please tell me what I am doing wrong? How can I get the data from the sensor? Is it possible to watch it in raviz?

Thak you very much!

Regards.