SICKAG/sick_scan #63: mysterious IP address change

jbreiden2, 2019-12-09:
I have a TIM571 lidar on ethernet and am trying to get data. The driver thinks it needs to change the IP addresses (to the same value that is passed in through the hostname parameter) and exits. I believe this is incorrect. If I edit `sick_scan_common.cpp` to force `setNewIPAddr` to false, I am able to successfully talk to the lidar and retrieve data. Is this a bug?

```
$ roslaunch ./install/share/sick_scan/launch/sick_tim_5xx.launch hostname:=192.168.1.55
... logging to /home/redacted/.ros/log/b70497fc-d237-11e5-8faf-78d00428e3d2/roslaunch-redacted-7719.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://redacted:34527/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /sick_tim_5xx/cloud_topic: cloud
 * /sick_tim_5xx/frame_id: cloud
 * /sick_tim_5xx/hostname: 192.168.1.55
 * /sick_tim_5xx/intensity: True
 * /sick_tim_5xx/max_ang: 2.35619449
 * /sick_tim_5xx/min_ang: -2.35619449
 * /sick_tim_5xx/port: 2112
 * /sick_tim_5xx/range_max: 100.0
 * /sick_tim_5xx/scanner_type: sick_tim_5xx
 * /sick_tim_5xx/timelimit: 5
 * /sick_tim_5xx/use_binary_protocol: True

NODES
  /
    sick_tim_5xx (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://redacted:11311

process[sick_tim_5xx-1]: started with pid [7739]
[ INFO] [1457001050.769576625]: sick_generic_caller V. 001.003.020
[ INFO] [1457001050.769609580]: Program arguments: /home/redacted/ros_catkin_ws/install/lib/sick_scan/sick_generic_caller
[ INFO] [1457001050.769621585]: Program arguments: __name:=sick_tim_5xx
[ INFO] [1457001050.769629799]: Program arguments: __log:=/home/redacted/.ros/log/b70497fc-d237-11e5-8faf-78d00428e3d2/sick_tim_5xx-1.log
[ INFO] [1457001050.781107158]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1457001050.781143549]: Binary protocol activated
[ INFO] [1457001050.781158697]: Start initialising scanner [Ip: 192.168.1.55] [Port: 2112]
[ INFO] [1457001050.811092825]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1457001050.825629784]: Parameter setting for <active_echo: 0>
[ INFO] [1457001051.025995637]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1457001051.031433837]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1457001051.031566028]: Sending  : <STX><STX><STX><STX><Len=0017>sWN EIIpAddr 0xc0 0xa8 0x01 0x37 CRC:<0x32>
[ INFO] [1457001051.033717970]: Receiving: <STX>sWA EIIpAddr <ETX>
[ INFO] [1457001051.033759907]: Sending  : <STX><STX><STX><STX><Len=0015>sMN mEEwriteall CRC:<0x21>
[ INFO] [1457001051.109640576]: Receiving: <STX>sAN mEEwriteall \x01<ETX>
[ INFO] [1457001051.109801563]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1457001051.115244765]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1457001051.115404463]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1457001051.120006369]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1457001051.120138271]: Sending  : <STX><STX><STX><STX><Len=0013>sMN mSCreboot CRC:<0x2c>
[ INFO] [1457001051.197314197]: Receiving: <STX>sAN mSCreboot <ETX>
[ INFO] [1457001051.197373189]: IP address changed. Node restart required
[ INFO] [1457001051.197386667]: Exiting node NOW.
[sick_tim_5xx-1] process has finished cleanly
```

michael1309, 2019-12-10:
Could you reproduce this phenomena with the most recent version from the github-repo master branch?

jbreiden2, 2019-12-10:
Yes, I built from source.

michael1309, 2019-12-10:
Can you upload your launch file?

michael1309, 2019-12-10:
The most recent version in the master branch is "1.5.0". You built from "001.003.020". Could you update to "1.5.0" from this repo and try it again?

michael1309, 2019-12-12:
Please verify that the variable "new_IP_address" is deleted after launching a launch file which uses this variable (e.g. sick_tim_5xx_new_ip.launch). As an alternative you can also restart roscore.

michael1309, 2019-12-12:
We guess that this recommendation solves the problem. In other case please feel free to reopen this issue.

closed 2019-12-12 by michael1309

jbreiden2, 2019-12-12:
>Please verify that the variable "new_IP_address" is deleted after launching a launch file 
>which uses this variable (e.g. sick_tim_5xx_new_ip.launch). As an alternative you can 
>also restart roscore.

Restarting roscore solved the problem. Thank you.

While we are talking, I should mention that the instructions for installing from source at these two locations disagree with each other. The one on wiki.ros.org should probably say "catkin_make install" install of "catkin_make".

http://wiki.ros.org/sick_scan                              
https://github.com/SICKAG/sick_scan