SICKAG/sick_scan #65: `range_min` and `range_max` ignored.

themightyoarfish, 2019-12-13:
Despite the fact that the launch file for LMS 5xx lists the two parameters as possible settings, actually assigning them has no effect on the pointcloud. I searched in the repo and all they seem to be used for is setting some metadata fields in a ´sensor_msgs::LaserScan` message which is not what is used for the LMS. Perhaps the documentation on these parameters should be improved.

michael1309, 2019-12-17:
Good point. We will verify that. In principle, however, we recommend that you also check the distance in the application anyway. At least as a workaround, this is the preferred solution in this case.

michael1309, 2020-03-02:
In the meantime we have decided not to use the parameters range_min and range_max according to your suggestion. In almost all cases known to us, this will be filtered directly in the target application if necessary.

closed 2020-03-02 by michael1309