SICKAG/sick_scan #66: Timing issues?

fehlfarbe, 2019-12-16:
Hi!
We replaced a Sick TiM laserscanner on our robot with a Sick MRS1000 (with IMU) and switched from the sick_tim ros node to sick_scan. The robots navigates in a building with Google Cartographer that is running on a different computer. Both systems are connected via wifi and the time is synchronized with chrony. After the change to Sick MRS1000 and sick_scan the cartographer node crashes after some seconds or minutes because the imu or scan timestamps aren't in order.
Odometry from the robot controller isn't effected so I don't think it's an time sync issue with chrony (makestep is disabled). When I use `cloud_output_mode = 0` cartographer crashes after some minutes because it gets wrong IMU data. When I use `cloud_output_mode = 2` it crashes immediately after startup so I assume it has the same timing problem with cloud data:

```
F1216 15:03:25.232957  4597 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (637121018052091973 vs. 637121018051480653) Non-sorted data added to queue: '(0, points2)'
[FATAL] [1576505005.233187751]: F1216 15:03:25.000000  4597 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (637121018052091973 vs. 637121018051480653) Non-sorted data added to queue: '(0, points2)'
*** Check failure stack trace: ***
    @     0x7efc90b8d5cd  google::LogMessage::Fail()
    @     0x7efc90b8f433  google::LogMessage::SendToLog()
    @     0x7efc90b8d15b  google::LogMessage::Flush()
    @     0x7efc90b8fe1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x75844b  cartographer::sensor::OrderedMultiQueue::Dispatch()
    @           0x758de7  cartographer::sensor::OrderedMultiQueue::Add()
    @           0x756950  cartographer::sensor::Collator::AddSensorData()
    @           0x672fed  cartographer::mapping::CollatedTrajectoryBuilder::AddData()
    @           0x643b78  cartographer::mapping::CollatedTrajectoryBuilder::AddSensorData()
    @           0x635c3c  cartographer_ros::SensorBridge::HandleOdometryMessage()
    @           0x5e8b37  cartographer_ros::Node::HandleOdometryMessage()
    @           0x608ece  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x6136f5  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7efc8e4ffd8d  ros::SubscriptionQueue::call()
    @     0x7efc8e4a5838  ros::CallbackQueue::callOneCB()
    @     0x7efc8e4a723b  ros::CallbackQueue::callAvailable()
    @     0x7efc8e503e39  ros::SingleThreadedSpinner::spin()
    @     0x7efc8e4e8e9b  ros::spin()
    @           0x5e17a6  cartographer_ros::(anonymous namespace)::Run()
    @           0x5de474  main
    @     0x7efc8cef5830  __libc_start_main
    @           0x5e12a9  _start
    @              (nil)  (unknown)
```

I'm running the latest commit from master branch ( 6a6968ee286e6557aefe01a4212a07eb829e2fa0 )

MRS1000 launch:
```
<node machine="robot" name="sick_mrs_1xxx" pkg="sick_scan" type="sick_generic_caller" respawn="true" output="screen">
        <param name="scanner_type" type="string" value="sick_mrs_1xxx"/>
        <param name="min_ang" type="double" value="-2.3998277" />
        <param name="max_ang" type="double" value="+2.3998277" />
        <param name="intensity_resolution_16bit" type="bool" value="false" />
        <param name="hostname" type="string" value="192.168.0.1" />
        <param name="port" type="string" value="2112" />
        <param name="timelimit" type="int" value="5" />
        <param name="cloud_topic" type="string" value="points2"/>
        <param name="frame_id" type="str" value="laser" />

        <param name="filter_echos" type="int" value="2" />
        <param name="imu_enable" type="bool" value="True" />
        <param name="cloud_output_mode" type="int" value="2" />
        <param name="imu_enable_additional_info" type="bool" value="True" />
      </node>
```

michael1309, 2019-12-17:
Based on my experience I can only recommend the `cloud_output_mode = 2`. The mode was introduced especially for the cartographer as experimental mode. Google Cartographer expects a high rate of pointcloud messages so that the data per message is not too old. The IMU-messages should fit in time. You can check it by swiveling the scanner up and down in front of a wall and plotting the distance and IMU values against each other in rqt_plot.

fehlfarbe, 2019-12-18:
Hi,

I created some plots. This is with `scan_mode = 2`. On the y-axis there is the timestamp of the PointCloud2 message header and on the x-axis I use the current ROS timestamp. After start the timestamp jumps backwards and cartographer crashes:
[image]

`scan_mode = 0` works for a while but after some minutes I get the same error message with IMU data. When I zoom into the data there is also a time jump in the message header:
[image]
[image]

michael1309, 2019-12-24:
Hello,

thanks for the analysis and the report. We have analyzed this phenomenon. In our opinion, the problem is caused by a feature that is very useful in principle. We have integrated a so-called "software PLL" that converts the ticks of the scanner into a time stamp that corresponds to the actual generation time of the scan. But as long as the PLL is not locked (it needs some scans at the beginning of the transmission), the timestamp of the IP data reception is used. The transition from the unlocked state to the locked state then results in the timestamp return. The getCorrectedTimeStamp function returns as return value whether the PLL is locked or not.

We recommend to use this return value to avoid sending an IMU message or a Pointcloud2 message in case of the unlocked state.

More details about the mechanism can be found at https://github.com/SICKAG/sick_scan/blob/master/doc/timing.md

You can see the relevant code parts in the attached screenshot

[image]

closed 2019-12-24 by michael1309