SICKAG/sick_scan #78: sick_rms_3xx.launch data received not working

alfred31, 2020-02-28:
Hello..

First of all, thank you for the amazing job you made.

The thing is that I am also using a sick radar, and I successfully launched the driver, that is already working, with no errors. The problem I am getting is that I don't really know which topic should give the information of the radar, I mean, and don't see any changes in the topics when I put objects closer or further to the sensor. Besides, I was able to see the pointclouds coming from the topics in rviz, but I just don't get what it works, they appear to be always the same...

Could you please tell me what I am doing wrong? How can I get the data from the sensor? Is it possible to watch it in raviz?

Thak you very much!

Regards.

alfred31, 2020-02-28:
Also I made it work with the windows software, and I could see it working when someone was around the radar, however I am not able to reproduce the same performance with the ros software.

Thank you for your help.

Regards

michael1309, 2020-03-02:
I think you use the launch file radar_object_marker.launch for the visualization. Actually, this is where the radar messages are converted one-to-one into visualization messages. It might be better for your application if you change the tracking method. Otherwise, there is actually no explanation why the radar should behave differently under Windows than under Linux or ROS. Maybe you have deactivated the raw targets by mistake?

alfred31, 2020-03-03:
Hello,

Thank you for your answer.
I'll dig into what you say, and see if that is the problem.
I will come back with the outcomes.

Regards

alfred31, 2020-03-03:
Hello, 
I launched everything and I am still getting to the same situation. This is what I got using the windows SOPA program you provided on your webpage :
[image]

Now using the ROS driver,firstly  I launch the sick_rms_3xx.launch (after having the IP address configured) , and I get the following output:
roslaunch sick_scan sick_rms_3xx.launch 
... logging to /home/gupg/.ros/log/acc594c0-5d30-11ea-8e99-f0d5bfebbcac/roslaunch-gupg-ThinkPad-T460-5135.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server 

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /sick_rms_3xx/emul_sensor: False
 * /sick_rms_3xx/hostname: 192.168.25.35
 * /sick_rms_3xx/port: 2112
 * /sick_rms_3xx/range_max: 250.0
 * /sick_rms_3xx/scanner_type: sick_rms_3xx
 * /sick_rms_3xx/timelimit: 5
 * /sick_rms_3xx/tracking_mode: 0
 * /sick_rms_3xx/transmit_objects: True
 * /sick_rms_3xx/transmit_raw_targets: True

NODES
  /
    sick_rms_3xx (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://localhost:11311

process[sick_rms_3xx-1]: started with pid [5157]
[ INFO] [1583227688.078129974]: sick_generic_caller V. 1.5.0
[ INFO] [1583227688.078254660]: Program arguments: /home/gupg/warthog/warthog_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1583227688.078313091]: Program arguments: __name:=sick_rms_3xx
[ INFO] [1583227688.078342151]: Program arguments: __log:=/home/gupg/.ros/log/acc594c0-5d30-11ea-8e99-f0d5bfebbcac/sick_rms_3xx-1.log
[ INFO] [1583227688.152165126]: Found emul_sensor overwriting default settings. Emulation: False
[ INFO] [1583227688.154581534]: Start initialising scanner [Ip: 192.168.25.35] [Port: 2112]
[ INFO] [1583227688.334767452]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1583227688.357410957, 1637.410000000]: Parameter setting for <active_echo: 0>
[ INFO] [1583227688.563059115, 1637.620000000]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1583227688.565604969, 1637.620000000]: Receiving: <STX>sAN SetAccessMode 1<ETX>
[ INFO] [1583227688.765315586, 1637.820000000]: Sending  : <STX>sWN EIHstCola 0<ETX>
[ INFO] [1583227688.768612760, 1637.820000000]: Receiving: <STX>sWA EIHstCola<ETX>
[ INFO] [1583227688.962950765, 1638.020000000]: Sending  : <STX>sMN mSCloadappdef<ETX>
[ INFO] [1583227688.987848010, 1638.040000000]: Receiving: <STX>sAN mSCloadappdef<ETX>
[ INFO] [1583227689.179559426, 1638.240000000]: Sending  : <STX>sRN FirmwareVersion<ETX>
[ INFO] [1583227689.181899308, 1638.240000000]: Receiving: <STX>sRA FirmwareVersion A 1.3.0.309R<ETX>
[ INFO] [1583227689.378624001, 1638.440000000]: Sending  : <STX>sRN SCdevicestate<ETX>
[ INFO] [1583227689.380911570, 1638.440000000]: Receiving: <STX>sRA SCdevicestate 0<ETX>
[ INFO] [1583227689.579680345, 1638.640000000]: Sending  : <STX>sRN ODoprh<ETX>
[ INFO] [1583227689.582052532, 1638.640000000]: Receiving: <STX>sRA ODoprh 53<ETX>
[ INFO] [1583227689.782590264, 1638.840000000]: Sending  : <STX>sRN ODpwrc<ETX>
[ INFO] [1583227689.785034195, 1638.840000000]: Receiving: <STX>sRA ODpwrc 28<ETX>
[ INFO] [1583227689.988217298, 1639.050000000]: Sending  : <STX>sRN LocationName<ETX>
[ INFO] [1583227689.991013192, 1639.050000000]: Receiving: <STX>sRA LocationName B SN 20020230<ETX>
[ INFO] [1583227690.003097081, 1639.060000000]: Raw target transmission is switched [ON]
[ INFO] [1583227690.003165703, 1639.060000000]: Object transmission is switched [ON]
[ INFO] [1583227690.003231709, 1639.060000000]: Tracking mode is set to id [0] [BASIC]
[ INFO] [1583227690.003374963, 1639.060000000]: Sending  : <STX>sRN DItype<ETX>
[ INFO] [1583227690.005628524, 1639.060000000]: Receiving: <STX>sRA DItype D RMS3xx-xxxxxx<ETX>
[ INFO] [1583227690.005858494, 1639.060000000]: Sending  : <STX>sRN SerialNumber<ETX>
[ INFO] [1583227690.008143849, 1639.060000000]: Receiving: <STX>sRA SerialNumber 8 20020230<ETX>
[ INFO] [1583227690.009445567, 1639.060000000]: Sending  : <STX>sRN OrdNum<ETX>
[ INFO] [1583227690.010995578, 1639.070000000]: Receiving: <STX>sRA OrdNum 7 1083661<ETX>
[ INFO] [1583227690.011153053, 1639.070000000]: Sending  : <STX>sWN TransmitTargets 1<ETX>
[... 1 lines cut ...]
[ INFO] [1583227690.012769173, 1639.070000000]: Sending  : <STX>sWN TransmitObjects 1<ETX>
[ INFO] [1583227690.015292305, 1639.070000000]: Receiving: <STX>sWA TransmitObjects<ETX>
[ INFO] [1583227690.015465564, 1639.070000000]: Sending  : <STX>sWN TCTrackingMode 0<ETX>
[ INFO] [1583227690.016674803, 1639.070000000]: Receiving: <STX>sWA TCTrackingMode<ETX>
[ INFO] [1583227690.016842036, 1639.070000000]: Sending  : <STX>sMN Run<ETX>
[ INFO] [1583227690.021280218, 1639.070000000]: Receiving: <STX>sAN Run 1<ETX>
[ INFO] [1583227690.021459251, 1639.070000000]: Sending  : <STX>sRN SCdevicestate<ETX>
[ INFO] [1583227690.031996582, 1639.080000000]: Receiving: <STX>sRA SCdevicestate 0<ETX>
[ INFO] [1583227690.032799490, 1639.080000000]: Waiting for scanner ready state since 0 secs
[ INFO] [1583227691.021331592, 1640.080000000]: Sending  : <STX>sRN SCdevicestate<ETX>
[ INFO] [1583227691.023915192, 1640.080000000]: Receiving: <STX>sRA SCdevicestate 1<ETX>
[ INFO] [1583227691.024468897, 1640.080000000]: Scanner ready for measurement after 1 [sec]
[ INFO] [1583227691.024834684, 1640.080000000]: Sending  : <STX>sEN LMDradardata 1<ETX>
[ INFO] [1583227691.026854172, 1640.080000000]: Receiving: <STX>sEA LMDradardata 1<ETX>

and secondly the radar_object_marker.launch.

In rviz I visualize the output clouds and the market topic /radar_marker, and what I visualize is this:
[image]
 the radar marker just goes backwards and forwards in the X direction, which is different to what I am getting in windows, plus I only got to see one marker, and the orientation and Y position remain still. So I don't know where the problems is....
Is there something wrong? Any thoughts that will solve this issue?

Thanks a lot for your help.

Regards.

alfred31, 2020-03-18:
Hello again,

Has anyone checked if the driver for the radar works properly?

Just to know of it is already working which would mean that I did something wrong or on the contary I should check the driver in order to make my radar work with ros...

Best regards,

Thanks for the support.

michael1309, 2020-03-18:
Hi Alfred,

As you know, there are two types of messages:
The raw targets always point directly towards the radar or away from the radar, as these raw targets do not contain any tracking information. Rather, they serve as input values for tracking. The tracking targets on the other hand should follow the objects. You can check the results very well, e.g. by using 'rostopic echo' to display the topics on the console. You can also use the example programs to write a small program (or extend the example software) that allows you to receive and interpret the data from the radar yourself. For example you can use the routine 
`void callback(const sick_scan::RadarScan::ConstPtr &oa)` in the file `https://github.com/SICKAG/sick_scan/blob/master/tools/radar_object_marker/src/radar_object_marker.cpp`.
We also recommend that you use a Doppler generator for testing. If a Doppler generator is not available, get a large fan (e.g. [](https://www.amazon.de/Pro-Breeze-Chrom-Bodenventilator-Geschwindigkeitsstufen-Ventilatorkopf/dp/B07BQDHS7P/ref=sr_1_11_sspa? __mk_en_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=Gro%C3%9Fer+stand fan&qid=1584552382&sr=8-11-  spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExRk5KWUNNREk3Tk1PJmVuY3J5cHRlZElkPUEwMDY1ODQxQVVGWURDREJENzVJJJmVuY3J5cHRlZEFkSWQ9QTA2MDg1OTZTV0QyMFpPWlBGWlkmd2lkZ2V0TmFtZT1zcF9tdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl )),  to generate raw targets (not tracks).

alfred31, 2020-03-25:
Hello Michael,

Thanks again for your answer. I took a look at all the things you mentioned, but again I think I am missing something. Just a few questions to enlighten me:
Is there anyway I can get something similar to this in rviz? 
[image]
 If yes, which topics should I look at in rviz? 

I just can't reproduce what I saw in windows using ros... In windows I was able to see the detections, but with rviz I can't see anything understandable. I took a look at /cloud_radar_track and I was only able to see sparse points at very far distance from where the radar frame was, and even though I was moving around the room the points were not changing they were just periodically moving form one point to another.

Best Regards,
Thanks again for your support.

alfred31, 2020-03-25:
[Uploading sick_radar.ogv.zip…]()

alfred31, 2020-03-25:
[sick_radar.ogv.zip](https://github.com/SICKAG/sick_scan/files/4380699/sick_radar.ogv.zip)

michael1309, 2020-03-26:
Hello @alfred31 
Thank you very much for your information. We have taken your ticket as an impulse to test the complete support of the radar once again and to document the start-up in more detail. From our point of view the radar runs as described. We can see from your video that you have already been able to do this successfully. The tutorial information (incl. video) is now available under this link.
https://github.com/SICKAG/sick_scan/blob/master/doc/radar.md

You will find the statement to your video in a follow-up note.

michael1309, 2020-03-26:
Hello @alfred31 
1. In your viedeo you can see tracks with 20 m/s, probably the simulation is still active.
Double check if the parameter emul sensor is set to false.
```xml
    <param name="emul_sensor" type="bool" value="false" />
```
2. in our video you can see how the raw targets are visualized from the topic ``/cloud_radar_rawtarget``, this might help you.
3. Is the TF tree set up correctly? There might be an error here that leads to the rotated view in your video. 
4. To test the radar it is recommended to move a metallic object towards the radar.
When the radar is rotated in space, a large number of targets with different speeds are generated. However, due to the principle of the radar, no track can be generated from these targets.

closed 2020-03-26 by michael1309

alfred31, 2020-04-06:
Thanks for everything!

Amazing work.

Best Regards.