SICKAG/sick_scan #85: Node crashes on startup LMS511

friebelt, 2020-04-17:
Hi,
when I call `roslaunch sick_scan sick_lms_5xx.launch` it crashes and reports a pointer error. A older version of the node is working with the scanner (LMS511-10100).
I use Debian 9 with ROS Melodic

>cat /proc/version
Linux version 4.9.0-12-amd64 (debian-kernel@lists.debian.org) (gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1) ) #1 SMP Debian 4.9.210-1 (2020-01-20)

>roslaunch sick_scan sick_lms_5xx.launch 2>&1 | tee sick_scan_log.txt
cat sick_scan_log.txt
^[[0m[ INFO] [1587115921.422322184]: sick_generic_caller V. 1.5.5^[[0m
^[[0m[ INFO] [1587115921.422454246]: Program arguments: /home/user/test_ws/install/lib/sick_scan/sick_generic_caller^[[0m
^[[0m[ INFO] [1587115921.422488607]: Program arguments: __name:=sick_lms_5xx^[[0m
^[[0m[ INFO] [1587115921.422493750]: Program arguments: __log:=/home/user/.ros/log/49baf4fc-808e-11ea-a5ec-78d00426a53f/sick_lms_5xx-2.log^[[0m
^[[0m[ INFO] [1587115921.437153946]: Found sopas_protocol_type param overwriting default protocol:^[[0m
^[[0m[ INFO] [1587115921.438015746]: Binary protocol activated^[[0m
^[[0m[ INFO] [1587115921.438047405]: Start initialising scanner [Ip: 192.168.0.1] [Port: 2112]^[[0m
^[[0m[ INFO] [1587115921.459779431]: Publishing laserscan-pointcloud2 to cloud^[[0m
^[[0m[ INFO] [1587115921.463261441]: Parameter setting for <active_echo: 0>^[[0m
^[[0m[ INFO] [1587115921.664634935]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>^[[0m
^[[0m[ INFO] [1587115921.668357377]: Receiving: <STX>sAN SetAccessMode \x01<ETX>^[[0m
^[[0m[ INFO] [1587115921.868762053]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>^[[0m
^[[0m[ INFO] [1587115921.870065448]: Receiving: <STX>sWA EIHstCola <ETX>^[[0m
^[[0m[ INFO] [1587115922.070492275]: Sending  : <STX><STX><STX><STX><Len=0019>sRN FirmwareVersion CRC:<0x24>^[[0m
^[[0m[ INFO] [1587115922.071928090]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x31\x2e\x35\x30\x2e\x36\x20\x20<ETX>^[[0m
^[[0m[ INFO] [1587115922.272362499]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>^[[0m
^[[0m[ INFO] [1587115922.273689633]: Receiving: <STX>sRA SCdevicestate \x00<ETX>^[[0m
^[[0m[ INFO] [1587115922.474259249]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODoprh CRC:<0x41>^[[0m
^[[0m[ INFO] [1587115922.475596288]: Receiving: <STX>sRA ODoprh \x00\x00\xfd\x0c<ETX>^[[0m
^[[0m[ INFO] [1587115922.677456557]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODpwrc CRC:<0x52>^[[0m
^[[0m[ INFO] [1587115922.678767079]: Receiving: <STX>sRA ODpwrc \x00\x00\x04\x3a<ETX>^[[0m
^[[0m[ INFO] [1587115922.880634668]: Sending  : <STX><STX><STX><STX><Len=0016>sRN LocationName CRC:<0x55>^[[0m
^[[0m[ INFO] [1587115922.881966501]: Receiving: <STX>sRA LocationName \x00\x0b\x6e\x6f\x74\x20\x64\x65\x66\x69\x6e\x65\x64<ETX>^[[0m
^[[0m[ INFO] [1587115922.883829850]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>^[[0m
^[[0m[ INFO] [1587115922.885134177]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>^[[0m
^[[0m[ INFO] [1587115922.885287113]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)^[[0m
^[[0m[ INFO] [1587115922.885361427]: [From:To] -5.00000 [deg] to 185.00000 [deg] (in 1/10000th deg: from 0xFFFF3CB0 to 0x1C3A90)^[[0m
^[[0m[ INFO] [1587115922.885419233]: MIN_ANG:   -1.658 [rad]  -95.000 [deg]^[[0m
^[[0m[ INFO] [1587115922.885474946]: MAX_ANG:    1.658 [rad]   95.000 [deg]^[[0m
^[[0m[ INFO] [1587115922.885742581]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x09 0xc4 0xff 0xff 0x3c 0xb0 0x00 0x1c 0x3a 0x90 CRC:<0x8d>^[[0m
^[[0m[ INFO] [1587115922.886960324]: Receiving: <STX>sWA LMPoutputRange <ETX>^[[0m
^[[0m[ INFO] [1587115922.887188706]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>^[[0m
^[[0m[ INFO] [1587115922.888483781]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>^[[0m
^[[0m[ INFO] [1587115922.888634816]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)^[[0m
^[[0m[ INFO] [1587115922.891252824]: MIN_ANG (after command verification):   -1.658 [rad]  -95.000 [deg]^[[0m
^[[0m[ INFO] [1587115922.891312491]: MAX_ANG (after command verification):    1.658 [rad]   95.000 [deg]^[[0m
^[[0m[ INFO] [1587115922.891349399]: LMS 5xx detected overwriting output channel flag ID^[[0m
^[[0m[ INFO] [1587115922.891390711]: LMS 5xx detected overwriting resolution flag (only 8 bit supported)^[[0m
^[[0m[ INFO] [1587115922.891654540]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x*** Error in `/home/user/test_ws/install/lib/sick_scan/sick_generic_caller': free(): invalid pointer: 0x000056480a04cbb0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x70bfb)[0x7f10a83bfbfb]
/lib/x86_64-linux-gnu/libc.so.6(+0x76fc6)[0x7f10a83c5fc6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7780e)[0x7f10a83c680e]
/home/user/test_ws/install/lib/libsick_scan_lib.so(_ZN9__gnu_cxx13new_allocatorIfE10deallocateEPfm+0x20)[0x7f10d2475844]
/home/user/test_ws/install/lib/libsick_scan_lib.so(_ZNSt16allocator_traitsISaIfEE10deallocateERS0_Pfm+0x2b)[0x7f10d2471f70]
/home/user/test_ws/install/lib/libsick_scan_lib.so(_ZNSt12_Vector_baseIfSaIfEE13_M_deallocateEPfm+0x32)[0x7f10d246caee]
/home/user/test_ws/install/lib/libsick_scan_lib.so(_ZNSt12_Vector_baseIfSaIfEED2Ev+0x41)[0x7f10d2463167]
/home/user/test_ws/install/lib/libsick_scan_lib.so(_ZNSt6vectorIfSaIfEED1Ev+0x41)[0x7f10d2458bbd]
/home/user/test_ws/install/lib/libsick_scan_lib.so(_ZN9sick_scan14SickScanCommon8loopOnceEv+0x3d3e)[0x7f10d243cfba]
/home/user/test_ws/install/lib/sick_scan/sick_generic_caller(_Z16mainGenericLaseriPPcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x16f9)[0x56480970bc01]
/home/user/test_ws/install/lib/sick_scan/sick_generic_caller(main+0x798)[0x56480970ff04]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1)[0x7f10a836f2e1]
/home/user/test_ws/install/lib/sick_scan/sick_generic_caller(_start+0x2a)[0x564809709d4a]
======= Memory map: ========
5648096f8000-564809717000 r-xp 00000000 08:12 2366822                    /home/user/test_ws/install/lib/sick_scan/sick_generic_caller
564809916000-564809918000 r--p 0001e000 08:12 2366822                    /home/user/test_ws/install/lib/sick_scan/sick_generic_caller
564809918000-564809919000 rw-p 00020000 08:12 2366822                    /home/user/test_ws/install/lib/sick_scan/sick_generic_caller
564809f39000-56480a068000 rw-p 00000000 00:00 0                          [heap]
7f1060000000-7f1060021000 rw-p 00000000 00:00 0
7f1060021000-7f1064000000 ---p 00000000 00:00 0
7f1068000000-7f1068021000 rw-p 00000000 00:00 0
...

michael1309, 2020-04-23:
Unfortunately, I can't replicate the crash. In my system the driver runs here without crash with the LMS511. Could you exchange the CMakeLists.txt with the file in the ticket as a test and then run the code in debug mode and send me the log file?
[CMakeLists.txt](https://github.com/SICKAG/sick_scan/files/4524128/CMakeLists.txt)

friebelt, 2020-04-28:
Hi thanks for the reply,
I recompiled the node with the supplied CMakeLists and set the log level in `$ROS_ROOT/config/rosconsole.config`  to debug.
The Problem still persists. I've uploaded all files from the log folder. I'll also try to get a new Firmware for the scanner if this is possible.

[master.log](https://github.com/SICKAG/sick_scan/files/4545700/master.log)
[roslaunch-L0ggingB0x-4249.log](https://github.com/SICKAG/sick_scan/files/4545702/roslaunch-L0ggingB0x-4249.log)
[rosout.log](https://github.com/SICKAG/sick_scan/files/4545703/rosout.log)
[rosout-1-stdout.log](https://github.com/SICKAG/sick_scan/files/4545704/rosout-1-stdout.log)

Best regards,
Tobi

michael1309, 2020-04-29:
Hi Tobi,
could you send in addition the console output and the log file generated by the launch?
You can find the log file name for example by inspecting 
```
ps -aef|grep "sick_generic_caller"
```
or in the roslaunch.log

Unfortunately it is not possible to see the problem directly from the log files. Could you possibly do the following steps:
1. Start roscore
2. Open a second Terminal
3. Launch the driver:
```
roslaunch sick_scan sick_lms_5xx.launch
```
4. In a third terminal:
```
ps -aef|grep "sick_generic_caller"
```
The output looks something like this:
```
rosuser 27781 27756 3 09:31 ? 00:00:09 /home/rosuser/catkin_ws_ros/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/rosuser/.ros/log/67af4cea-89eb-11ea-9c58-70b3d5570ff0/sick_lms_5xx-2.log
rosuser 27876 2763 0 09:35 pts/1 00:00:00 grep --color=auto sick_generic_caller
```
5. Cancel the program with kill.
In this case:
```
kill -9 27781
```
6. Change to the Repo. directory
Here:
```
~/catkin_ws_ros/src/sick_scan$
```
7. Start gdb as follows:
```
gdb --args /home/rosuser/catkin_ws_ros/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/rosuser/.ros/log/t.log
```
8.
```
run 
```

9. After crash you can check the backtrace
```
bt
```

9. Report the result in this issue (e.g. as a screenshot).

friebelt, 2020-04-29:
Alright I followed all the steps
Here is the output of gdb (there was no additional log file written):
[image]

The .log file stated by ps is not written either, so I redirected the output to file with ` roslaunch sick_scan sick_lms_5xx.launch 2>&1 | tee log.txt`
Result: [log.txt](https://github.com/SICKAG/sick_scan/files/4552582/log.txt)

I've also tried the Scanner on a different pc (same os and same ros) with the same outcome. It seems to be a problem with the scanner, a setting, or the fw. It is still running with a older version of the node though.
Let me know if I can be be of assistance somehow.

Greetings,
Tobi

michael1309, 2020-04-29:
Hi Tobi,
As you have seen, I have removed almost all sources for debugging from the library definition in the CMakeLists.txt In order not to patch the build process too much, only the file driver/src/softwarePLL.cpp is the source used to build the lib.This means that gdb cannot find this library, because exactly one function from this source is not found.

a) Could you copy the libsick_scan_lib.so from devel/lib to src/sick_scan just for testing and repeat the gdb process? 
b) Could you also send the screenshot of the driver startup (as you did when you opened the ticket)?
c) Which version of this node runs without problems with your lidar? 

Thanks in advance!

friebelt, 2020-04-30:
Hi!
I'm sorry I've never used gdb and have always been protected by a safeguarding IDE. Many things to learn for me :)
The working version of the node is:  sick_generic_caller V. 001.002.005 (this is what it tells me at startup)

I copied the lib folder and started gdb again, here is the output:

> user@L0ggingB0x:~/test_ws$ gdb --args /home/user/test_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/user/.ros/log/t.log
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/user/test_ws/devel/lib/sick_scan/sick_generic_caller...done.
(gdb) run
Starting program: /home/user/test_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/user/.ros/log/t.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[ INFO] [1588249579.197577086]: sick_generic_caller V. 1.5.5
[ INFO] [1588249579.197726320]: Program arguments: /home/user/test_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1588249579.197763631]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1588249579.197773123]: Program arguments: __log:=/home/user/.ros/log/t.log
[DEBUG] [1588249579.197872960]: remap: __name => sick_lms_5xx
[DEBUG] [1588249579.197905699]: remap: __log => /home/user/.ros/log/t.log
[DEBUG] [1588249579.231996131]: Adding tcp socket [7] to pollset
[DEBUG] [1588249579.232271586]: UDPROS server listening on port [48394]
[New Thread 0x7fffa7f2c700 (LWP 18743)]
[New Thread 0x7fffa772b700 (LWP 18744)]
[New Thread 0x7fffa6f2a700 (LWP 18745)]
[New Thread 0x7fffa6729700 (LWP 18746)]
[DEBUG] [1588249579.235332425]: Started node [/sick_lms_5xx], pid [18710], bound on [L0ggingB0x], xmlrpc port [37539], tcpros port [43683], using [real] time
Warning: recursive print statement has occurred.  Throwing out recursive print.
[DEBUG] [1588249579.237430591]: XML-RPC call [getParam] returned an error (-1): [Parameter [/sick_lms_5xx/new_IP_address] is not set]
[DEBUG] [1588249579.239476169]: XML-RPC call [getParam] returned an error (-1): [Parameter [/sick_lms_5xx/range_min] is not set]
[DEBUG] [1588249579.239987234]: XML-RPC call [getParam] returned an error (-1): [Parameter [/sick_lms_5xx/time_increment] is not set]
[DEBUG] [1588249579.240298872]: XML-RPC call [getParam] returned an error (-1): [Parameter [/sick_lms_5xx/emul_sensor] is not set]
[ INFO] [1588249579.240613570]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1588249579.240629481]: Binary protocol activated
[ INFO] [1588249579.240646503]: Start initialising scanner [Ip: 192.168.1.140] [Port: 2112]
[DEBUG] [1588249579.241840755]: XML-RPC call [getParam] returned an error (-1): [Parameter [/sick_lms_5xx/diagnostic_period] is not set]
[DEBUG] [1588249579.261457587]: XML-RPC call [getParam] returned an error (-1): [Parameter [/sick_lms_5xx/cloud_topic] is not set]
[ INFO] [1588249579.261479457]: Publishing laserscan-pointcloud2 to cloud
[New Thread 0x7fffa5e35700 (LWP 18747)]
[DEBUG] [1588249579.264828610]: XML-RPC call [getParam] returned an error (-1): [Parameter [/sick_lms_5xx/new_IP_address] is not set]
[DEBUG] [1588249579.265156792]: XML-RPC call [getParam] returned an error (-1): [Parameter [/sick_lms_5xx/ntp_server_address] is not set]
[DEBUG] [1588249579.265438201]: XML-RPC call [getParam] returned an error (-1): [Parameter [/sick_lms_5xx/active_echos] is not set]
[ INFO] [1588249579.265457118]: Parameter setting for <active_echo: 0>
[DEBUG] [1588249579.455398885]: Accepted connection on socket [7], new socket [15]
[DEBUG] [1588249579.456080878]: Adding tcp socket [15] to pollset
[DEBUG] [1588249579.456175663]: TCPROS received a connection from [127.0.0.1:59610]
[DEBUG] [1588249579.457017624]: Connection: Creating TransportSubscriberLink for topic [/rosout] connected to [callerid=[/rosout] address=[TCPROS connection on port 43683 to [127.0.0.1:59610 on socket 15]]]
[DEBUG] [1588249579.466217942]: Command: <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1588249579.466647538]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1588249579.470339409]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[DEBUG] [1588249579.670856051]: Command: <STX>sWN EIHstCola 1<ETX>
[ INFO] [1588249579.671044271]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1588249579.672345099]: Receiving: <STX>sWA EIHstCola <ETX>
[DEBUG] [1588249579.872852729]: Command: <STX>sRN FirmwareVersion<ETX>
[ INFO] [1588249579.873099522]: Sending  : <STX><STX><STX><STX><Len=0019>sRN FirmwareVersion CRC:<0x24>
[ INFO] [1588249579.874423167]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x31\x2e\x35\x30\x2e\x36\x20\x20<ETX>
[DEBUG] [1588249580.074967426]: Command: <STX>sRN SCdevicestate<ETX>
[... cut ...]
#11 0x00005555556dc5b6 in mainGenericLaser (argc=1, argv=0x7fffffffdee8, nodeName="sick_lms_5xx")
    at /home/user/test_ws/src/sick_scan/driver/src/sick_generic_laser.cpp:358
#12 0x00005555556dd8fa in main (argc=3, argv=0x7fffffffdee8) at /home/user/test_ws/src/sick_scan/driver/src/sick_generic_caller.cpp:195

**and the output when I start it with roslaunch**
I had to copy it to a file since it is to long:

[sick_console_output.log](https://github.com/SICKAG/sick_scan/files/4558446/sick_console_output.log)

michael1309, 2020-05-03:
Thanks for your effort. It seems that when the driver releases a float vector (by using the destructor), the program crashes. 
I will check the code again for this. But it might be a side effect that is not so easy to reproduce, because the driver is running on my test computer.

friebelt, 2020-05-04:
I found a solution. I've reset the device to factory default and did the FW update and now it is running. Could have thought of that earlier.

Thanks for your help!

michael1309, 2020-05-04:
Great! Could you send me the FW version before and after the update?

friebelt, 2020-05-05:
Shure! The version before the update was 1.50.6-07.08.2015 and has been updated to V1.83.1

Best Regrads
Tobias

closed 2020-05-14 by friebelt