SICKAG/sick_scan #86: Feature - scan range calibration

MitchAllain, 2020-05-15:
This is not an issue with the driver; however recently we have been seeing inconsistency in systematic error from new TiM571-2050101 units. About 20% of new sensors are arriving with > 60 mm of systematic range error. The remaining 80% tend to be within 20 or 10 mm.

We have spoken with SICK technical support, and aside from exchanging these units, they have recommended adding a range scaling offset from calibration.

Is a measurement range offset something that has been discussed in the context of this driver? I believe it would be relatively easy to add, but I was curious if this issue has come up elsewhere. This would also allow sensors which are within SICK's spec to be calibrated to even higher accuracy.

Thanks, Mitchell

michael1309, 2020-05-16:
We understand your point. We will clarify it with the SICK management in May and will then give you feedback on the decision.

stg710, 2020-06-08:
@MitchAllain: Thanks for your request. Am I right in assuming that the topic was also brought up via the SICK USA service department? This is already in progress. We will close this ticket and inform you about the progress.

MitchAllain, 2020-06-08:
> @MitchAllain: Thanks for your request. Am I right in assuming that the topic was also brought up via the SICK USA service department? This is already in progress. We will close this ticket and inform you about the progress.

Yes, in fact we discovered that we were able to drastically improve measurement error by covering the rear of the scanner with an opaque cover. We have notified SICK of this resolution, and it does not seem that we will be needing a range calibration feature. As always, appreciate your willingness to help, and we'll provide as much info to SICK as we can to help others avoid this issue when mounting the TiM scanners.

closed 2020-06-08 by MitchAllain

michael1309, 2020-06-09:
Very good to hear.  If you should try something anyway, I recommend to make the change behind line 3515 in the file sick_scan_common.cpp (V.1.7.2) and adjust 'range_meter' accordingly. But this should not be necessary anymore due to the new information. Thank you for sharing these results with us.