SICKAG/sick_scan #89: Changed password, can no longer login

rachel-spykerman, 2020-06-05:
Hi,

I'm running into a few issues with the sick_scan package as well as Sopas tool to interface with the SICK LIDAR's. I have two SICK LIDAR's set up (a front one and back one, both TiM781S). I have a top level ros launch file that sets a few parameters for each LIDAR (see attached txt file
[sick_sensors.txt](https://github.com/SICKAG/sick_scan/files/4737243/sick_sensors.txt)
) and then calls the sick_scan provided launch file for the TiM781S. I'm running into several errors when launching, the first of which seems to be an issue with logging into each LIDAR. The setAccessMode returns a 0, and subsequent errors occur with EIHstCola, LMPoutputRange, and LMDscandatacfg. I'm able to receive data on all expected topics, I can see the scans in RVIZ, and I can detect obstacles, so everything seems to be functioning correct, but I don't know how to get rid of these errors. 

Since it doesn't seem to be able to login when calling setAccessMode, I've read on another issue (https://github.com/SICKAG/sick_scan/issues/4) that the password needs to be a default password of "client". I had initially changed it to something different, but to try and fix this login issue, I tried changing it to "client" through Sopas tool, but now I'm not able to log into the LIDAR at all. Neither my old password or the "client" password work.

Two questions: 
1) How do I resolve not being able to log into the SICK LIDAR after changing the password?
2) How do I resolve the roslaunch errros? 

Please let me know if you need any more info.

Here's the roslaunch output: 

started roslaunch server http://beta1:33383/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /sick_scan_back/cloud_topic: back_cloud
 * /sick_scan_back/frame_id: /laser_back
 * /sick_scan_back/hostname: 192.168.1.204
 * /sick_scan_back/intensity: True
 * /sick_scan_back/max_ang: 2.35619449
 * /sick_scan_back/min_ang: -2.35619449
 * /sick_scan_back/port: 2112
 * /sick_scan_back/range_max: 100.0
 * /sick_scan_back/scanner_type: sick_tim_7xxS
 * /sick_scan_back/timelimit: 5
 * /sick_scan_back/use_binary_protocol: False
 * /sick_scan_front/cloud_topic: front_cloud
 * /sick_scan_front/frame_id: /laser_front
 * /sick_scan_front/hostname: 192.168.1.203
 * /sick_scan_front/intensity: True
 * /sick_scan_front/max_ang: 2.35619449
 * /sick_scan_front/min_ang: -2.35619449
 * /sick_scan_front/port: 2112
 * /sick_scan_front/range_max: 100.0
 * /sick_scan_front/scanner_type: sick_tim_7xxS
 * /sick_scan_front/timelimit: 5
 * /sick_scan_front/use_binary_protocol: False

NODES
  /
    sick_scan_back (sick_scan/sick_generic_caller)
    sick_scan_front (sick_scan/sick_generic_caller)
    zone_activation_node (sick_safety/zone_activation_node)

auto-starting new master
process[master]: started with pid [8061]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1dc0cb7e-a744-11ea-af21-ac40ea0193f7
process[rosout-1]: started with pid [8098]
started core service [/rosout]
process[sick_scan_front-2]: started with pid [8116]
[${node}] [1591372159.553648299] [ INFO] [main():184]: sick_generic_caller V. 1.6.0
[${node}] [1591372159.553701310] [ INFO] [main():192]: Program arguments: /opt/ros/kinetic/lib/sick_scan/sick_generic_caller
[${node}] [1591372159.553715154] [ INFO] [main():192]: Program arguments: scan:=scan_front
[${node}] [1591372159.553728935] [ INFO] [main():192]: Program arguments: cloud:=cloud_front
[${node}] [1591372159.553771774] [ INFO] [main():192]: Program arguments: __name:=sick_scan_front
[${node}] [1591372159.553799879] [ INFO] [main():192]: Program arguments: __log:=/home/rachel/Dev/canvas/beta-borg/artifacts/log/1dc0cb7e-a744-11ea-af21-ac40ea0193f7/sick_scan_front-2.log
process[sick_scan_back-3]: started with pid [8117]
[${node}] [1591372159.562670592] [ INFO] [main():184]: sick_generic_caller V. 1.6.0
[${node}] [1591372159.562714863] [ INFO] [main():192]: Program arguments: /opt/ros/kinetic/lib/sick_scan/sick_generic_caller
[... 84 lines cut ...]
[/sick_scan_back] [1591372161.080137578] [ERROR] [SickScanCommon::sendSopasAndCheckAnswer():750]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX>sWN LMPoutputRange 1 D05 FFF92230 225510<ETX>Answer= >>>sFA A<<<

[/sick_scan_back] [1591372161.080183106] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():717]: Sending  : <STX>sRN LMPoutputRange<ETX>
[/sick_scan_front] [1591372161.081917162] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():723]: Receiving: <STX>sAN Run 1<ETX>
[/sick_scan_front] [1591372161.081962641] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():717]: Sending  : <STX>sEN LMDscandata 1<ETX>
[/sick_scan_back] [1591372161.082342933] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():723]: Receiving: <STX>sRA LMPoutputRange 1 D05 FFF92230 225510<ETX>
[/sick_scan_back] [1591372161.082382281] [ INFO] [SickScanCommon::init_scanner():1885]: Angle resolution of scanner is 0.33330 [deg]  (in 1/10000th deg: 0xD05)
[/sick_scan_front] [1591372161.083066414] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():723]: Receiving: <STX>sEA LMDscandata 1<ETX>
[/sick_scan_back] [1591372161.083881968] [ INFO] [SickScanCommon::init_scanner():1909]: MIN_ANG (after command verification):   -2.356 [rad] -135.000 [deg]
[/sick_scan_back] [1591372161.083919071] [ INFO] [SickScanCommon::init_scanner():1911]: MAX_ANG (after command verification):    2.356 [rad]  135.000 [deg]
[/sick_scan_back] [1591372161.083941803] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():717]: Sending  : <STX>sWN LMDscandatacfg 01 00 1 1 0 0 00 0 0 0 1 1<ETX>
[/sick_scan_back] [1591372161.085029091] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():723]: Receiving: <STX>sFA A<ETX>
[/sick_scan_back] [1591372161.085071723] [ERROR] [SickScanCommon::sendSopasAndCheckAnswer():750]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX>sWN LMDscandatacfg 01 00 1 1 0 0 00 0 0 0 1 1<ETX>Answer= >>>sFA A<<<

[/sick_scan_back] [1591372161.085098133] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():717]: Sending  : <STX>sRN LMDscandatacfg<ETX>
[/sick_scan_back] [1591372161.086212068] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():723]: Receiving: <STX>sRA LMDscandatacfg 1 0 1 0 0 0 0 0 1 0 0 1<ETX>
[/sick_scan_back] [1591372161.087236901] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():717]: Sending  : <STX>sMN LMCstartmeas<ETX>
[/sick_scan_back] [1591372161.089414073] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():723]: Receiving: <STX>sFA 1<ETX>
[/sick_scan_back] [1591372161.089450352] [ERROR] [SickScanCommon::sendSopasAndCheckAnswer():750]: Error Sopas answer mismatch Command or Error message not definedAnswer= >>>sFA 1<<<

[/sick_scan_back] [1591372161.089493173] [ERROR] [SickScanCommon::init_scanner():2244]: Command or Error message not defined
[/sick_scan_back] [1591372161.089537408] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():717]: Sending  : <STX>sMN Run<ETX>
[/sick_scan_back] [1591372161.090640815] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():723]: Receiving: <STX>sAN Run 1<ETX>
[/sick_scan_back] [1591372161.090662248] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():717]: Sending  : <STX>sEN LMDscandata 1<ETX>
[/sick_scan_back] [1591372161.091796792] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():723]: Receiving: <STX>sEA LMDscandata 1<ETX>

stg710, 2020-06-08:
Hi @rachel-spykerman, the launch file "sick_tim_7xxS.launch" works with the default password stated in the manual on page 32
https://cdn.sick.com/media/docs/4/54/254/Operating_instructions_TiM781S_en_IM0083254.PDF

rachel-spykerman, 2020-06-08:
So both LIDAR's I set up had their passwords changed from SICKSAFE to something else. Does sick_scan expect the default password to be used? If changed back to the default password, would the above setAccessMode errors be resolved? I run into other errors with EIHstCola, LMPoutputRange, and LMDscandatacfg. When checking the parameters I've set in my launch files and the parameters I've set in Sopas, I don't see any mismatch. At this point, I'm assuming these errors occur because the setAccessMode failed.

michael1309, 2020-06-09:
The hash codes for the access mode stored in the code are encoded in the following commands:
```
    sopasCmdVec[CMD_SET_ACCESS_MODE_3] = "\x02sMN SetAccessMode 3 F4724744\x03\0";
    sopasCmdVec[CMD_SET_ACCESS_MODE_3_SAFETY_SCANNER] = "\x02sMN SetAccessMode 3 6FD62C05\x03\0";
```
For the safety scanner, the hash code '6FD62C05' corresponds to the standard password SICKSAFE (according to section 6.5.6 of the above mentioned document https://cdn.sick.com/media/docs/4/54/254/Operating_instructions_TiM781S_en_IM0083254.PDF ) of the safety scanner.

Have you tried to reset the scanner to the delivery status? After that, logging in should be possible again.  Otherwise, I recommend to contact the support team to reset the lidar to the default settings.

michael1309, 2020-06-09:
[image]
[image]

michael1309, 2020-06-09:
If you have now assigned your own password, you must patch the code according to this calculation. 
sopasCmdVec[CMD_SET_ACCESS_MODE_3_SAFETY_SCANNER] = "\x02sMN SetAccessMode 3 **6FD62C05**\x03\0";

rachel-spykerman, 2020-06-09:
Great, thank you! That worked.

closed 2020-06-26 by michael1309