SICKAG/sick_scan #92: Compatibility with LMS400

jajberni, 2020-07-07:
Is there any way to make this driver run with the LMS400. It seems like the LMS4xxx is an evolution of the LMS400, but despite activating the Binary mode and modifying the port, there is no way it can talk to the LMS400. It seems like the command/acknowledgment/reply is different from other models:

[ INFO] [1594130468.203699150]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1594130468.207510700]: Receiving: <STX>sMA SetAccessMode <ETX>
[ERROR] [1594130468.208400347]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>Answer= >>>sMA SetAccessMode <<<

[ INFO] [1594130468.002088373]: Parameter setting for <active_echo: 0>
[ INFO] [1594130468.203699150]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1594130468.207510700]: Receiving: <STX>sMA SetAccessMode <ETX>
[ERROR] [1594130468.208400347]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>Answer= >>>sMA SetAccessMode <<<

[ INFO] [1594130468.208650591]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1594130468.209044997]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1594130468.409501611]: Sending  : <STX>sWN EIHstCola 1<ETX>
[ERROR] [1594130474.252637879]: no answer received after 5000 ms. Maybe sopas mode is wrong.

Is there any way around this or any way to update the firmware on the LMS400?

Thank you in advance!

stg710, 2020-07-10:
Hey @jajberni, there is no LMS400 ROS driver from SICK. The device has been discontinued one year ago. The Successor (LMS4000) is fully compatible with this ROS driver. The underlying API is completely different compared to the LMS400.

michael1309, 2020-07-11:
@stg710 Thanks for clarifying. So we close this ticket.

closed 2020-07-11 by michael1309