SICKAG/sick_scan #94: Angle scandata for output

williamledda, 2020-08-07:
Hello, 
I have a TIM718s and I would like to configure the angle of the scan data for output. 
For doing this, I'm sending the "sWN LMPoutputRange" telegram that, according to the manual (8014631, section 4.3.2), should work even for TIMxxx series, but I'm getting a  Sopas_Error_VARIABLE_WRITE_ACCESSDENIED as reply. Is really compatible? What I'm doing wrong? I'm connected as Authorized client.

The command that I'm sending is (0.333 res., range 85 - 95 deg.)

sWN LMPoutputRange 1 D05 CF850 E7EF0

Reply:
sFA A

I have the same error when issuing the command from the SOPAS ET terminal.

michael1309, 2020-08-07:
Hello @williamledda:

So you are using a 7xxxS so a security scanner.
Here you have to use a different login password to change scan parameters.
The error message ```Sopas_Error_VARIABLE_WRITE_ACCESSDENIED``` matches a wrong password. 
That's why you get the same error when using Sopas ET.
The procedure is as follows:
Log in with Password.
Change parameters.
Log out.
Start scandata acquisition. 

See also section 6.56 in the [manual](https://cdn.sick.com/media/docs/4/54/254/Operating_instructions_TiM781S_en_IM0083254.PDF) for the default Password.
I hope we could help you.

best regards Michael.

williamledda, 2020-08-07:
Hi Michael, 
thank you, it was actually a login problem that I did not realized from the error message (a bit misleading in the description: "It is not allowed to write values to this variable. Probably the variable is defined as read-only.")
Thank you for the quick reply!

closed 2020-08-07 by williamledda