SICKAG/sick_scan #95: Issues Driver ROS LMS511

esterge, 2020-08-18:
Dear Colleagues,

One customer is using a driver for the integration of the LMS511 lasers for ROS which is available at http://wiki.ros.org/sick_scan and from time to time we encounter the followin problem
 
"Lookup would require extrapolation into the past. Requested time 1594073504.179511308 but the earliest data is at time 1594073508.659801551, when looking up transform from frame [lms511_1]"
 
We don't know how to fix this.
Could you help us?
 
 Additionally, Indra has another doubt. Also they need to know if it can e used port 2111 for tasks like login; contamination sensors status, etc and, simultaneously, use port 2112 with ROS and binary protocol for the application.
Is it possible?

Best regards

michael1309, 2020-08-24:
Hi,

have you tried to use the software-pll mechanism? See #98.

michael1309, 2020-08-24:
Currently we only support one port for the communication. But it is no problem to switch from ASCII to binary and vice versa.

esterge, 2020-08-26:
Hi Michael,

I don't know if the customer is using the software-pll, I will check it, 
Anyway I have sent to them the link with the explanation.
Regarding the other topic I have explained to the customer.

Best regards

closed 2020-09-14 by michael1309