SICKAG/sick_scan #98: About time stamp

chance2006, 2020-08-21:
Dear Sir:
I need to use the TIM‘s time stamp(time stamp,Time since start up in µs ,Time of transmission in µs), Because I want to Syn. with other device. 
[image]

Coud you help me to acchive it?
 
Best Regards!
Bob Hao

michael1309, 2020-08-24:
HI Bob,

exactly for this case we introduced the mechanism of the so called SoftwarePLL. 
You can find more information here: https://github.com/SICKAG/sick_scan/blob/master/doc/software_pll.md

The mechanism can be switched on by setting the flag `sw_pll_only_publish` to true.
Best regards,
Michael

closed 2020-08-24 by michael1309

chance2006, 2020-08-31:
Dear Sir：
Thanks very much！I have understood it.
Best Regards!
Chester Hao