SICKAG/sick_scan #100: LMS511-20100 Pro Driver use questions

WangQJuan, 2020-08-26:
5xx.launch
In the process of using, I want to configure the angular resolution and rotation speed of the laser. There was a problem.
At present, I can only use the default 35Hz

I'd like to ask you how to do it. Thank you

1.set   <param name="use_binary_protocol" type="bool" value="true"/>
	<param name="ang_res" type="double" value="0.1667" />
        <param name="scan_freq" type="double" value="25" />
[ERROR] [1598438538.450053089]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0036>sMN mLMPsetscancfg 0x2b 0x02 0x05 0x00 0x00 0x20 0x01 0x20 0x2b 0x01 0x06 0x06 0x07 0x20 0x00 0x20 0x00 CRC:<0x53>Answer= >>>sFA\x00\x08<<<

2.set   <param name="use_binary_protocol" type="bool" value="false"/>
	<param name="ang_res" type="double" value="0.1667" />
        <param name="scan_freq" type="double" value="25" />
[ WARN] [1598438519.558494118]: Data length is outside acceptable range 1-381 (1141). Ignoring scan
[ WARN] [1598438519.597887486]: Data length is outside acceptable range 1-381 (1141). Ignoring scan
[ WARN] [1598438519.638020280]: Data length is outside acceptable range 1-381 (1141). Ignoring scan
[ WARN] [1598438519.677837551]: Data length is outside acceptable range 1-381 (1141). Ignoring scan

michael1309, 2020-08-29:
What is your preferred setting for angular resolution and rotation speed?

WangQJuan, 2020-08-29:
I understand that angular resolution is relative to rotation speed            I need a 0.1667 0.5 configuration. Of course, I want to be able to configure all the resolutions supported by the device

chance2006, 2020-09-02:
Dear Micheal：
WangQJuan is our customer. He found that he must get  an error if he change the frequency or angular resolution. They directly change 48&49 lines in our launch file
https://github.com/SICKAG/sick_scan/blob/master/launch/sick_lms_5xx.launch
The error 
 [ERROR] [1598438538.450053089]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <Len=0036>sMN mLMPsetscancfg 0x2b 0x02 0x05 0x00 0x00 0x20 0x01 0x20 0x2b 0x01 0x06 0x06 0x07 0x20 0x00 0x20 0x00 CRC:<0x53>Answer= >>>sFA\x00\x08<<<
Woud you kindly check it? Thanks very much!

Best Regards!
Chester Hao

michael1309, 2020-09-02:
Dear Chester,

Thanks for your feedback. We have fixed this inconsistency and checked in the new version 1.7.8. 
Please check out this version in the master branch. 
This should solve the customer's issue.

Best Regards,
Michael

commit 2fd93bc8 (2020-09-02): fixes #100
closed 2020-09-02 by michael1309