SICKAG/sick_scan #102: Sick_generic_caller

schleptech, 2020-09-14:
Getting attached error[image]

michael1309, 2020-09-14:
We have seen that you build the ROS node under Raspberry-Pi. This platform is not officially supported by us. But it seems that the ROS system does not start correctly. You can try to start roscore once in a separate thread and then run the launch command again.

In addition (or as an alternative) you can try to start `sick_generic_caller` directly. The roslaunch command will in detail start a process with argument like this:
```
sick_generic_caller __name:=sick_lms_1xxx __log:=/tmp/debuglog.log
```
You could check this by using the `ps -afx|grep sick_generic_caller`.

Also you must source the project ros environment by 
using a command sequence like this:
```
cd ~/ros_catkin_ws
source ./devel/setup.bash
```
Otherwise the launch command cannot find `sick_generic_caller`.

Best regards,
Michael

schleptech, 2020-09-14:
Thanks for your response, Michael.

I will give that a try this evening to see. What platforms do you officially support? If that doesn’t work, I will start over with clean slate from one of your recommended platforms. Ubuntu?

michael1309, 2020-09-15:
The default support is for Ubuntu x86 systems.

closed 2020-10-09 by michael1309