SICKAG/sick_scan #103: sick_generic_caller use 100% cpu after Network reconnected

twxjyg, 2020-09-22:
Our robot has a router, and our Tim571 laser scanner and ECU connected to the router. When I reboot our router, the sick_generic_caller ros node will be stoped to send laser scan data. At the same time, the cpu load will be 100%. Seems like the ros node stacked by the network. 
After I  reboot the sick_generic_caller ros node, every thing be good. laser scan data published and CPU load becomes normal (around 0.7&).

michael1309, 2020-09-22:
Hello twxjyg,

We cannot reproduce the error you describe.

For testing we have chosen two setups:
1. setup: Tim571 Static Ip Notebook Static Ip
    Test: Network cable removed and reconnected.
Result: After reconnecting some packets are dropped by the software PLL and logged.
Afterwards the operation is continued normally.

2. setup: Tim571 Static Ip Notebook Static Ip Connection via Router
Test: Power cycling of the router.
Result: The network connection is lost. CPU load remains constantly low.

In general the driver should always be restarted when the network connection is physically disconnected.
We also recommend the use of Static IP
 
best regards
Michael

twxjyg, 2020-09-22:
Thank you Michael,
Here is my launch file major part, I think we are using static IP, and our High CPU load issue can is reproducable, do you have other suggestion about our configuration ? for example do you have re-connection options ?

```xml

<launch>
  <arg name="hostname" default="192.168.8.100" />
  <arg name="cloud_topic" default="cloud" />
  <arg name="frame_id" default="laser" />
  <node name="sick_tim_5xx" pkg="sick_scan" type="sick_generic_caller"  respawn="false" output="log">

  <!--<param name="robot_description" command="$(find xacro)/xacro.py '$(find sick_scan)/urdf/example.urdf.xacro'" />-->

    <!-- Default: min_angle = -135 degree, max_angle = 135 -->
    <!-- default values: -->
    <!--
      <param name="scanner_type" type="string" value="sick_tim_5xx" />
      <param name="min_ang" type="double" value="-2.35619449" />
      <param name="max_ang" type="double" value="2.35619449" />
      <param name="intensity" type="bool" value="True" />
      <param name="skip" type="int" value="0" />
      <param name="frame_id" type="str" value="laser" />
      <param name="time_offset" type="double" value="-0.001" />
      <param name="publish_datagram" type="bool" value="False" />
      <param name="subscribe_datagram" type="bool" value="false" />
      <param name="device_number" type="int" value="0" />
      <param name="range_min" type="double" value="0.05" />
    -->
    <param name="scanner_type" type="string" value="sick_tim_5xx" />

    <!-- -135° -->
    <param name="min_ang" type="double" value="-1.65" />

    <!-- 135° -->
    <param name="max_ang" type="double" value="1.65" />

    <param name="use_binary_protocol" type="bool" value="true" />
    <param name="range_max" type="double" value="25.0" />
    <param name="intensity" type="bool" value="True" />

    <param name="hostname" type="string" value="$(arg hostname)" />
    <param name="cloud_topic" type="string" value="$(arg cloud_topic)"/>
    <param name="frame_id" type="str" value="$(arg frame_id)" />
    <param name="port" type="string" value="2112" />
    <param name="timelimit" type="int" value="5" />
    <remap from="scan" to="raw_scan" />
  </node>

</launch>

```

michael1309, 2020-10-09:
Is this still an issue? As mentioned we could not reproduce it. For this reason we will close this ticket. If there is still a finding you can reopen it.

closed 2020-10-09 by michael1309